Koenkk/zigbee2mqtt #8380: TuYa WHD02 paired as TS0001

drdownload, 2021-08-15:
# Bug Report
## What happened
I have this Switch: https://zigbee.blakadder.com/Tuya_WHD02.html => looks exactly like in this picture. 

It gets paired as TS0001. (which has no power on setting that I would need)

Blakadder also mentions that it sometimes uses TS0001 as Zigbee ID => understandable that it pairs as TS0001.

Also the WHD02 page has another  image https://www.zigbee2mqtt.io/devices/WHD02.html. So maybe there different switches with the same model-name and same/different Zigbee IDs, or maybe  an older version.

## What did you expect to happen
Get paired as WHD02.

## How to reproduce it (minimal and precise)
Just hold the pairing button on the  device and it gets paired as TS0001

## Debug Info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC2652RB
Adapter firmware version: zStack3x0 20210120

Koenkk, 2021-08-15:
- First make sure to update to the latest z2m (1.21.0, you are running 1.18.1).
If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0
- If that doesn't help share the data/database.db entry of your device here.

closed 2021-09-22 by github-actions[bot]

wileu, 2022-01-29:
Hi. I have the same problem as above.
Z2M 1.22.2
Adapter: CC2652P
firmware: zStack3x0 2021217

vindolin, 2022-03-31:
I have the exact same problem.
I'm on 1.24.0-dev commit: [95921479]

Here's my database.db
[zigbee2mqtt_database.zip](https://github.com/Koenkk/zigbee2mqtt/files/8388329/zigbee2mqtt_database.zip)