Koenkk/zigbee2mqtt #8381: TuYa TS0002 misbehaves.

SargonofAssyria, 2021-08-15:
This device https://www.zigbee2mqtt.io/devices/TS0002.html#tuya-ts0002
Has in the database
`{"id":45,"type":"Router","ieeeAddr":"0x60a423fffe112aa3","nwkAddr":60887,"manufId":4098,"manufName":"_TZ3000_svoqrno4","powerSource":"Mains (single phase)","modelId":"TS0002","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0002","manufacturerName":"_TZ3000_svoqrno4","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca1b0bc","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca1b0bc","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":3},"lastSeen":1629016828713}
`
But in the state.json file
`    "0x60a423fffe112aa3": {
        "state_l1": "OFF",
        "state_l2": "OFF",
        "state": "ON",
        "brightness": 203,
        "color_temp": 300,
        "color": {
            "x": 0.4599,
            "y": 0.4106
        }
    },
`
In every message is brightness, color_temp and color present.
`info  2021-08-15 12:11:29: MQTT publish: topic 'zigbee2mqtt/Slaapkamer', payload '{"brightness":203,"color":{"x":0.4599,"y":0.4106},"color_temp":300,"last_seen":"2021-08-15T12:11:29+02:00","linkquality":39,"state":"ON","state_l1":"ON","state_l2":"OFF"}'
`
I already deleted the state.json file but that does not solve the problem.
Using:
`Version: 2021.1 (build 13360)
Build Hash: eee333c30
Compile Date: 2021-07-04 11:58:11
dzVents Version: 3.1.8
Python Version: 3.7.3 (default, Jan 22 2021, 20:04:44) [GCC 8.3.0]

and 

Zigbee2MQTT version 1.21.0-dev commit: c18a698e
Coordinator type zStack3x0
Coordinator revision 20210708
Frontend version 0.5.25
`
How do I get rid of the brightness, color_temp and color?

Koenkk, 2021-08-15:
Is this device in a group? If yes it might be that the state came from here, I believe this is already fixed in a recent z2m version and it should not readd the colortemp etc after you clear the current state (stop z2m and remove it from `state.json`).

SargonofAssyria, 2021-08-15:
No this device is NOT in a group, but was sometime ago. I have the the latest dev from today and deleted the state.json before I started zigbee2mqtt again. But the brightness, color_temp and color are still present on the device.

Koenkk, 2021-08-16:
I don't understand why those get added back, could you provide the debug log starting from the point where you start z2m with a clean state.json until these properties get published?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

SargonofAssyria, 2021-08-16:
This device works correctly now and the brightness, color_temp and color are not present anymore, but I got a few other errors back in Domoticz and from the plugin.
`2021-08-16 18:23:27.008 Error: Zigbee: (Zigbee) 'onMessage' failed 'TypeError':'unsupported operand type(s) for /: 'NoneType' and 'int''.
2021-08-16 18:23:27.008 Error: Zigbee: (Zigbee) ----> Line 297 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMessage
2021-08-16 18:23:27.008 Error: Zigbee: (Zigbee) ----> Line 137 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMessage
2021-08-16 18:23:27.008 Error: Zigbee: (Zigbee) ----> Line 167 in '/home/pi/domoticz/plugins/zigbee2mqtt/mqtt.py', function onMessage
2021-08-16 18:23:27.008 Error: Zigbee: (Zigbee) ----> Line 217 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMQTTPublish
2021-08-16 18:23:27.008 Error: Zigbee: (Zigbee) ----> Line 63 in '/home/pi/domoticz/plugins/zigbee2mqtt/devices_manager.py', function handle_mqtt_message
2021-08-16 18:23:27.008 Error: Zigbee: (Zigbee) ----> Line 56 in '/home/pi/domoticz/plugins/zigbee2mqtt/adapters/base_adapter.py', function handle_mqtt_message
2021-08-16 18:23:27.008 Error: Zigbee: (Zigbee) ----> Line 27 in '/home/pi/domoticz/plugins/zigbee2mqtt/adapters/lumi/DJT11LM.py', function convert_message
2021-08-16 18:23:27.008 Error: Zigbee: (Zigbee) ----> Line 31 in '/home/pi/domoticz/plugins/zigbee2mqtt/adapters/adapter_with_battery.py', function convert_message
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) 'onMessage' failed 'TypeError':'conversion from NoneType to Decimal is not supported'.
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) ----> Line 297 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMessage
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) ----> Line 137 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMessage
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) ----> Line 167 in '/home/pi/domoticz/plugins/zigbee2mqtt/mqtt.py', function onMessage
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) ----> Line 217 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMQTTPublish
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) ----> Line 63 in '/home/pi/domoticz/plugins/zigbee2mqtt/devices_manager.py', function handle_mqtt_message
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) ----> Line 64 in '/home/pi/domoticz/plugins/zigbee2mqtt/adapters/base_adapter.py', function handle_mqtt_message
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) ----> Line 132 in '/home/pi/domoticz/plugins/zigbee2mqtt/devices/device.py', function handle_message
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) ----> Line 111 in '/home/pi/domoticz/plugins/zigbee2mqtt/devices/device.py', function get_device_args
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) ----> Line 44 in '/home/pi/domoticz/plugins/zigbee2mqtt/devices/temperature_humidity_sensor.py', function get_string_value
2021-08-16 18:25:22.588 Error: Zigbee: (Zigbee) ----> Line 28 in '/home/pi/domoticz/plugins/zigbee2mqtt/devices/temperature_humidity_sensor.py', function get_temperature_value`

SargonofAssyria, 2021-08-16:
Found out that the device was sending 'null' for several values. After inclusion again the errors are gone.

closed 2021-08-16 by SargonofAssyria

WollyWonka, 2021-12-27:
I'm experiencing the same issue as mentioned here. However, I don't understand the "solution" (After inclusion) mentioned by SargonofAssyria. I was trying to pair a device that has been paired before successfully (I'm rebuilding my system from scratch as I was getting many issues before.). 

Domoticz Version 2021.1 (build 13953)
Zigbee2mqtt plugin version 3.1.0-beta
Zigbee2mqtt version 1.22.1-dev
zStack12 20201128