Koenkk/zigbee2mqtt #8382: Namron Led bulb 6w RGBW

larronni, 2021-08-15:
## Information about the device + link
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['3802967'],
    model: '3802967',
    vendor: 'Namron',
    description: 'Led bulb 6w RGBW',
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;

URL:
[https://www.namron.com/products/led-paere-6w-rgbw-gu10-zigbee/](url)

## data/database.db entry of the device
{"id":76,"type":"Router","ieeeAddr":"0x5c0272fffe445322","nwkAddr":58599,"manufId":4761,"manufName":"Namron AS","powerSource":"Mains (single phase)","modelId":"3802967","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64642],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"3802967","manufacturerName":"Namron AS","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":101,"hwVersion":1,"dateCode":"20200423-675","swBuildId":"2.4"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":555}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":101,"hwVersion":1,"dateCode":"20200423-675","swBuildId":"2.4","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1629018866467}

Koenkk, 2021-08-15:
Please also add the color temperature range: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature

larronni, 2021-08-16:
Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":555}

Koenkk, 2021-08-17:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-17 by Koenkk