Koenkk/zigbee2mqtt #8383: soft_reset_timeout does not work with tcp connection

sjorge, 2021-08-15:
## What happened
While migrating to a new VM I noticed the `soft_reset_timeout` property. Seemed like a good thing to enable.

It will always reset when the timeout is reached, even if there is activity happening.

## What did you expect to happen
I guess to not do anything when using a tcp connection to the stick or work as expected.

## How to reproduce it (minimal and precise)

Set the timeout to say 5 sec, connect to the stick over tcp.

## Debug info
Zigbee2MQTT version: Zigbee2MQTT version 1.21.0-dev (commit #c18a698e)
Adapter hardware: CC1352P-2
Adapter firmware version:

Koenkk, 2021-08-15:
`soft_reset_timeout` is deprecated, why do you want to use it?

sjorge, 2021-08-15:
Must have skipped over that it's deprecated.

closed 2021-08-15 by sjorge