Koenkk/zigbee2mqtt #8384: Using converter to change device funccion

pedroxns, 2021-08-15:
## Is your feature request related to a problem? Please describe
I´m trying to make a doorbell sensor using a contact sensor as a dry contact sensor

## Describe the solution you'd like
Make zigbee2mqtt and/or homekit see a contact sensor as a doorbell.

## Describe alternatives you've considered
connect a sensor from a different brand than the ones that are already into my network, all my sensor are Aqara - thinking about using a sonoff one. 
Change the Sonoff sensor atributes and make homekit sees it as a doorbell that rings when the contact is closed (doorbell switch presssed)

## Additional context
Add any other context or screenshots about the feature request here.

danieledwardgeorgehitchcock, 2021-08-16:
If this is the only device of that type you have in your Zigbee2MQTT network, you could use the same code from the device in an external converter.

Zigbee2MQTT prioritises external converters over its built in converters so, when you remove the device from the network and re-join it, it should use the one that you have created

rcarmo, 2021-09-22:
I've done the exact same thing as @pedroxns (bell, Node-RED and all), but would love to be able to remap a contact sensor as an occupation sensor in Zigbee2MQTT directly because it would be a cleaner solution (and require no maintenance after the configuration). 

Is there any documented way to change a device type using a converter? I thought it only handled events, so any examples would be great.

rcarmo, 2021-09-22:
In the meantime I went digging through the `externalConverters` example and see that there is apparently no way to override a _single_ device, just a device type. Since I am using multiple contact sensors and only want to convert _one_ of them to an occupation sensor (or a single use button, which might be simpler), there doesn't seem to be a way to do that.

pedroxns, 2021-09-22:
Hey, how are you?
After a few days trying to make the doorbell with the sensor I decided to try a "cleaner" way to do it.
In the end, the project was done with a tasmotized sonoff mini and homebridge-camera-Ffmpeg plug-in. I think it's better this way because there is no need of and external program running a script, everything is done by tasmota rules, making the relays decouple to stop sending power to the buzzer (from 18:30 to 8:00) and reverting it to default at 8:01 until 18:29, so actually I'm not even using zigbee and stopped trying to make the converter changes (as I've tryed a lot without any success).
But in fact the main issue was that HomeKit just recognize a doorbell with a video camera. The zigbee converter would have to make the contact sensor be seen as a doorbell camera, not as a "blind" doorbell, or as a movement sensor that triggers a fake door bell camera.
So, if you have a project like the one I did before, I sugest you to take a look at this other way to make it, the mqtt message send by the sonoff can be read by any other plugin or program, as my automation is a hybrid HomeKit + Alexa, a my Apple devices gives me notifications and I have Alexa routines as well, depending on what I want and on the time of the day. I even have a Home app button to change the buzzer status (on/off).

rcarmo, 2021-09-22:
I've gone down that path, but I prefer to have the ability to reconfigure an individual sensor to masquerade as another kind of device. Makes things simpler to maintain.

closed 2021-10-31 by github-actions[bot]