Koenkk/zigbee2mqtt #8385: UnhandledPromiseRejectionWarning and zigbee2mqtt fails

Jogobo, 2021-08-15:
# Bug Report
## What happened
zigbee2mqtt fails nearly every day with UnhandledPromiseRejectionWarning

## What did you expect to happen
zigbee2mqtt running without failure

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version:  1.21.0
Adapter hardware: Slaesh
Adapter firmware version: 20210708
Node version: 14.17.5
npm version: 7.20.6

zigbee2mqtt.log:
```
Aug 15 18:29:19 jogolinux npm[21019]: (node:21030) UnhandledPromiseRejectionWarning: Error: SRSP - SYS - version after 6000ms
Aug 15 18:29:19 jogolinux npm[21019]:     at Timeout._onTimeout (/usr/local/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
Aug 15 18:29:19 jogolinux npm[21019]:     at listOnTimeout (internal/timers.js:557:17)
Aug 15 18:29:19 jogolinux npm[21019]:     at processTimers (internal/timers.js:500:7)
Aug 15 18:29:19 jogolinux npm[21019]: (Use `node --trace-warnings ...` to show where the warning was created)
Aug 15 18:29:19 jogolinux npm[21019]: (node:21030) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 245)
Aug 15 18:29:19 jogolinux npm[21019]: (node:21030) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit cod
e.

```

Koenkk, 2021-08-15:
Please provide the herdsman debug logging of this.

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Jogobo, 2021-08-16:
Debug logging is activated. Waiting for the next failure.

Jogobo, 2021-08-19:
As I don't know which part of the debug log is important I added the complete logfile (145 MB) zip compressed.
[zigbee2mqtt.log.debug.zip](https://github.com/Koenkk/zigbee2mqtt/files/7015502/zigbee2mqtt.log.debug.zip)

Koenkk, 2021-08-19:
This is not the herdsman debug logging but the normal debug logging (See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging ). I would also suggest to remove `availability_timeout` from your `configuration.yaml`, this should drastically reduce the stress on your coordinator.  I'm currently working on refactoring the availability feature (#6281)

Jogobo, 2021-08-21:
zigbee2mqtt failed again last night. Difference with `availability_timeout` disabled is that there is no "UnhandledPromiseRejection" error in the logs and no "Timeout.js" errors. When zigbee2mqtt starts to be unresponsive there are pairs of log messages like this:
`Aug 20 23:30:06 jogolinux npm[13534]: ^[[31mZigbee2MQTT:error^[[39m 2021-08-20 23:30:06: Publish 'set' 'state' to 'mqttGroup_Fensterlicht' failed: 'Error: Command 1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":0}) failed (SRSP -
 AF - dataRequestExt after 6000ms)'
Aug 20 23:30:06 jogolinux npm[13534]: ^[[32mZigbee2MQTT:info ^[[39m 2021-08-20 23:30:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'mqttGroup_Fensterlicht' failed: 'Error: Command 1 genLevelCtrl.moveToLevelWithOnOff({\"level\":0,\"transtime\":0}) failed (SRSP - AF - dataRequestExt after 6000ms)'","meta":{"friendly_name":"mqttGroup_Fensterlicht"},"type":"zigbee_publish_error"}'
`

Koenkk, 2021-08-21:
Do you have the herdman debug logging of this?

Jogobo, 2021-08-21:
Sorry, but no. I disabled herdsman debgging after sending the last logfile.

Jogobo, 2021-08-21:
It might be that battery devices are the main cause for these errors.
I have several battery based motion sensors (Osram Smart+). Tonight the light did not work though the LED of the motion sensor showed that a movement was detected. So I tried to switch on the lights manually. There was a delay but the lights could be turned on and off.
After restarting zigbee2mqqt everything worked like a charm and no delay could be detected. Maybe I can turn on herdsman debugging again tomorrow when the next failure occurs. I will upload the logfile later.

Jogobo, 2021-08-21:
Interesting fact:
The JSON message for the motion sensors looks like this:
`    "battery_low": false,
    "last_seen": "2021-08-21T19:04:12.523Z",
    "linkquality": 0,
    "occupancy": false,
    "tamper": true,
    "temperature": 20.28`

But the device is not tampered.

Jogobo, 2021-08-31:
Since my last restart of Z2M it is running without any problems. Still debugging herdsman but so far no malfunction. I have no idea what's going on.

Jogobo, 2021-09-19:
Changed to 1.21.1. Now the problem is back. Attached part of the debug log.
[zigbee_debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/7191105/zigbee_debug.txt)

Koenkk, 2021-09-20:
Thanks, what is the output of `dmesg` the moment after it fails?

Jogobo, 2021-09-21:
The dmesg output is really hard to get. It is not instantly obvious that the system failed again. I control most of my lights (some by motion sensors, some by timers) with z2m. So I only realize a failure when a light, that is supposed to turn on, does not turn on. That might be minutes or hours after z2m failed.
I'm running z2m on a Dell Server with an "Intel(R) Xeon(R) CPU E3-1225 v3 @ 3.20GHz" and 16 GB RAM. OS is openSuSE 15.1. Keyboard and Mouse are directly connected via USB. `node` version is 14.17.5, `npm` is 7.20.6, `java` is 1.8.0_275.

closed 2021-10-31 by github-actions[bot]