Koenkk/zigbee2mqtt #8391: Zigbee2MQTT wont start after HA container switch

drdownload, 2021-08-15:
## What happened
I switched to the new HA repo to upgrade to latest zigbee2mqtt. First z2m wont start with a message that the pan id is already in use. after coping my backup of coordinator.backup and replugging the stick it moved along but stopped at. "tables need to be same length"

Zigbee2MQTT:error 2021-08-15 23:04:50: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-08-15 23:04:50: Failed to start zigbee
Zigbee2MQTT:error 2021-08-15 23:04:50: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-08-15 23:04:50: Exiting...
Zigbee2MQTT:error 2021-08-15 23:04:50: Error: All table entries need to be the same length

## What did you expect to happen
z2m would start

## How to reproduce it (minimal and precise)
see above

## Debug info
Zigbee2MQTT version: 1.18.1 => 1.21-
Adapter hardware: CC2652RB
Adapter firmware version: zStack3x0 20210120

Koenkk, 2021-08-16:
Looks something wrong with the backup/restore mechanism, could you provide the herdsman debug logging so @castorw can check?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

drdownload, 2021-08-17:
Im not sure if I caputured everything, if someting interesting is missing give me a headsup.

https://pastebin.com/Pi8QAL4F

drdownload, 2021-08-20:
Since my family asks why some automations are not working as they are used to: would it get fixed by reflashing the zigbee coordinator and repairing (I have only 2 hidden Zigbee devices)

Koenkk, 2021-08-21:
@drdownload that will definitely fix it.