Koenkk/zigbee2mqtt #8403: Mi smartplug not reporting power

sygys, 2021-08-17:
I have a mi smartplug correctly connected to Z2MQTT and it shows in the list as interview completed true. It reports all values except 2. One of them is power and the other is current. I can however turn it on and off and when it was connected using ZHA it the power usage did work.

In the logs i get these errors in z2m:

Debug Received Zigbee message from 'Wasmachine smartplug on_off', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,1,3,40,26,152,57,0,0,0,0,149,57,129,249,234,66,150,57,0,64,13,69,151,57,0,0,0,0,5,33,1,0,154,32,0,8,33,22,1,7,39,0,0,0,0,0,0,0,0,9,33,3,15,11,32,0,155,16,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug lumi.plug.mmeu01: unknown index 5 with value 1
Debug lumi.plug.mmeu01: unknown index 154 with value 0
Debug lumi.plug.mmeu01: unknown index 8 with value 278
Debug lumi.plug.mmeu01: unknown index 7 with value null
Debug lumi.plug.mmeu01: unknown index 9 with value 3843
Debug lumi.plug.mmeu01: unknown index 11 with value 0
Debug lumi.plug.mmeu01: unknown index 155 with value 1
Info MQTT publish: topic 'zigbee2mqtt/Wasmachine smartplug on_off', payload '{"auto_off":null,"consumer_connected":true,"consumer_overload":null,"consumption":117.49,"current":0,"energy":117.49,"led_disabled_night":null,"linkquality":39,"power":0,"power_outage_memory":null,"state":"ON","temperature":26,"update":{"state":"idle"},"update_available":false,"voltage":226}'

Received MQTT message on 'zigbee2mqtt/Wasmachine smartplug on_off/get' with data '{"power":""}'
Debug Publishing get 'get' 'power' to 'Wasmachine smartplug on_off'
Error Publish 'get' 'power' to 'Wasmachine smartplug on_off' failed: 'TypeError: Cannot read property 'read' of undefined'

Koenkk, 2021-08-17:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/7745

Can you attach a known load to the device (e.g. a 60 watt bulb) and provide such logging again:

```
Debug Received Zigbee message from 'Wasmachine smartplug on_off', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,1,3,40,26,152,57,0,0,0,0,149,57,129,249,234,66,150,57,0,64,13,69,151,57,0,0,0,0,5,33,1,0,154,32,0,8,33,22,1,7,39,0,0,0,0,0,0,0,0,9,33,3,15,11,32,0,155,16,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug lumi.plug.mmeu01: unknown index 5 with value 1
Debug lumi.plug.mmeu01: unknown index 154 with value 0
Debug lumi.plug.mmeu01: unknown index 8 with value 278
Debug lumi.plug.mmeu01: unknown index 7 with value null
Debug lumi.plug.mmeu01: unknown index 9 with value 3843
Debug lumi.plug.mmeu01: unknown index 11 with value 0
Debug lumi.plug.mmeu01: unknown index 155 with value 1
Info MQTT publish: topic 'zigbee2mqtt/Wasmachine smartplug on_off', payload '{"auto_off":null,"consumer_connected":true,"consumer_overload":null,"consumption":117.49,"current":0,"energy":117.49,"led_disabled_night":null,"linkquality":39,"power":0,"power_outage_memory":null,"state":"ON","temperature":26,"update":{"state":"idle"},"update_available":false,"voltage":226}'
```

I want to check if any of the `unknown index XX with value XX` should be the power. (please post in #7745)

closed 2021-08-17 by Koenkk

sygys, 2021-08-18:
I re-paired the device a couple of times and it seems to work now. I guess something went wrong in the interview