Koenkk/zigbee2mqtt #8406: Xiaomi devices join even though permit_join set to false

rocket59, 2021-08-17:
Running 1.21 using a Conbee II. Even though permit join is disabled previously unpaired xiaomi sensors will join when the sensor is reset (I have another zigbee mesh running on a different channel. This did not occur when I was using a CC2531 as coordinator.

Oddly this behaviour only occurs if the permit join drop down box is set to 'all' (I'm using the z2m front end). If permit join is limited to one of the routers in the my z2m mesh network then new devices will not pair. So it looks like the conbee is ignoring the permit join restriction and pairing to any device regardless of whether permit join is disabled or not.

Koenkk, 2021-08-17:
Did this Xiaomi device ever connect to the network previously?

rocket59, 2021-08-17:
I don't believe so but I'll see if I can find something that I can say for sure was never connected and test it out. Will revert with the result once I've managed to find one.

rocket59, 2021-08-17:
Ok I found another couple of xiaomi sensors as well as a brand new smartthings sensor. Both paired immediately to the mesh with permit join disabled. Another xiaomi sensor paired even when permit join was restricted to one of the routers. 

As a related issue I found that I can't delete any of the devices in the blocklist using the frontend. Clicking on the '-' for a device causes it to disappear but instantly reappear.

Koenkk, 2021-08-18:
- Could you share the sniff when a device that has not been paired previously pairs with permit join disabled? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
- @chrishae is this a known issue of the conbee?

> As a related issue I found that I can't delete any of the devices in the blocklist using the frontend. Clicking on the '-' for a device causes it to disappear but instantly reappear.

Already fixed in the dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

rocket59, 2021-08-18:
Will do, thanks.

ChrisHae, 2021-08-20:
Not as far as I know. When the network is closed no device will join.
I have just tested it with the newest zigbee2mqtt version (1.21.0) and a xiaomi vibration sensor. It did not join a closed network.
But I just used the configuration.yaml to open and close the network. Perhaps it gets opened in his case for some reason.

rocket59, 2021-08-20:
> * Could you share the sniff

[image]

Does this help? First time using wireshark so not sure what to provide

z-x-f, 2021-08-20:
Export it and attach your password

rocket59, 2021-08-20:
Ok so I did what the poster did here https://github.com/Koenkk/zigbee2mqtt/issues/538#issuecomment-487324640  as well as change the pan id and lo and behold things seem to be working correctly now. I did not delete the existing devices, however they would not re-pair if permit join is set to false (as it should be) and new devices also do not pair. 

Thanks @Koenkk. For future reference though and to help you troubleshoot, should I be attaching the export of the entire sniffed output as suggested by @z-x-f (thanks for that!) although I'm not sure what the password should be?

Koenkk, 2021-08-20:
@rocket59 with password he means network key, 2 things go wrong in this sniff:
- packet 19: a router (what device is this? ieeAddr = `0x680ae2fffe793921`) sending a association response, this should only be done when joining is enabled.
- packet 31: the coordinator (conbee) sends a transport key, this shouldn't be done when permit join is disabled. 

Both point are not directly a bug of z2m since both are handled in the router/conbee firmware. To be sure that joining is disabled. Can you:

- start sniff
- enable join
- wait 5 seconds
- disable join
- wait 5 seconds
- pair device

After that please share the sniff with your network key here.

rocket59, 2021-08-20:
Unfortunately I probably can't provide any useful insights any more as after messing with z2m a bit (as I mentioned above) everything seems to be working correctly (ie devices now only pair when permit join is enabled, I tested with multiple devices including previously paired ones and a brand new one). 

(However in response to the points raised the router referred to is a smartthings plug which was already paired to the network. I have no idea why it was doing that)

At this point I can only keep any eye on it. If strange behaviour resurfaces I will definitely send the sniff as per your instructions. I guess at this stage the issue can probably be marked as closed?

rocket59, 2021-08-20:
Hi again @Koenkk Sorry I misunderstood your post. Here is the sniff file 
[Z2M sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/7024371/Z2M.sniff.zip)

the network key is 78:CE:42:4E:13:03:35:C0:0F:60:7B:E8:82:47:24:F5

So looks like everything is ok? I reset a new device after 10s but no joining occurs.

Koenkk, 2021-08-21:
@rocket59 this sniff looks OK indeed, as you can see the `Beacon` of the Xiaomi device is not followed up by a `Beacon` from any of the routers/coordinator in your network.

rocket59, 2021-08-21:
Nice - thanks for looking at this. The Conbee II really is a massive step up from a CC2531 as well as being readily available. I don't need touchlink (at least not now), although are you aware if this is likely to be supported? That seems to  be the only thing that you can't do with a conbee.

closed 2021-08-22 by rocket59

ChrisHae, 2021-08-23:
@rocket59 Touchlink is a feature that will eventually be implemented in the Conbee2 firmware. But there is no ETA yet.

rocket59, 2021-08-23:
No problem - thanks for the update.