Koenkk/zigbee2mqtt #8408: YSRSAI YSR-MINI-01_rgbcct color temp range max 370 mireds on HA 500 on web panel z2m

shirou93, 2021-08-17:
Zigbee2MQTT version: 1.21.0
Adapter hardware: CC2652RB
Adapter firmware version: 20210708

https://www.zigbee2mqtt.io/devices/YSR-MINI-01_rgbcct.html#ysrsai-ysr-mini-01_rgbcct

max on ha:
[image]

z2m panel:
[image]

Koenkk, 2021-08-18:
Could you provide the color temp range? https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature

shirou93, 2021-08-18:
` MQTT publish: topic 'zigbee2mqtt/Kuchnia LED', payload '{"brightness":254,"color":{"h":133,"hue":133,"s":100,"saturation":100,"x":0.164,"y":0.588},"color_mode":"xy","color_temp":111,"last_seen":"2021-08-18T19:17:52+02:00","linkquality":78,"state":"OFF"}'

Info Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":160,"colorTempPhysicalMax":370}`

Koenkk, 2021-08-19:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-19 by Koenkk