Koenkk/zigbee2mqtt #8409: Wrong HA status topic in 1.21

DubhAd, 2021-08-17:
## What happened
Looking in the Z2M UI I see that the status topic for HA is
```
            "homeassistant_status_topic": "hass/status",
```
The documentation and source say it should be `homeassistant/status`.

It _isn't_ configured in `configuration.yaml`, or any other configuration file.

## What did you expect to happen

The status topic to match the documentation.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware:  CC26X2R1 (ZZH)
Adapter firmware version: 20210708

Koenkk, 2021-08-18:
This is a legacy thing, previously ha did not use a default birth/last will topic so users might have the following in their config: https://www.home-assistant.io/docs/mqtt/birth_will/

Z2M always subscribes to both `hass/status` and `homeassistant/status`.  (https://github.com/Koenkk/zigbee2mqtt/blob/70891eec39018303a43349c875ad04d44b3b43cf/lib/extension/homeassistant.js#L704)

DubhAd, 2021-08-18:
> This is a legacy thing, previously ha did not use a default birth/last will topic so users might have the following in their config: https://www.home-assistant.io/docs/mqtt/birth_will/
> 
> Z2M always subscribes to both `hass/status` and `homeassistant/status`. (

Ah, so the UI is misleading then... that's good to know.

Thanks

closed 2021-08-18 by DubhAd