Koenkk/zigbee2mqtt #8410: npm run build fails under windows

vladtsit, 2021-08-17:
## npm run build fails 

## successful build

## clone 1.21.0 to windows, run: npm run build

## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware: not applicable
Adapter firmware version: not applicable

**npm run build
*
> zigbee2mqtt@1.21.0 build
> node_modules/.bin/tsc && node index.js writehash

'node_modules' is not recognized as an internal or external command,
operable program or batch file.

nurikk, 2021-08-18:
do `npm ci` first

vladtsit, 2021-08-18:
I did npm ci before. Then npm start failed with npm run build error.

Koenkk, 2021-08-19:
Works fine for me, what npm version are you using?

```
C:\Users\koen\Desktop\zigbee2mqtt>npm --version
6.14.10

C:\Users\koen\Desktop\zigbee2mqtt>npm run build

> zigbee2mqtt@1.21.0 build C:\Users\koen\Desktop\zigbee2mqtt
> tsc && node index.js writehash

C:\Users\koen\Desktop\zigbee2mqtt>
```

egony, 2021-08-22:
> I did npm ci before. Then npm start failed with npm run build error.

**npm ci --production**  - build failed when **npm start**
**npm ci** - build ok, z2m started

node-v10.23.0-x64

Koenkk, 2021-08-22:
`npm ci --production - build failed when npm start` yes this is expected, since no typescript is available then.

if you do a:
```
npm ci
npm run build
rm -rf node_modules
npm ci --production
```

it will work fine.

vladtsit, 2021-08-23:
> Works fine for me, what npm version are you using?

npm -v
7.15.0

node -v
v14.17.5

something wrong with new versions?

Koenkk, 2021-08-23:
Did an attempt to fix it, can you check the latest dev branch? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

vladtsit, 2021-08-23:
now it works! Thank you!