Koenkk/zigbee2mqtt #8414: Mercator Ikuü inline Dimmer (Zigbee) SISWD01

reidyrobot, 2021-08-18:
Hello team 

I've had my electrician install this rebranded tuya dimmer device and am using it with Zigbee2MQTT version 1.21.0-dev installed in a docker container. Like most of these tuya devices in the many threads I've read I can successfully pair it but the device shows in the device list as unsupported:

Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ojzhk75b'
warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

I feel quite the noob here and I really have tried to add support for the new device by creating an external converter but I just couldn't get Zigbee2MQTT to even start when referencing the external .js file. I think possibly I'm missing something very fundamental in that I can only seem to bind the /app/data folder outside the container. Do I need to bind /run/udev as well??

**Information about the device + link**
The device range:
https://www.ikuu.com.au/
The specific device :
https://www.ikuu.com.au/product/inline-switch-with-dimmer/
https://www.ikuu.com.au/downloadable-assets/product-specifications/MIDS000033Z%20Data%20Sheet.pdf

**data/database.db entry of the device**
{"id":2,"type":"Router","ieeeAddr":"0xbc55acfffwf0d7ff","nwkAddr":34586,"manufId":4098,"manufName":"_TZE200_ojzhk75b","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,4,5,6,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ojzhk75b","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1627361770238}

I gather a few very similar tuya devices (including the mercator ones) have been added to the supported list in the latest version of zigbee2mqtt so I'm just hoping against hope that it may not be a difficult thing to add this one. I'd really like to be able to do it myself of course - would be really grateful for any guidance, although I realise i haven't given much to go on.

Koenkk, 2021-08-18:
Did an attempt to add it, it should be recognised as a `TS0601_dimmer` but I'm not sure if dimming/turning on/off will work correctly.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

reidyrobot, 2021-08-19:
> Did an attempt to add it, it should be recognised as a `TS0601_dimmer` but I'm not sure if dimming/turning on/off will work correctly.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

thanks Koenkk, it actually seems to be working perfectly in home assistant now - turning off and on and dimming all functioning well.  fantastic work!

Koenkk, 2021-08-19:
Great, assuming this can be closed now.

closed 2021-08-19 by Koenkk