Koenkk/zigbee2mqtt #8415: GU10 ZigBee Smart Bulb (Chinese Tuya White Label)

nwaelti, 2021-08-18:
Hello all, 

I've been able to sucessfully add other Zigbee Tuya devices in the past days, but this one gives me a "Failed interview" back, can someone point me in the right direction? I'm on debian with a  CC2652RB USB Key.
This is the debuglog
[image]
[image]

Thanks a lot in advance,
Nick

Koenkk, 2021-08-18:
Node descriptor failing indicates the device did not respond to anything (it's the first step of the interview procedure). Have you tried pairing it closer to the coordinator?

If that doesn't help, can you make a sniff of the pairing? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

nwaelti, 2021-08-18:
Well... easiest things are often the best ones :) I feel a bit bad about it but you were right. I assumed zigbee interview would also work in mesh and that one had to place the devices to their final location... It's now working great, thanks a milion

closed 2021-08-18 by nwaelti

Koenkk, 2021-08-19:
> I assumed zigbee interview would also work in mesh 

It does most of the times, but some chinese devices are tricky.