Koenkk/zigbee2mqtt #8424: Hive Alarm Keypad

robertalexa, 2021-08-18:
Background info: I have been looking for a zigbee alarm keypad and there doesn't seem to be a lot going on in the market, or at least nothing that can be bought easily in the UK, and not as part of a bundle with extra items.

Hive has released a new Keypad as part of their HomeShield range. So i bought one :) which I am not certain i will keep, but nonetheless I would like to give it support in z2m before (possibly) returning it.

I am writing this more as a log than a request, as mentioned, i will try to support it myself.

@peterguy04 i know you have sniffed zigbee traffic between the Hive Hub and Hive devices in the past, would you be kind to offer some guidance as to how I do this? It would be my first time trying this. I have a CC, i can flash it with the right firmware, it is more of a case of what tools do i use to see what is going on? I know you have provided screenshots before.

Thanks in advance

Product: https://www.hivehome.com/shop/smart-home-security/hive-keypad
Manual: https://bucketeer-e9a3f077-0ec6-47cc-a7da-226f8fd8c41c.s3.eu-west-1.amazonaws.com/public/img/attachments/keypadguide-66cb9cc3ea75a09afa80f8d710ccfc9d063aee17c8d15ecee486bfc593709791.pdf

data/database.db entry of the device
`{"id":5,"type":"EndDevice","ieeeAddr":"0x000d6ffffec6dcc1","nwkAddr":28324,"manufId":4456,"manufName":"LDS","powerSource":"Battery","modelId":"KEYPAD001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1025,"inClusterList":[0,1,3,32,1280,2821],"outClusterList":[25,1281],"clusters":{"genBasic":{"attributes":{"modelId":"KEYPAD001","manufacturerName":"LDS","powerSource":3,"zclVersion":3,"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"21022020","swBuildId":"1.15"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0018e32e48","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":33}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e32e48","endpointID":1},{"cluster":1280,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e32e48","endpointID":1},{"cluster":1281,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e32e48","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e32e48","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e32e48","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e32e48","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"21022020","swBuildId":"1.15","zclVersion":3,"interviewCompleted":true,"meta":{"configured":528200716},"lastSeen":1629316413769}`

robertalexa, 2021-08-18:
This is what i've got so far
[image]

This is by re-using code from the other supported keypads.

What i have noticed is that once you press a button, the device doesn't accept any other button presses, not even if you follow the sequence from the manual.

My guess is that the device is meant to receive some sort of confirmation from z2m.

Will definitely require sniffing in order to understand the functionality.

peterguy04, 2021-08-19:
Pretty sure I used this. It was so noisy on mine I had to make use of the filters once I found my device.  I can't remember if i needed to get the zigbee channel  on the hive hub to listen to, I had to contact hive customer support to get that info but can't remember if it was necessary 

https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

robertalexa, 2021-08-30:
PR created to support the device. More info there

closed 2021-08-30 by robertalexa