Koenkk/zigbee2mqtt #8425: LXEK-3 new device

emms007, 2021-08-18:
Hello,

Could you please add support for the following device:
Entry=
    {
        zigbeeModel: ['LXEK-3'],
        model: '9CZA-P470T-A1A',
        vendor: 'ADEO',
        description: 'ENKI LEXMAN E14 LED RGBW',
        extend: extend.light_onoff_brightness_colortemp_color(),
    },

Database info=
{"id":7,"type":"Router","ieeeAddr":"0x5c0272fffed50564","nwkAddr":18404,"manufId":4727,"manufName":"Adeo","powerSource":"Mains (single phase)","modelId":"LXEK-3","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64642],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LXEK-3","manufacturerName":"Adeo","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200526","swBuildId":"2.4.0"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200526","swBuildId":"2.4.0","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1629311884638}

Shop=
https://www.leroymerlin.fr/produits/decoration-eclairage/ampoule-et-led/ampoule-led/ampoule-e14/ampoule-connectee-led-spherique-e14-intensite-couleur-variables-enki-lexman-82473186.html

Thanks

sjorge, 2021-08-19:
Can you grab the color temperature info: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature

It's needed for lint checks to pass on zhc.

emms007, 2021-08-19:
Thanks, please see color temp info:
{"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}

Koenkk, 2021-08-21:
Added thanks to @sjorge 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-21 by Koenkk

antonellopasella, 2024-01-31:
@emms007 How did you managed to pair with z2m?

I do the 6 times on/off but the bulb start r-gb-w and then stay white with no join to z2m