Koenkk/zigbee2mqtt #8428: Moving map

sygys, 2021-08-19:
Could someone please let the zigbee devices on the map stop moving. Damn its annoying...! And what could possible be the purpose of devices making 360's over the screen?? I really wonder why it is made this way... You cant even select a device while its moving across the screen with the speed of light. Its like the solar system on steroids but made with the devices on your zigbee network.

sygys, 2021-08-19:
Oh now i see... its the impersonation of the bigbang https://youtu.be/amit7tUk4yo

sjorge, 2021-08-19:
You should open an issue here: https://github.com/nurikk/z2m-frontend/issues

The map has always been unsable for me too, I also have a large number of devices and it's pure chaos. I sadly still have to manually generate them using graphviz, but even then it's hell with large number of devices if I don't request the routes and filter out all other connections.

sygys, 2021-08-19:
It would help if they all stopped moving and when dragged and dropped they stay where you drop them... Couldn't be that hard to manage

Koenkk, 2021-08-19:
As mentioned here https://github.com/Koenkk/zigbee2mqtt/issues/new/choose, please create frontend issues here https://github.com/nurikk/z2m-frontend/issues

closed 2021-08-19 by Koenkk