Koenkk/zigbee2mqtt #8429: State not updating if made manually turn on/off

Drealine, 2021-08-19:
## What happened
When I turn on a light or outlet manually (not using HA), the state is not retrieve. I've to force manually to refresh the state directly into the web Z2M interface.

## What did you expect to happen
The state would be updated

## How to reproduce it (minimal and precise)
If I turn on the light from an interuptor, I have to force the status update to know that the light is on.

## Debug info
Zigbee2MQTT version: 1.21.0-4
Adapter hardware: Conbee II
Adapter firmware version: 0x26680700

danieledwardgeorgehitchcock, 2021-08-19:
This may be the default behaviour of your light. Most devices send their state when they are turned on - Zigbee2MQTT has no way of knowing otherwise what the state is

Drealine, 2021-08-19:
As discuss on Discord, it is possible to make like this ? : 

[image]

Rira1234, 2021-09-12:
I have the same problem with my dimmer switch (Model: Eco-Dim.05) (Zigbee Model: EcoDim-Zigbee 3.0).
 @Drealine: I don't fully understand your comment. Could you elaborate how this would update the state (on/off and dimmer level) of the device?

closed 2021-10-20 by github-actions[bot]

Drealine, 2021-11-13:
Yes. You can close.

rafareusch, 2022-05-18:
> Yes. You can close.
@Drealine 

Do you mind sharing what did you do to solve it? I'm having the same issue.

jzielke84, 2022-07-30:
How I hate those people like @Drealine ...