Koenkk/zigbee2mqtt #8430: Z2M 1.21 with CC2652 - devices won't finish interview

DubhAd, 2021-08-19:
### Possibly fixed in dev

## What happened

On 1.18.3 with a ZZH I had no issues pairing devices. Since upgrading to 1.21 devices not in direct reach of the coordinator won't pair at all, and devices in direct reach will pair, but interview won't complete. This included to attempt to re-include a device that had previously been paired in that location.

On a CC2531 (firmware 20190619) devices pair quickly and without issue, including the Terncy devices that wouldn't pair at all with the ZZH.

The ZZH is on a USB extension cable, on a USB 2.0 port, over 2 meters from the nearest WiFi device. I'm using channel 20 with my WiFi on channels 1 and 6. 

Devices _do_ pair - they just don't complete interview.

I've seen others report the same with not being able to pair via a router (@Tediore, @code-in-progress) on 1.21, but they could on older versions. 

**Moving to the dev release** (commit c134d4e) _mostly_ solves this. Pairing via a router still doesn't seem to work, and pairing in place is very hit and miss - much more so than before.

## What did you expect to happen

Devices to pair and complete interview.

## How to reproduce it (minimal and precise)

1. Put Z2M in pairing mode
2. Put device in pairing mode
3. Keep device awake
4. See interview never start or never complete

## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware: CC26X2R1
Adapter firmware version: 20210708 and 20201026

Current mesh has a mix of 63 devices.

sjorge, 2021-08-19:
There was a new issue during that got fixed in 1.21 as mentioned in the release notes:
https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.0

> - This release fixes an issue in the backup/restore mechanism (#7680) which prevented devices to join via a router after reflashing the firmware of your adapter. Since this was caused by an incomplete backup, users who already reflashed their coordinators earlier have to repair the routers where joninig is not working for. This only applies to adapters with a Z-Stack_3.x.0 (e.g. zzh, CC2652R launchpad, Slaesh's) firmware. Important: before reflashing your adapter, make sure to start and stop this release at least once! This ensures your backup (coordinator_backup.json) is correct.
> - New firmwares are available for Texas Instruments CC2652R, CC2652RB, CC1352P-2 based adapters: 20210708 (changelog). This i.a. enables control for the LEDs (will be on when joining is permitted) and fixes an issue of devices not being able to join when joining is only enabled on a specific router. Important: before reflashing first read the point above.

DubhAd, 2021-08-19:
> There was a new issue during that got fixed in 1.21 as mentioned in the release notes:

Right, and I'm on 1.21 and running that firmware...

sjorge, 2021-08-19:
But it doesn't magically fix the problem if you upgrade the firmware before being on 1.21. You'll need to remove/add all routers to fix it, starting near the coordinator obviously and work your way outward :(

Diddlik, 2021-08-19:
I updated mm to 1.21. Restarted it 2 times, after them updated firmware to the latest version: 20210708 but have the same problem. Pairing over routers does not work.
Is it really, I should remove all routers and repair they again?

sjorge, 2021-08-19:
I believe if the the issues already present before you updated to z2m 1.21... then yes.

CC @Koenkk

Koenkk, 2021-08-19:
@sjorge is right, you need to repair the routers you want to pair via if you ever reflashed the firmware before updating to 1.21.0. Technical details can be found here, second point: https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.0

Note that interview failing is a different kind of issue, this indicates the device paired but z2m was unable to gather all info of it. If you have an interview failing try to pair the device closer to the coordinator, if that doesn't help provide the herdsman debug log of this. 
See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Tediore, 2021-08-19:
Makes sense, thank you both. Can confirm re-pairing the routers fixed that issue on my end. I misunderstood the release notes to mean if you flash the `20210708` firmware before upgrading to 1.21.0 you'll need to re-pair all affected routers, but now I see it means if you have ever flashed _any_ version of firmware to the coordinator after creating a zigbee2mqtt network.

I think I notice a secondary issue involving re-pairing a device that used to be on the network, but I want to try it with a few different devices first to make sure it's not just that one specific device.

DubhAd, 2021-08-19:
> @sjorge is right, you need to repair the routers you want to pair via if you ever reflashed the firmware before updating to 1.21.0. Technical details can be found here, second point: https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.0

> but now I see it means if you have ever flashed _any_ version of firmware to the coordinator after creating a zigbee2mqtt network.

Damn, that makes sense now - and it's not at all how I read the release notes.

That's going to be "fun" 😂 Guess I'll be experimenting tomorrow with some spare router devices.

> Note that interview failing is a different kind of issue, this indicates the device paired but z2m was unable to gather all info of it. If you have an interview failing try to pair the device closer to the coordinator, if that doesn't help provide the herdsman debug log of this.

Thanks, the next time that happens I'll do that.

Diddlik, 2021-08-19:
> 
> 
> Makes sense, thank you both. ~Can confirm re-pairing the routers fixed that issue on my end.~ I misunderstood the release notes to mean if you flash the `20210708` firmware before upgrading to 1.21.0 you'll need to re-pair all affected routers, but now I see it means if you have ever flashed _any_ version of firmware to the coordinator after creating a zigbee2mqtt network.
> 
> I think I notice a secondary issue involving re-pairing a device that used to be on the network, but I want to try it with a few different devices first to make sure it's not just that one specific device.
> 
> Edit: I'm actually still not able to pair devices via routers even after re-pairing all of them. The first time must have been a fluke since I was in the basement relatively close to the coordinator. Attempting to pair to any router on the main floor results in nothing actually happening, and doing permit join (all) also doesn't work, likely because i'm out of the range of the coordinator. I've tried with several different devices (a couple of different contact sensors and one water leak sensor), and the results are all the same; I need to move within range of the coordinator for pairing to work. Currently running 1.21.0-dev (commit [c134d4e](https://github.com/Koenkk/zigbee2mqtt/commit/c134d4e975dee47939a132022ed8e3d5c3fc078e))

I can confirm this.

DubhAd, 2021-08-20:
> Edit: I'm actually still not able to pair devices via routers even after re-pairing all of the routers. The first time must have been a fluke since I was in the basement relatively close to the coordinator. Attempting to pair to any router on the main floor results in nothing actually happening, and doing permit join (all) also doesn't work, likely because i'm out of the range of the coordinator. I've tried with several different devices (a couple of different contact sensors and one water leak sensor), and the results are all the same; I need to move within range of the coordinator for pairing to work. Currently running 1.21.0-dev (commit [c134d4e](https://github.com/Koenkk/zigbee2mqtt/commit/c134d4e975dee47939a132022ed8e3d5c3fc078e))

Can confirm, probably. I removed and re-joined a "spare" router (Slaesh stick with router firmware). Then I tried to pair a spare door sensor through it and couldn't.

Then again, I now can't pair _anything_ ... so maybe there's a bigger issue here. I'll restart Z2M later and enable debug on the herdsman to see if I can find anything useful in either case.

DubhAd, 2021-08-20:
> Then again, I now can't pair _anything_ ... so maybe there's a bigger issue here. I'll restart Z2M later and enable debug on the herdsman to see if I can find anything useful in either case.

Well, turns out it did pair, it just didn't show up in the UI until I restarted Z2M...

Now I don't know whether it paired through the router or not, but I'll tackle that later.

Diddlik, 2021-08-20:
The only thing that worked for me was this:

do not unpair/pair each router individually, but unpair all routers and then pair them one by one.
Next, not all EndDevices were connected, some had to be paired again. This has worked so far.
And I can confirm. The devices are not visible in the frintend. Only after the restart

DubhAd, 2021-08-20:
> The only thing that worked for me was this:
> 
> do not unpair/pair each router individually, but unpair all routers and then pair them one by one.
> Next, not all EndDevices were connected, some had to be paired again. This has worked so far.

Ooof. I hope that's not necessary since that's going to be a nightmare.

djtrue, 2021-08-21:
The same faced this problem. I removed all the routers and connected the zany from near to far. But the pairing process was very difficult. The usual pairing messages did not come, and the devices only appeared after rebooting the z2m. A few days later, I discovered that the ability of the routers to mate again disappeared.  p.s. - 
sorry for the translator

Tediore, 2021-08-21:
> Well, turns out it did pair, it just didn't show up in the UI until I restarted Z2M...

I've noticed this too but only with devices that were joined to the network at one point before. If I add a new device to the network, it appears in the UI after pairing without restarting. In addition, if I unpair a device, delete its entry from the z2m configuration, and re-pair it, it behaves like a new device and appears in the UI after pairing.

Diddlik, 2021-08-24:
I deleted the devices that were close to the coordinator using the trash icon in z2m GUI. The others via force remove.
Some lamps were automatically in pairing mode, the others I had to reset according to manual or put in pairing mode.

MaZZly, 2021-08-24:
@Koenkk what is the best way to "start fresh", because it seems my network is broken due to this bug, and some of the devices have already dropped away testing @Diddlik's suggestion which worked initially, but now again everything stopped pairing... So I might as well start from scratch..

Is it enough to e.g. force_remove every device in Z2M GUI and then change the pan_id + restarting Z2M ?

djtrue, 2021-08-24:
> @Koenkk what is the best way to "start fresh", because it seems my network is broken due to this bug, and some of the devices have already dropped away testing @Diddlik's suggestion which worked [...]

I realized that this is a global problem and started everything from scratch.  Changed pan_id, channel and key.  After adding 10 nearby routers, the network stopped accepting new connections.  I gave it up.  I returned the backup and reconnected these 10 devices + about 10 more end devices (a total of 96 devices in the network) Then I reconnected all the routers.  Everything seemed to work, but after 2 days, some routers again lost the ability to connect other devices and some did not (I understood this because the area of ​​the territory is large and all routers are scattered around the site).  What is known for sure is that newly added devices (both endpoints and routers) are not displayed in the gui until the reboot. 

Sorry for the translation

Koenkk, 2021-08-24:
@MaZZly bring down all the routers in your current network, stop z2m, change the `pan_id`, also take this opportunity to switch to a better channel https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html, start z2m

z-x-f, 2021-08-25:
> I've noticed this too but only with devices that were joined to the network at one point before. If I add a new device to the network, it appears in the UI after pairing without restarting. In [...]

https://github.com/nurikk/z2m-frontend/issues/746

Tediore, 2021-08-28:
> do not unpair/pair each router individually, but unpair all routers and then pair them one by one.
Next, not all EndDevices were connected, some had to be paired again. This has worked so far.

This seems to have _mostly_ fixed the problem for me. Joining devices away from the coordinator via _permit join: all_ works seemingly without issues, and I am finally able to pair via specific routers... except for one router. That one router is the same model as several others, so who knows why it's being stubborn.

djtrue, 2021-08-29:
> @MaZZly отключите все маршрутизаторы в вашей текущей сети, остановите z2m, измените `pan_id`, также воспользуйтесь этой возможностью, чтобы переключиться на лучший канал https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html , начать z2m

The FAQ says that changing pan_id will require reconnecting all devices, not just routers.  Or is it wrong?

nickw444, 2021-09-20:
@djtrue 

> Then I reconnected all the routers. Everything seemed to work, but after 2 days, some routers again lost the ability to connect other devices and some did not (I understood this because the area of [...]

I actually think I'm having this issue as well; I raised the bug report issue here: https://github.com/Koenkk/zigbee2mqtt/issues/8803 with a tonne of sniffer traces. Have a few stubborn routers which won't allow end devices to pair with them, even after starting the network entirely from scratch (and every time I rebuild the network, different devices have the issue)