Koenkk/zigbee2mqtt #8432: Paul Neuhaus Q-remote - action_group not sent with button on / off

rodnheli, 2021-08-19:
Hi,
I bought a Paul Neuhaus Q-remote - [https://www.zigbee2mqtt.io/devices/100.462.31.html](url) - and after several hours I can't get any further - I have to say that I'm a beginner an probably missing something :-)
I want to control zigbee and wlan (tuya) lights in Home Assistant with the Q-remote and Conbee II adapter.

My problem is that the action group 1, 2, 3 are not sent in combination with the buttons ON and OFF.

[image]

**Example**:

I selet the action group 1 on the remote and then i press the botton "brightness down".
All seems to be fine:

```
{
    "action": "brightness_step_down",
    "action_group": 1,
    "action_step_size": 25,
    "action_transition_time": 0.01,
    "linkquality": 255
}
```
If i press the action group button 1 on the remote and the "on" button - action_group is not sent:

```
Message 59 received on zigbee2mqtt/QRemote_1 at 10:57:
{
    "action": "",
    "linkquality": 255
}
QoS: 0 - Retain: false
Message 58 received on zigbee2mqtt/QRemote_1 at 10:57:
{
    "action": "on",
    "linkquality": 255
}
QoS: 0 - Retain: false
Message 57 received on zigbee2mqtt/QRemote_1 at 10:57:
{
    "action": "",
    "linkquality": 255
}
QoS: 0 - Retain: false
Message 56 received on zigbee2mqtt/QRemote_1 at 10:57:
{
    "action": "on",
    "linkquality": 255
}
```

_So my problem is - i have NO idea how to power on / off each group if the action_group is not sent._

I've also seen that there should be a `scene_*.` - [https://www.zigbee2mqtt.io/devices/100.462.31.html#action-enum](url) - but didn't find out on the remote --> mqtt listener nor in HASS:

[image]

I've read through other threads / issues - downloaded some node red flows - but it seems they all send
the action_group when pressing the group button.

Using zigbee2mqtt HASS integration - current version: 1.21.0-4 .

Probably someone can lead me into the right direction.
Many thanks!

rodnheli, 2021-08-22:
I got yesterday my Trust ZYCT 202 remote - this one is sending the action_group on every button - so on / off works as expected.

```
Debug Received Zigbee message from 'Remote_ZYCT202', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 24305
Info MQTT publish: topic 'zigbee2mqtt/Remote_ZYCT202', payload '{"action":"on","action_group":24305,"linkquality":175}'
Info MQTT publish: topic 'zigbee2mqtt/Remote_ZYCT202/action', payload 'on'

Debug Received Zigbee message from 'Remote_ZYCT202', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID 24305
Info MQTT publish: topic 'zigbee2mqtt/Remote_ZYCT202', payload '{"action":"off","action_group":24305,"linkquality":247}'
Info MQTT publish: topic 'zigbee2mqtt/Remote_ZYCT202/action', payload 'off'
```

In my opinion the Neuhau q-remote is not useable with Home Assistant and mixed bulbs (zigbee, wlan, ...) because it does not send the action_group for buttons "on" and "off"...

closed 2021-09-29 by github-actions[bot]

DavidFFerreira, 2021-11-11:
Hi, did you got this working?

rodnheli, 2021-11-11:
> Hi, did you got this working?

No, i gave up playing some hours with this :-)
Bought a namron / sunricher remote -> SR-ZG9001K12-DIM-Z4 which works well with action groups - did realize than the actions with node red...

[image]

DavidFFerreira, 2021-11-24:
@rodnheli how did you pair. I've got this controller but i cannot even pair with z2m edge

pepsinio, 2025-01-06:
I have the same issue. It used to work just fine. Since upgraded to Z2M 2.0 and latest Zstack action group is gone. Any ideas how to fix this?