Koenkk/zigbee2mqtt #8433: Problems with interview of previously paired devices

yoprogramo, 2021-08-19:
## What happened
I start a brand new homeassistant instance from scratch, installed zigbee2mqtt and none of the previously working devices end the interview

## What did you expect to happen

Interview completed and device paired

## How to reproduce it (minimal and precise)

* Have a homeassistant with working zigbee2mqtt devices paired. Get rid of the SD (destroyed and no backup)
* Install a new homeassistant with a new zigbee2mqtt instance from scratch (same hardware)
* Try to pair a device that previously worked

## Debug info
Zigbee2MQTT version: 1.21.0-4
Adapter hardware: CC2531
Adapter firmware version: 

Last debug after trying to pair a device:

```
2021-08-19T10:19:02.614Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-08-19T10:19:02.615Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-08-19T10:19:02.616Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-19T10:19:12.618Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '29159' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2021-08-19T10:19:12.619Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 29159
2021-08-19T10:19:12.620Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":29159,"options":0,"radius":30}
2021-08-19T10:19:12.622Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,231,113,0,30,236]
2021-08-19T10:19:12.629Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2021-08-19T10:19:12.629Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2021-08-19T10:19:12.630Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2021-08-19T10:19:12.630Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2021-08-19T10:19:12.631Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-19T10:19:15.633Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":29159,"nwkaddrofinterest":29159}
2021-08-19T10:19:15.634Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,231,113,231,113,35]
2021-08-19T10:19:15.643Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-08-19T10:19:15.644Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-08-19T10:19:15.645Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-08-19T10:19:15.645Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-08-19T10:19:15.646Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-19T10:19:25.648Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0003ccf6aa', attempt 3
2021-08-19T10:19:25.649Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":29159,"nwkaddrofinterest":29159}
2021-08-19T10:19:25.650Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,231,113,231,113,35]
2021-08-19T10:19:25.660Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-08-19T10:19:25.661Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-08-19T10:19:25.661Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-08-19T10:19:25.662Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-08-19T10:19:25.663Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-19T10:19:35.665Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '29159' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2021-08-19T10:19:35.665Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 29159
2021-08-19T10:19:35.667Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":29159,"options":0,"radius":30}
2021-08-19T10:19:35.668Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,231,113,0,30,236]
2021-08-19T10:19:35.675Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2021-08-19T10:19:35.675Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2021-08-19T10:19:35.676Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2021-08-19T10:19:35.680Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2021-08-19T10:19:35.681Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-19T10:19:38.683Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":29159,"nwkaddrofinterest":29159}
2021-08-19T10:19:38.684Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,231,113,231,113,35]
2021-08-19T10:19:38.695Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-08-19T10:19:38.696Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-08-19T10:19:38.696Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-08-19T10:19:38.697Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-08-19T10:19:38.698Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-19T10:19:48.706Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0003ccf6aa', attempt 4
2021-08-19T10:19:48.707Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":29159,"nwkaddrofinterest":29159}
2021-08-19T10:19:48.708Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,231,113,231,113,35]
2021-08-19T10:19:48.719Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-08-19T10:19:48.719Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-08-19T10:19:48.720Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
[... 25 lines cut ...]
2021-08-19T10:20:11.770Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-19T10:20:21.773Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '29159' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2021-08-19T10:20:21.773Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 29159
2021-08-19T10:20:21.775Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":29159,"options":0,"radius":30}
2021-08-19T10:20:21.777Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,231,113,0,30,236]
2021-08-19T10:20:21.788Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2021-08-19T10:20:21.788Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2021-08-19T10:20:21.789Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2021-08-19T10:20:21.791Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2021-08-19T10:20:21.792Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-19T10:20:24.794Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":29159,"nwkaddrofinterest":29159}
2021-08-19T10:20:24.796Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,231,113,231,113,35]
2021-08-19T10:20:24.806Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-08-19T10:20:24.806Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-08-19T10:20:24.807Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-08-19T10:20:24.808Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-08-19T10:20:24.808Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-19T10:20:34.810Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0003ccf6aa', attempt 6
2021-08-19T10:20:35.570Z zigbee-herdsman:controller:device:log Interview - failed for device '0x00158d0003ccf6aa' with error 'Error: Interview failed because can not get node descriptor ('0x00158d0003ccf6aa')
    at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:459:19)
    at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:358:13)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:534:17)'
2021-08-19T10:20:35.573Z zigbee-herdsman:controller:error Interview failed for '0x00158d0003ccf6aa with error 'Error: Interview failed because can not get node descriptor ('0x00158d0003ccf6aa')'
Zigbee2MQTT:error 2021-08-19 12:20:35: Failed to interview '0x00158d0003ccf6aa', device has not successfully been paired
```

mcsikerull, 2021-08-19:
I have the same problem. Waiting for a decision

vdkeybus, 2021-08-20:
Did you also reset the devices ?

yoprogramo, 2021-08-20:
> Did you also reset the devices ?

Yes, several times, this causes the start interview...

DubhAd, 2021-08-20:
Some of us have been seeing [similar issues](https://github.com/Koenkk/zigbee2mqtt/issues/8430).

Koenkk, 2021-08-21:
If you want to start a new network on the CC2531 make sure to use a different pan_id (https://www.zigbee2mqtt.io/information/configuration.html) otherwise it will conflict with the already existing network AND make sure all your routers are off power (to make sure the old network is down)

yoprogramo, 2021-08-21:
> If you want to start a new network on the CC2531 make sure to use a different pan_id (https://www.zigbee2mqtt.io/information/configuration.html) otherwise it will conflict with the already existing network AND make sure all your routers are off power (to make sure the old network is down)

Yes, changing the pan_id makes the possible to pair the devices (I still have some issues with one of them but I thing the pairing problem is solved doing this).

closed 2021-08-21 by yoprogramo