Koenkk/zigbee2mqtt #8449: IKEA E1757 Blind - Add State Tracking

krisoshau, 2021-08-20:
## Is your feature request related to a problem? Please describe
Because there is no native (Open/Close) for this device like a lock device. I have to manually fire scripts to open and close the blinds. I also cannot track the state remotely because of this.

## Describe the solution you'd like
I'd like to be able to see the state (Open, Closed) in the entities. The blind also publishes how far up/down it is in intervals of 2 via MQTT as it goes,

https://i.imgur.com/zSDcJGz.png (Entities Panel)
https://pastebin.com/cbUa22ZY (MQTT Log from opening/closing)

## Describe alternatives you've considered
Tracking each individual part manually with scripts etc but it's alot of borking around for something that could be done at a higher level

sjorge, 2021-08-20:
There is no attribute in the zigbee cluster that tracks a lock/unlock state like for locks.

I just key all my scripting of the position attribute, where 100 = closed and < 100 is open.

krisoshau, 2021-08-22:
Sounds like this is not solvable with this device.

closed 2021-08-22 by krisoshau