Koenkk/zigbee2mqtt #8451: Ikuu \ Tuya TS0011 - No status update with manual operation

activex, 2021-08-20:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/8427

<div type='discussions-op-text'>

<sup>Originally posted by **activex** August 19, 2021</sup>
Hi all,

I've got a Ikuu \ Tuya TS0011 single gang wired wall switch [Link](https://www.ikuu.com.au/product/single-switch/), registration on Z2M worked fine after the 3rd attempt and I can control the switch through the dashboard and manually.

When I press the switch button manually, the status is not being updated in the dashboard. The status is off, even though all lights are on.

Is that normal behaviour? How can I investigate further or fix this issue.

For context, I use a Conbee II on a raspberry pi 4, running Z2M 1.21 and homebridge to get my zigbee devices into HomeKit.

Thanks in advance!</div>

Switch: Ikuu Single Switch - Identified as Tuya TS0011 [Product Link](https://www.ikuu.com.au/product/single-switch/)

I've also removed the switch from Z2M and added it to the Ikuu Zigbee Hub using the Ikuu app. In the Ikuu app, any changes are reflected immediately.

activex, 2021-08-20:
Updated the issue formatting as it was copying all from the discussion.

ddemidov, 2021-09-04:
Same issue with Lonsonho 1gang switch (qs-zigbee-s05-l) identified as Tuya TS0011 by Z2M. Manual state change is not reflected in Z2M, refresh button under Exposes/State works.

closed 2021-10-12 by github-actions[bot]