Koenkk/zigbee2mqtt #8452: Crashes once a day since 1.21.0

illivanillie, 2021-08-20:
#in

## What happened
I have massive problems with zigbee2mqtt.
zigbee2mqtt once a day fails and is unable to start again. „Failed to start zigbee-herdsman“
I can only overcome this by stopping HA and pull the plug for a few seconds. Then zigbee is able to start again. Seems like it is an adapter problem, but just reconnecting the adapter won’t work.

## What did you expect to happen

zigbee2mqtt to keep working or at least be able to start again without physical interaction, because it is in a remote spot.

## How to reproduce it (minimal and precise)

was not able to determine any specific cause. Randomly, about once a day.
The problem last for about 2 weeks. So it could correlate with 1.21.0 or/and with the new adapter firmware

## Debug info
Zigbee2MQTT version: latest
Adapter hardware: CC26X2R1
Adapter firmware version: latest, reflashed several times.

Koenkk, 2021-08-20:
Please provide the herdsman debug logging from starting z2m till the crash.

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

illivanillie, 2021-08-20:
Should I update to 1.21.0-4 before the log?

Koenkk, 2021-08-21:
That doesn't matter.

illivanillie, 2021-08-23:
By chance I was able to copy this. I got a notification that a few lights were unavailable for a few times (=zigbee2mqtt was not available). Maybe you can read something. There was no crash so far again.
Tbh I am still trying to figure out, how and where exactly the herdsman logs are to find. Could you help me out?

`2021-08-23T15:45:00.354Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - simpleDescReq - {"status":0}
2021-08-23T15:45:00.355Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14]
2021-08-23T15:45:00.356Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [69,132,0,0,0,0,0,8,6,9,1,5,0,0,0,0,204]
2021-08-23T15:45:00.356Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,69,132,0,0,0,0,0,8,6,9,1,5,0,0,0,0,204]
2021-08-23T15:45:00.356Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 2 - 5 - 132 - [0,0,0,0,0,8,6,9,1,5,0,0,0,0] - 204
2021-08-23T15:45:00.356Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":0,"status":0,"nwkaddr":0,"len":8,"endpoint":6,"profileid":265,"deviceid":5,"deviceversion":0,"numinclusters":0,"inclusterlist":[],"numoutclusters":0,"outclusterlist":[]}
2021-08-23T15:45:00.357Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-23T15:45:00.357Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":0,"nwkaddrofinterest":0,"endpoint":5}
2021-08-23T15:45:00.358Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,4,0,0,0,0,5,33]
2021-08-23T15:45:00.363Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,4,0,96,254,14]
2021-08-23T15:45:00.363Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,4,0,96,254,14]
2021-08-23T15:45:00.363Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 4 - [0] - 96
2021-08-23T15:45:00.363Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - simpleDescReq - {"status":0}
2021-08-23T15:45:00.363Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14]
2021-08-23T15:45:00.364Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [69,132,0,0,0,0,0,8,5,8,1,5,0,0,0,0,206]
2021-08-23T15:45:00.365Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,69,132,0,0,0,0,0,8,5,8,1,5,0,0,0,0,206]
2021-08-23T15:45:00.365Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 2 - 5 - 132 - [0,0,0,0,0,8,5,8,1,5,0,0,0,0] - 206
2021-08-23T15:45:00.365Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":0,"status":0,"nwkaddr":0,"len":8,"endpoint":5,"profileid":264,"deviceid":5,"deviceversion":0,"numinclusters":0,"inclusterlist":[],"numoutclusters":0,"outclusterlist":[]}
2021-08-23T15:45:00.366Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-23T15:45:00.366Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":0,"nwkaddrofinterest":0,"endpoint":4}
2021-08-23T15:45:00.367Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,4,0,0,0,0,4,32]
2021-08-23T15:45:00.371Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,4,0,96,254,14]
2021-08-23T15:45:00.372Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,4,0,96,254,14]
2021-08-23T15:45:00.372Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 4 - [0] - 96
2021-08-23T15:45:00.372Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - simpleDescReq - {"status":0}
2021-08-23T15:45:00.372Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14]
2021-08-23T15:45:00.373Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [69,132,0,0,0,0,0,8,4,7,1,5,0,0,0,0,192]
2021-08-23T15:45:00.374Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,69,132,0,0,0,0,0,8,4,7,1,5,0,0,0,0,192]
2021-08-23T15:45:00.374Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 2 - 5 - 132 - [0,0,0,0,0,8,4,7,1,5,0,0,0,0] - 192
2021-08-23T15:45:00.374Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":0,"status":0,"nwkaddr":0,"len":8,"endpoint":4,"profileid":263,"deviceid":5,"deviceversion":0,"numinclusters":0,"inclusterlist":[],"numoutclusters":0,"outclusterlist":[]}
2021-08-23T15:45:00.374Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-23T15:45:00.375Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":0,"nwkaddrofinterest":0,"endpoint":3}
2021-08-23T15:45:00.375Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,4,0,0,0,0,3,39]
2021-08-23T15:45:00.381Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,4,0,96,254,14]
2021-08-23T15:45:00.381Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,4,0,96,254,14]
2021-08-23T15:45:00.381Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 4 - [0] - 96
2021-08-23T15:45:00.381Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - simpleDescReq - {"status":0}
2021-08-23T15:45:00.381Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14]
2021-08-23T15:45:00.382Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [69,132,0,0,0,0,0,8,3,4,1,5,0,0,0,0,196]
2021-08-23T15:45:00.383Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,69,132,0,0,0,0,0,8,3,4,1,5,0,0,0,0,196]
2021-08-23T15:45:00.383Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 2 - 5 - 132 - [0,0,0,0,0,8,3,4,1,5,0,0,0,0] - 196
[... cut ...]
2021-08-23T15:45:00.842Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82,254,3]
2021-08-23T15:45:00.843Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82,254,3]
2021-08-23T15:45:00.843Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82
2021-08-23T15:45:00.843Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2021-08-23T15:45:00.843Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3] `

Koenkk, 2021-08-23:
I don't see any error in that log. Use `docker logs zigbee2mqtt > log.txt 2>&1` to grab all logs, next time please post the logs on pastebin.com

jmyoung, 2021-08-28:
I'm having the same issue.  CC2530+CC2591, using the firmware from 20190608 (haven't been bothered to update it yet), and using the latest zigbee2mqtt Docker image, the coordinator works for _exactly_ 24 hours, and I mean _exactly_ - to the minute.  After that it just doesn't do anything else until I power cycle the adapter (which involves shutting down the rpi it's on, of course).

No useful logs unfortunately :/

goliath888, 2021-08-28:
I was gonna open an issue but I think I have the same problem. I am using docker image 1.21.0 with the CC2530.

As @jmyoung mentioned, exactly 24 hours after start the container, logs has no more update of any device. If I try to turn on any light, I get this message:

> error 2021-08-28 21:09:43: Publish 'set' 'state' to 'hallway_front' failed: 'Error: Command 0x588e81fffec9ca45/1 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

If I try to reboot the container, It won't be able to do it with this error message: 

> Using '/app/data' as data directory
> Zigbee2MQTT:info  2021-08-28 21:25:50: Logging to console and directory: '/app/data/log/2021-08-28.21-25-50' filename: log.txt
> Zigbee2MQTT:info  2021-08-28 21:25:50: Starting Zigbee2MQTT version 1.21.0 (commit #70891ee)
> Zigbee2MQTT:info  2021-08-28 21:25:50: Starting zigbee-herdsman (0.13.124)
> Zigbee2MQTT:error 2021-08-28 21:26:10: Error while starting zigbee-herdsman
> Zigbee2MQTT:error 2021-08-28 21:26:10: Failed to start zigbee
> Zigbee2MQTT:error 2021-08-28 21:26:10: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
> Zigbee2MQTT:error 2021-08-28 21:26:10: Exiting...
> Zigbee2MQTT:error 2021-08-28 21:26:10: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
>     at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
>     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
>     at Zigbee.start (/app/lib/zigbee.js:67:27)
>     at Controller.start (/app/lib/controller.js:106:27)
>     at start (/app/index.js:94:5)
> Using '/app/data' as data directory

I have to unplug the CC2530 and plug it again. Then everything works fine...for another 24 hours. 

I go back to 1.20 and it goes perfect for days with no issues. 

Is there anything i can help with?

gionni12, 2021-08-28:
hi.idem

jmyoung, 2021-08-30:
> I have to unplug the CC2530 and plug it again. Then everything works fine...for another 24 hours.
> 
> I go back to 1.20 and it goes perfect for days with no issues.

In this vein, I rolled back to my old verison of zigbee2mqtt, which was pulled 15 months ago (yeah), and everything works fine again.  This problem seems to be specific to most recent versions.  Since @goliath888 has had success with 1.20, I'll try pulling 1.20.0 and see how that goes.

gionni12, 2021-08-30:
good morning..I don't know the method and I don't have a backup of my home assistant. can someone tell me how to go back?  Thank you

YassineBenayad, 2021-08-30:
Hello everybody,
I have the same issue since I migrated to z2m 1.21. I'm using a CC2530+CC2591 connected to a vmware host using a Silicon Labs CP2102 USB to UART Bridge Controller, the same setup works perfectly well if I down grade to version 1.18-1 or 1.17.0-1 (Home Assistant Add-on) from some pervious spanshots of my VM.

here is how the error look like:

> [10:53:19] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.21.0 start
> node index.js
Zigbee2MQTT:error 2021-08-30 10:53:48: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-08-30 10:53:48: Failed to start zigbee
Zigbee2MQTT:error 2021-08-30 10:53:48: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-08-30 10:53:48: Exiting...
Zigbee2MQTT:error 2021-08-30 10:53:48: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.js:67:27)
    at Controller.start (/app/lib/controller.js:106:27)
    at start (/app/index.js:94:5)

Westcott1, 2021-08-30:
I'm also using  1.21-dev and CC2530+CC2591.
After **exactly** 24 hours all devices steadily go offline with "failed to ping xxx"
It all works OK for another 24 hours if I stop z2m, un- and re-plug the stick, then restart z2m.

illivanillie, 2021-08-31:
For me it was ok for a few days. Just happened again and had a quick look.
See logs [here](https://pastebin.com/jG1LFVAK)

illivanillie, 2021-09-03:
For me the issue is definitely solved (for the moment), if I restart the HA host.
Any ideas on how to dig deeper into this?

ascanio65, 2021-09-03:
hi.for me desolution was a downgrade at version 1.20

Koenkk, 2021-09-03:
Can you try again with the latest-dev? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

jmyoung, 2021-09-05:
Swapped over now, we shall see in 24 hours.  I'll report back with findings :)

Westcott1, 2021-09-05:
I'm sorry, it didn't fix it for me.

jmyoung, 2021-09-06:
Didn't fix it for me.  Stopped responding after 24 hours again.

Koenkk, 2021-09-06:
@Westcott1 @jmyoung to debug this further I will need the herdsman debug logging

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Westcott1, 2021-09-06:
Staretd with - DEBUG=zigbee-herdsman* npm start
Now wait 24 hours...

jmyoung, 2021-09-07:
Mine too, just set it up, I see herdsman logging output in docker logs.  So now we wait.

Westcott1, 2021-09-07:
All devices dutifully went offline after 24 hours exactly.
Hopefully I captured the right bits in the attached file.
[herdsman.log](https://github.com/Koenkk/zigbee2mqtt/files/7120531/herdsman.log)

Koenkk, 2021-09-07:
@Westcott1 the log you provided is very short, please provide the log starting from the point where you start z2m until it fails.

Westcott1, 2021-09-07:
Sorry I don't know how to do that!
Since it can't be redirected into a file, the PuTTY window doesn't have a large enough buffer to hold the entire 24 hours of data.
I saw nothing untoward up to the point where everything went offline.
If you like, I can capture the start also.

colin19889, 2021-09-07:
I'm having the exact same 24-hour problem with Hassio + RPi4 + CC2530+2591 + CP2102. Going to update to 1.21.1-1 and I'll capture the logs and post them in about 24-hours.

jmyoung, 2021-09-08:
Got a full captured log for you, from start to finish.  There's...  a lot of stuff in it, I have quite a few temperature sensors...

[herdsman.log.gz](https://github.com/Koenkk/zigbee2mqtt/files/7125039/herdsman.log.gz)

What immediately strikes me is the last message in the log is;

```
[36mzigbee2mqtt_1  |[0m 2021-09-08T00:43:34.004Z zigbee-herdsman:controller:log Wrote coordinator backup to '/app/data/coordinator_backup.json'
```

Presumably that task runs after exactly 24 hours of uptime?

Westcott1, 2021-09-08:
Hi Jmyoung,
How did you manage to capture the full-size log?
I saw that coordinator message, but guessed that it was part of the shutdown process.

jmyoung, 2021-09-08:
I'm using Docker Compose, so I just did `docker-compose logs > foo.txt`.

That coordinator message appeared _before_ I shut down the container, and notably it also happened exactly (well, to within a minute) 24 hours after z2m was started.

Koenkk, 2021-09-08:
@jmyoung do I understand correctly that z2m just freezes completely after the line below?

```
[36mzigbee2mqtt_1  |[0m 2021-09-08T00:43:34.004Z zigbee-herdsman:controller:log Wrote coordinator backup to '/app/data/coordinator_backup.json'
``` 

What is logged when you try to control a bulb after this? Would be good if you could also include the z2m logging itself, you can do this with: (replace the `CONTAINER_ID` ofc)

```
docker logs CONTAINER_ID > log.txt 2>&1
```

Westcott1, 2021-09-08:
Mine ended with this -

```
Zigbee2MQTT:error 2021-09-07 10:29:25: Failed to ping 'Dimmer_1'
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":53,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,245,2,11,0,0]}} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,53,0,30,6,0,25,245,2,11,0,0,213] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":513} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,1,2,51] +6s
  zigbee-herdsman:controller:log Creating coordinator backup +57s
  zigbee-herdsman:adapter:zStack:startup:backup creating backup +24s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +6s
  zigbee-herdsman:controller:endpoint Read 0x00124b001c467b6c/11 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +1m
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b001c467b6c:61298/11 (0,0,1) +38s
^C^C  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":513} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,1,2,51] +6s
Zigbee2MQTT:error 2021-09-07 10:29:49: Failed to stop Zigbee2MQTT
```

jmyoung, 2021-09-08:
> @jmyoung do I understand correctly that z2m just freezes completely after the line below?

In my case, yes - but I also didn't just leave it sitting there for a long time.  I configured an automation in HomeAssistant to yell at me via Pushover when it hasn't heard from anything in 10 minutes, so I got to it pretty quick when it failed.

> What is logged when you try to control a bulb after this? 

I don't have any controllable devices set up at the moment.  However, after the failure, I get no more updates from anything - no temperature sensors or anything else.

> Would be good if you could also include the z2m logging itself, you can do this with: (replace the `CONTAINER_ID` ofc)

I already had the logging set to 'warn' for z2m itself as well as herdsman.  I'll pull it down in the morning and restart it all with everything on debug - I expect there will be a _lot_ of output, and I'll have to wait the 24 hours.

jmyoung, 2021-09-11:
Full log attached, z2m was in debug logging mode, herdsman logging enabled.  As before, the last line written by Herdsman to the log was about the coordinator backup, followed by a few state updates from z2m, and that's all she wrote.

```
zigbee2mqtt_1  | 2021-09-11T06:23:59.410Z zigbee-herdsman:adapter:zStack:startup:backup fetched adapter network security material table
zigbee2mqtt_1  | 2021-09-11T06:23:59.436Z zigbee-herdsman:controller:log Wrote coordinator backup to '/app/data/coordinator_backup.json'
zigbee2mqtt_1  | Zigbee2MQTT:debug 2021-09-11 15:57:26: Saving state to file /app/data/state.json
zigbee2mqtt_1  | Zigbee2MQTT:debug 2021-09-11 16:02:26: Saving state to file /app/data/state.json
zigbee2mqtt_1  | Zigbee2MQTT:debug 2021-09-11 16:07:26: Saving state to file /app/data/state.json
```

[full-log.txt.gz](https://github.com/Koenkk/zigbee2mqtt/files/7147337/full-log.txt.gz)

Koenkk, 2021-09-11:
@jmyoung could you try and see if the issue still happens when using the source routing firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Westcott1, 2021-09-11:
I switched to a zzh CC2652R coordinator (from a CC2630) 2 days ago, and it has run OK since then.

jmyoung, 2021-09-13:
> @jmyoung could you try and see if the issue still happens when using the source routing firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Sorry, but I can't change the firmware on my coordinator to that.  I have 15 devices, however I'm not able to get most of my devices within range of at least one separate router (I do have two other routers, but they typically only wind up with 1-2 devices on them due to their location).

I'll just stick with 1.20.0 instead.  That works.

igogold, 2021-09-13:
I have the same issue with 24 hours lifetime of working z2m and after that I should reset adapter to make it working again. But I have a way to send reset command to the adapter without rebooting my device (I'm unable to replug adapter). I'm using [CC2530+CC2592 module](https://wirenboard.com/wiki/WBE2R-R-ZIGBEE_ZigBee_Extension_Module) for Russian Wiren Board smart home controller based on [Ebyte module](https://www.ebyte.com/en/product-view-news.aspx?id=123). The firmware is a latest one (non source route), 21 devices total (2 routers).

Is it possible to disable (via config) creation of coordinator backup at all? It also need more time to stop z2m service with a backup (about 90 seconds now), not only hang coordinator after 24 hours.

colin19889, 2021-09-13:
> But I have a way to send reset command to the adapter without rebooting my device (I'm unable to replug adapter).
> 
@igogold - How are you resetting the adapter without rebooting? I send a reboot command at 0300 everyday to avoid the problem but resetting the adapter would be better.

igogold, 2021-09-14:
@colin19889 At developer support board I found info adapter module has reset line connected to gpio of wiren board controller. And it's possible to change state of gpio using linux interface via sysfs. There is [a post](https://support.wirenboard.com/t/perestal-zapuskatsya-zigbee2mqtt-error-srsp-sys-ping-after-6000ms/7770/6) (in Russian). I think a presence of such reset line depends on a way the module was created and connected to main device.

I'm resetting an adapter every time before start zigbee2mqtt.service as it hang just after stopping the service (after creating a coordinator backup?). So, to restart z2m I need reset adapter anyway.

colin19889, 2021-10-03:
Has there been any progress made on this issue? I'm still scheduling a 3am restart but the reboot occasionally causes issues with my battery operated devices meaning I have to manually reset them.

Koenkk, 2021-10-03:
Not yet, @castorw is checking this

castorw, 2021-10-17:
Okay so let's give a quick update. I have tried reproducing this issue on CC2531 running Z-Stack 1.2. So far without success. From what I read in this post this seems to affect CC2530. CC2531 should be almost the same as CC2530 except it has a built-in USB controller. However there might be other differences in the IC design.

Unfortunately I have no CC2530 device on hand so I am still trying to reproduce this on CC2531. Upgraded one of my prod nwks to affected z2m version 1.21.2.

**Please try updating firmware!**
This issue resembled signs of **memory leak**, so the only sensible advice I am able to give right now: please try updating your radios to the latest firmware available. Please let me know if this helps.

igogold, 2021-10-19:
> Zip contains one for the CC2530 and CC2530+CC2591

Thank you! But can you also build CC2530+CC2592 default firmware so I can try it too as my CC2530 adapter comes with CC2592 amp?

Koenkk, 2021-10-21:
@colin19889 does this problem disappear when switching back to the old fw?

Koenkk, 2021-10-29:
Could you guys test the following firmwares? https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_Home_1.2/bin/source_routing (among other changes, it also includes a fix for this problem). Please let me know if it works OK then I can release these fws officially.

igogold, 2021-11-01:
Is only "source routing" firmware have this bug fixed? Or in the "default" it must be fixed too? I flash "default" develop firmware ("revision":20211030) and it seems the bug is still here. On this firmware the stopping of z2m service lasts about 90 sec, there are no devices information in the created coordinator_backup.json and adapter must be reset/power cycle to work again. The same as it was on last stable "default" 20201127 firmware version.
Was firmware from https://github.com/Koenkk/zigbee2mqtt/issues/8452#issuecomment-946473353 a "source routing" one? Maybe it was fixed for me because I moved from default to source routing, not because of fixes?

Koenkk, 2021-11-01:
@igogold could you check if the source routing fw in that zip fixes it?

igogold, 2021-11-01:
Maybe I'll send you a debug logs when service stopping and creating backup? I don't know was this bug on the stable source routing firmware to know it was fixed. Also it's not easy for me to reflash the adapter, it's a whole quest with connection/disconnection of my controller with embedded zigbee adapter.

igogold, 2021-11-01:
```
Zigbee2MQTT:info  2021-11-01 12:40:17: Stopping zigbee-herdsman...
  zigbee-herdsman:controller:log Disable joining +669ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":2,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,3,2,11,0,0]}} +1m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,2,0,30,6,0,25,3,2,11,0,0,20] +1m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103,254,3,68,128,0,242,2,55] +677ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103,254,3,68,128,0,242,2,55] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +6ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +1m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,2,55] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,2] - 55 +4ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":2} +828ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +9ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":0,"tcsignificance":0} +334ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,0,0,26] +329ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1] +231ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [101,54,0,82] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +5ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +273ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +10ms
  zigbee-herdsman:controller:log Creating coordinator backup +406ms
  zigbee-herdsman:adapter:zStack:startup:backup creating backup +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +82ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +85ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240] +46ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240 +5ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +328ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,0,2,6,3,86,101,52,1,2,0,0,0,0,108] +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,0,2,6,3,86,101,52,1,2,0,0,0,0,108] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,0,2,6,3,86,101,52,1,2,0,0,0,0] - 108 +7ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20211030} +96ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +7ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - getExtAddr - {} +71ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,4,37] +70ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,8,97,4,19,140,49,31,0,75,18,0,133] +26ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,8,97,4,19,140,49,31,0,75,18,0,133] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 8 - 3 - 1 - 4 - [19,140,49,31,0,75,18,0] - 133 +3ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - getExtAddr - {"extaddress":"0x00124b001f318c13"} +44ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +11ms
  zigbee-herdsman:adapter:zStack:startup:backup fetched adapter ieee address +120ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":33} +50ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,33,0,17] +50ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2] +38ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [97,19,110,0,30] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,110,0,30] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [110,0] - 30 +5ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":110} +62ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +10ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvReadExt - {"id":33,"offset":0} +62ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,33,28,33,0,0,0,24] +63ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,112] +51ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,112] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [97,28,0,110,225,5,2,17,20,17,0,20,0,0,0,1,5,1,143,7,0,2,5,30,0,0,11,0,0,0,0,0,0,0,0,0,0,98,26,8,0,8,0,0,15,15,5,0,1,0,0,0,1,0,0,0,0,19,140,49,31,0,75,18,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0] +9ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":117} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,117,0,69] +6s
  zigbee-herdsman:adapter:zStack:startup:backup fetched adapter network security material table +18s
  zigbee-herdsman:controller:log Wrote coordinator backup to '/mnt/data/root/zigbee2mqtt/data/coordinator_backup.json' +2m
  zigbee-herdsman:adapter:zStack:znp:log Port closed +3m
Zigbee2MQTT:info  2021-11-01 12:41:48: Stopped zigbee-herdsman
Zigbee2MQTT:info  2021-11-01 12:41:48: Stopped Zigbee2MQTT
```

Koenkk, 2021-11-01:
I see the bug is indeed appearing in your logs.

Can somebody else check if the firmware from https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_Home_1.2/bin/default works?

snakuzzo, 2021-11-01:
Hi @Koenkk I tried [1.2 default](https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20211030.zip) and it seems ok.

[HERE](https://pastebin.com/SVg1sFxj) start and stop log (using latest dev image with docker + docker-compose)

igogold, 2021-11-02:
Is it possible to add an option to disable creation of coordinator backups (on stop and on 24h basis)? It's not a fix but it will work as workaround of this issue.

castorw, 2021-11-02:
> Is it possible to add an option to disable creation of coordinator backups (on stop and on 24h basis)? It's not a fix but it will work as workaround of this issue. 

The problem in the CC2530 firmware is caused by small buffer sizes for ZNP protocol. I would strongly suggest updating since this may cause other issues in time.

igogold, 2021-11-02:
@castorw sorry, but what update you suggest? Update CC2530 to more powerful adapter or only update firmware of this one? I'm already on latest develop fw, just not source route. Or the source route variant for CC2530 is the only stable firmware? But how much directly connected devices can be paired with the coordinator on this fw? In my config with about 7-8 direct devices on source route fw I lose one device and was unable to reconnect it back before fw reflash. So zigbee network with additional routers is the only solution for stable work of CC2530? Up to 40 devices as written in fw repo readme?

Koenkk, 2021-11-02:
@igogold i'm checking together with @snakuzzo to fix the default firmware for CC2530

castorw, 2021-11-03:
@igogold I was thinking of FW upgrade. Sorry I didn't read that the default firmware hasn't been fixed yet. Let's wait for the fix.

igogold, 2021-11-03:
Thank you @Koenkk ! Sorry but I will be able to test it only on Monday :(

igogold, 2021-11-08:
@Koenkk I've flashed your fixed firmware and now my adapter no hanging anymore on service stop. Also the stop time just one second, not 90+ as before. I'll check how it will work 24h later (when it create daily coordinator config backup). Thank you!

Koenkk, 2021-11-08:
@igogold thanks, please let me know tomorrow, if it is ok I will publish the firmwares on https://github.com/koenkk/z-stack-firmware

igogold, 2021-11-09:
Well, for now z2m works more than 24 hours and successfully updated coordinator_backup.json 20 minutes ago.

There are no devices in the file but adapter doesn't hang while doing backup. I think, CC2350 fw version works good.

Thank you @Koenkk !

[10 more replies]

closed 2021-11-17 by Koenkk