Koenkk/zigbee2mqtt #8453: Prepare the new state_class : total_increasing for Home Assistant

McGiverGim, 2021-08-20:
## Is your feature request related to a problem? Please describe
The `energy` sensors have now a `last_reset` attribute that is a const date, it is not refreshed when the device is really reset, so this can produce bad measurements. Home Assistant have detected this problem too, and have deprecated for the next version this way of doing things. 

Instead, they have added a new `state_class : total_increasing` that does the reset automatically. More info here: https://developers.home-assistant.io/blog/2021/08/16/state_class_total/

## Describe the solution you'd like
Change the `last_reset` and `state_class` value for energy sensors. I suppose is very similar to this PR: https://github.com/Koenkk/zigbee2mqtt/pull/8187/files

But I don't know how to deal with the backward compatibility (if needed).

## Describe alternatives you've considered
None.

## Additional context
None.

Koenkk, 2021-08-20:
@frenck will `total_increasing` land in HA 2021.9.0?

frenck, 2021-08-21:
Yes, but the current solution is backward compatible.
So, it will generate a warning.

Considering HA & Z2M releases are out of sync, I would skip a release in Z2M personally (which is why I didn't open a PR yet).