Koenkk/zigbee2mqtt #8454: Upgrade to Slae.sh-stick - Failed to determine if path is valid / Error while opening serialport

smathev, 2021-08-20:
## What happened
Upgraded from a 2531-stick to a slae.sh-stick. Flashed the firmware (2652RB* from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin), and now I can't get zigbee2mqtt to start in my docker. Since it's in a docker it's fairly hard to troubleshoot it, but I've followed every thread that had issues similar to mine.

## What did you expect to happen
I expected Zigbee2Mqtt to start. See below for error-log on zigbee2mqtt-docker start and my docker-compose file.

## How to reproduce it (minimal and precise)
Hard to say - I don't know what's going wrong :(

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2652 - Slae.sh
Adapter firmware version: CC2652RB_coordinator_20210708.bin

Docker-Compose file:

```
    zigbee2mqtt:
        image: koenkk/zigbee2mqtt
        container_name: zigbee2mqtt
        privileged: true
        volumes:
            - /run/udev:/run/udev:ro
            - ("relevant_homedir")/zigbee2mqtt/:/app/data:rw
        devices:
                #- /dev/ttyUSB0:/dev/ttyACM0
                #- "/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_4B_7B-if00-port0:/dev/ttyACM0"
            - /dev/ttyUSB.slaesh_stick:/dev/ttyUSB.slaesh_stick
        ports:
            - "8989:8989"
        environment:
            - TZ=Europe/Copenhagen
            - PUID=1000
            - PGID=1000
            - DEBUG=zigbee-herdsman*
```
I've made a udev-rule:
`SUBSYSTEM=="tty", ATTRS{idVendor}=="0451", ATTRS{idProduct}=="16a8", ATTRS{serial}=="00_12_4B_00_22_98_85_D5", SYMLINK="ttyUSB.slaesh_stick", OWNER="core"`
Changing the things that didn't line up with the slae.sh sticks infomartion, IE. idVendor, idProduct, and serial.

I've tested that the user starting the docker-container has write-access to the port:
```
test -w /dev/ttyUSB.slaesh_stick && echo success || echo failure
success
```

Docker error log, when starting Zigbee2Mqtt:
```
Using '/app/data' as data directory
Zigbee2MQTT:debug 2021-08-20 12:22:33: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2021-08-20 12:22:33: Logging to console and directory: '/app/data/log/2021-08-20.12-22-32' filename: log.txt
Zigbee2MQTT:debug 2021-08-20 12:22:33: Removing old log directory '/app/data/log/2021-08-20.12-03-23'
Zigbee2MQTT:info  2021-08-20 12:22:33: Starting Zigbee2MQTT version 1.21.0 (commit #70891ee
)
Zigbee2MQTT:info  2021-08-20 12:22:33: Starting zigbee-herdsman (0.13.124)
Zigbee2MQTT:debug 2021-08-20 12:22:33: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB.slaesh_stick"}}'
2021-08-20T10:22:33.208Z zigbee-herdsman:adapter:zStack:znp:error Failed to determine if path is valid: 'Error: ENOENT: no such file or directory, lstat '/dev/ttyUSB.slaesh_stick''
2021-08-20T10:22:33.419Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB.slaesh_stick"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2021-08-20T10:22:33.419Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB.slaesh_stick and {"baudRate":115200,"rtscts":false,"autoOpen":false}
Zigbee2MQTT:error 2021-08-20 12:22:33: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-08-20 12:22:33: Failed to start zigbee
Zigbee2MQTT:error 2021-08-20 12:22:33: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-08-20 12:22:33: Exiting...
Zigbee2MQTT:error 2021-08-20 12:22:33: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB.slaesh_stick'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
```

As far as I can tell, the issue is that docker isn't passing the USB-stick through to the container, which is weird for me. 

I've tried passing the /dev/ttyUSB.slaesh_stick through as a bunch of different devices - both /dev/ttyACM0, /dev/ttyUSB0 and I've also tried (before the udev-rule) to go by /dev/serial/by-id - but nothing lead to anything but the errors you see.

Could you please help me troubleshoot this?

DubhAd, 2021-08-20:
After you edited the compose file, did you rebuild the container:

```bash
docker-compose up -d zigbee2mqtt
```

smathev, 2021-08-20:
Hi. 

Thanks for getting back to me, everytime I've tried changing anything in my docker_compose file I run a bash script with the following lines:
```
            /usr/local/bin/docker-compose -f $file1 pull
            sleep 0.5
            /usr/local/bin/docker-compose -f $file1 down
            sleep 10
            /usr/local/bin/docker-compose -f $file1 rm
             sleep 0.2
             /usr/local/bin/docker-compose -f $file1 build --no-cache
```
where $file1 is the docker_compose file in which everything related to zigbee2mqtt, home-assistant, mqtt-broker, etc is located. IE. I rebuild the entire thing from scratch only keeping my config-folders intact.

smathev, 2021-08-31:
Any news, ideas or input to this?

Thanks in advance.

smathev, 2021-09-02:
Yeah, so leaving this comment for others to reference. The issue is fixed.

Turns out "priviliged" and devices-mapping doesn't mix in a docker compose-file.

So, after removing 'priviliged: true' from my docker-compose file the device was mapped correctly and everything is up and running.

closed 2021-09-02 by smathev