Koenkk/zigbee2mqtt #8457: Develco/Frient smoke detector - false (?) battery values?

rindlerblabla, 2021-08-20:
## What happened
I've got 6 pretty new (like two months old) Frient smoke detectors. Three of them now says "Battery 42%" (not 41, 38 or something else, the rest 100%. It also differs through the day, between 42% and 100%. Seems like there's some issue with the battery reporting?

[image]

## Debug info
zigbee2mqtt_version 1.21.0-dev commit: f9451ff
coordinator_type zStack3x0 (zzh)
coordinator_revision 20210120

(@theStigh)

sjorge, 2021-08-20:
I have a few of these and this is ‘normal’ behavior, they don’t support reporting batteryPercentageRemaining, so the battery % is based on voltage.

Which isn’t as accurate, voltage also fluctuates for me based on temperature so battery hops around a lot, usually twice uses day/night.

TheStigh, 2021-08-21:
@rindlerblabla This is not a Z2M issue, it would be what the device reports.

Are you using a Develco or Frient product, and what is the firmware version? Why do you refer to Develco/Frient?

rindlerblabla, 2021-08-21:
Okey! Then maybe it is better to "listen" to the binary `battery_low` or is that also based on a calculation on the voltage?

Those devices are SMSZB-120, branded Frient, but that's only the name on the box, it's the same company/products. Don't know what firmware, only the build date (20210531 01:34) is shown in the setting list.

sjorge, 2021-08-21:
I only added support for the develop air quality sensor, but for that one the technical document mentions the voltage cutoff or use for setting battery_low. Not sure for the other devices but I assume they are similar as all of them only support reporting of batteryVoltage

https://github.com/Koenkk/zigbee-herdsman-converters/blob/ba43a6efd4ef24d35df59d2d8803c7b6b27dd30a/converters/fromZigbee.js#L1563

rindlerblabla, 2021-08-21:
To me it looks like it's part of the IAS cluster and that the device report battery_low itself when the battery is below 2.8 VDC.

Anyhow, all my devices report `batteryVoltage":29` or `batteryVoltage":30`. 29 is calculated to 42% and 30 to 100%.

`debug 2021-08-21 21:43:59: Received Zigbee message from 'Brandvarnare trapphuset', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryAlarmState":0,"batteryVoltage":29}' from endpoint 35 with groupID 0`

sjorge, 2021-08-21:
Right the fire/smoke detector uses IAS, the air quality sensor does not, so it’s probably not keyed on batteryVoltage.

~ sjorge

rindlerblabla, 2021-08-21:
According to the manual, https://www.develcoproducts.com/media/1742/smszb-120-technical-manual-smoke-alarm.pdf

> Note: The attribute “BatteryVoltage” is measuring the battery voltage, in units of 100mV.  

So 29 = 2900 mV (~97% battery, not 42%).

sjorge, 2021-08-21:
@Koenkk i wonder if I used the wrong battery template then?

TheStigh, 2021-08-22:
> Okey! Then maybe it is better to "listen" to the binary `battery_low` or is that also based on a calculation on the voltage?
>
> Those devices are SMSZB-120, branded Frient, but that's only the name on [...]

[image]

rindlerblabla, 2021-08-22:
I am on firmware 4.0.2 as well.

Koenkk, 2021-08-22:
Maybe a different `voltageToPercentage` can be used, e.g. `3V_2500` 

https://github.com/Koenkk/zigbee-herdsman-converters/blob/46cdb1c63dc5b4a317ba84e5c7ec20e8adb2d5bd/devices/develco.js#L148

sjorge, 2021-08-23:
That definitely seems to give a reading for my Air Quality sensors, they now show closer to what I would expect given the battery age, I also swapped in a near deplete set and it showed < 20% and battery_low=true. I didn't have the correct batteries in a near depleted state to check the smoke and heat detectors.

rindlerblabla, 2021-08-24:
@sjorge Are your heat and smoke detectors showing 100% or less? Saw that you didn't have any near depleted batteries bug have you tried with ``3V_2500`` for the batteries you already have in place?

sjorge, 2021-08-24:
The one I have easy access too is at a 100% so it made no difference when testing.

~ sjorge

closed 2021-09-12 by Koenkk