Koenkk/zigbee2mqtt #8458: Bitron Video AV2010/25 - energy consumption is not exposed

doctaste, 2021-08-21:
I'm using a couple of these AV2010/25 power plugs and they're working fine so far.
They also provide the summarized energy consumption but i cannot use that in HA
as the sensor is not available under exposes.

- Zigbee2MQTT 1.21.0-dev ( Edge)
- HA core-2021.8.8
- supervisor-2021.06.8
- Home Assistant OS 6.2
- CC2531 ( Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}')

[image]

[image]

{
    "energy": 41.7,
    "linkquality": 57,
    "power": 0.1,
    "state": "ON"
}

nfo  2021-08-21 09:16:10: MQTT publish: topic 'zigbee2mqtt/Bitron Video AV2010/25-3', payload '{"energy":3.5,"linkquality":76,"power":0,"state":"OFF"}'

I would like to read the energy, can you add this ?

Thanks !

Koenkk, 2021-08-21:
Did an attempt to add it, can you check if it works in the latest-dev?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

doctaste, 2021-08-21:
Thanks very much.

Installed latest dev branch.

_Starting Zigbee2MQTT version 1.21.0-dev (commit #f9451ff)_

Now it does show "Null W" under power although there's load attached and energy doesn't show up neither. 

[image]

Under State "energy" is not listed anymore:

{
    "linkquality": 149,
    "power": null,
    "state": "ON"
}

Koenkk, 2021-08-22:
`f9451ff` is an old version, you need to uninstall and install again to update to the latest edge.

doctaste, 2021-08-22:
When i deinstall Edge and reinstall it from add-on store ( https://github.com/zigbee2mqtt/hassio-zigbee2mqtt) i get #f9451ff.

info  2021-08-22 20:53:31: Starting Zigbee2MQTT version 1.21.0-dev (commit #f9451ff)

How can i get a newer version ?

Koenkk, 2021-08-22:
@ciotlosm I noticed the pipeline failed because the dev branch was deleted, https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/blob/f82c683409d47609a9fc9189d0a9a32c7ee681cb/pipeline/edge-build-pipeline.yml#L5 is refering to so it's used as the default branch.

@doctaste up-to-date image will be available in 30 mins from now.

doctaste, 2021-08-22:
@Koenkk 

Thanks very much !

The zigbee2mqtt edge updated now to d07f1da and the Bitron AV2010/25 is now showing the energy consumption.

info  2021-08-23 00:17:47: Starting Zigbee2MQTT version 1.21.0-dev (commit #d07f1da)

[image]

[image]

Thanks again, issue can be closed.

br,

doctaste

closed 2021-08-23 by Koenkk

s7431952, 2021-09-15:
@Koenkk the latest release (1.21.1) broke Bitron AV2010/25 for me.
When I revert back to the previous release (1.21.0) everything works as expected.
I am getting as error:
Error Failed to configure 'Server', attempt 3 (Error: ConfigureReporting 0x00124b0007300c55/1 seMetering([{"attribute":"currentSummReceived","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:627:22) at Object.currentSummReceived (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:112:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/bitron.js:55:13) at Configure.configure (/app/lib/extension/configure.js:137:13))