Koenkk/zigbee2mqtt #8460: High latency / delay on lights sometimes. Very often 2 seconds.

kevincaradant, 2021-08-21:
Hi

I use Z2M and Home Assistant and mosquitto since one year now.
Each instance are running with Docker container.
Everything worked well until now.

Often but not always, I have some significatly high latency  between 1 and 3 seconds, between the click and the answer of the light for example.
I can see the signal of the button in Z2M but the light if there is only one or (one of 3 lights) if it's a group, can take time before to switch of color or brightness. 
I don't understand why :(
 
This is an example ( look the time, there is 2 seconds of delay ). But I can click 6 times without issue and suddently I got the issue and sometimes, it's on the first click that I have the issue, it's pretty random, I don't get it :
[image]

I tried to update everything that I can HA / Mosquitto and Z2M but it's exactly the same thing

Thank you ina dvance for your help

Regards

Adapter hardware: CC1352P-2

Version Zigbee2MQTT
1.21.0-dev commit: cb47efe
Type de Coordinateur
zStack3x0
Révision du Coordinateur
20201130
Version Interface
0.5.25

Version HA
Version | core-2021.8.8
Type d'installation | Home Assistant Container

kevincaradant, 2021-08-22:
Hi

I tried to reproduce today the issue because sometimes it works well and sometimes it doesn't.
In this moment i have delay on all of my lights.

I press a brightness and color from remote and I got it 8 seconds after ... something like that

```
2021-08-22T16:02:00.255Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2021-08-22 18:02:00: Publishing 'set' 'color' to 'gledopto_bedroom'
2021-08-22T16:02:00.257Z zigbee-herdsman:controller:endpoint Command 0x00124b00210f6f77/11 lightingColorCtrl.moveToColor({"transtime":1.6500000000000001,"colorx":21299,"colory":12452}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-08-22T16:02:00.257Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b00210f6f77:62461/11 (0,0,1)
2021-08-22T16:02:00.257Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62461,"destendpoint":11,"srcendpoint":1,"clusterid":768,"transid":147,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[1,89,7,51,83,164,48,1,0]}}
2021-08-22T16:02:00.258Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,253,243,11,1,0,3,147,0,30,9,1,89,7,51,83,164,48,1,0,31]
2021-08-22T16:02:00.274Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-08-22T16:02:00.274Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-08-22T16:02:00.274Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-08-22T16:02:00.274Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-08-22T16:02:00.274Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-22T16:02:00.278Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,147,85]
2021-08-22T16:02:00.278Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,147,85]
2021-08-22T16:02:00.278Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,147] - 85
2021-08-22T16:02:00.278Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":147}
2021-08-22T16:02:00.278Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-22T16:02:01.075Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,6,0]
2021-08-22T16:02:01.075Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0]
2021-08-22T16:02:01.075Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [9,58,2,1,0,66,0,143]
2021-08-22T16:02:01.075Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,9,58,2,1,0,66,0,143]
2021-08-22T16:02:01.075Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [120,8,0,0,4,1,122,253]
2021-08-22T16:02:01.075Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,9,58,2,1,0,66,0,143,120,8,0,0,4,1,122,253]
2021-08-22T16:02:01.076Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,186,182,27,195]
2021-08-22T16:02:01.076Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,9,58,2,1,0,66,0,143,120,8,0,0,4,1,122,253,0,186,182,27,195]
2021-08-22T16:02:01.076Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,9,58,2,1,0,66,0,143,120,8,0,0,4,1,122,253,0,186,182,27] - 195
2021-08-22T16:02:01.076Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":14857,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":66,"securityuse":0,"timestamp":555151,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,122,253,0]}}
2021-08-22T16:02:01.077Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":6,"data":{"type":"Buffer","data":[1,122,253,0]},"address":14857,"endpoint":2,"linkquality":66,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2021-08-22 18:02:01: Received Zigbee message from 'remote_bed_bedroom', type 'raw', cluster 'genOnOff', data '{"data":[1,122,253,0],"type":"Buffer"}' from endpoint 2 with groupID 0
2021-08-22T16:02:01.078Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-22T16:02:01.224Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,6,0]
2021-08-22T16:02:01.224Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0]
2021-08-22T16:02:01.224Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [9,58,3,1,0,66,0,13]
2021-08-22T16:02:01.225Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,9,58,3,1,0,66,0,13]
2021-08-22T16:02:01.225Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [157,8,0,0,4,1,123,253]
2021-08-22T16:02:01.225Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,9,58,3,1,0,66,0,13,157,8,0,0,4,1,123,253]
2021-08-22T16:02:01.226Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,186,182,27,164]
2021-08-22T16:02:01.226Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,9,58,3,1,0,66,0,13,157,8,0,0,4,1,123,253,0,186,182,27,164]
2021-08-22T16:02:01.226Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,9,58,3,1,0,66,0,13,157,8,0,0,4,1,123,253,0,186,182,27] - 164
[... cut ...]
2021-08-22T16:02:13.497Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,151,81]
2021-08-22T16:02:13.497Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,151,81]
2021-08-22T16:02:13.497Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,151] - 81
2021-08-22T16:02:13.497Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":151}
2021-08-22T16:02:13.497Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

kevincaradant, 2021-08-22:
Here a last example. 

```
Zigbee2MQTT:info  2021-08-22 18:10:54: MQTT publish: topic 'zigbee2mqtt/Z2M_all_lights', payload '{"brightness":90,"color":{"x":0.325,"y":0.19},"color_mode":"xy","color_temp":35,"state":"ON"}'
Zigbee2MQTT:info  2021-08-22 18:10:54: MQTT publish: topic 'zigbee2mqtt/gledopto_bedroom', payload '{"brightness":90,"color":{"x":0.325,"y":0.19},"color_mode":"xy","color_temp":35,"linkquality":96,"state":"ON"}'
2021-08-22T16:10:55.734Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,6,0]
2021-08-22T16:10:55.734Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0]
2021-08-22T16:10:55.735Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [9,58,3,1,0,39,0,115]
2021-08-22T16:10:55.735Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,9,58,3,1,0,39,0,115]
2021-08-22T16:10:55.735Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [92,38,0,0,4,1,126,253]
2021-08-22T16:10:55.735Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,9,58,3,1,0,39,0,115,92,38,0,0,4,1,126,253]
2021-08-22T16:10:55.736Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,113,148,27,188]
2021-08-22T16:10:55.736Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,9,58,3,1,0,39,0,115,92,38,0,0,4,1,126,253,0,113,148,27,188]
2021-08-22T16:10:55.736Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,9,58,3,1,0,39,0,115,92,38,0,0,4,1,126,253,0,113,148,27] - 188
2021-08-22T16:10:55.736Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":14857,"srcendpoint":3,"dstendpoint":1,"wasbroadcast":0,"linkquality":39,"securityuse":0,"timestamp":2514035,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,126,253,0]}}
2021-08-22T16:10:55.737Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":6,"data":{"type":"Buffer","data":[1,126,253,0]},"address":14857,"endpoint":3,"linkquality":39,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2021-08-22 18:10:55: Received Zigbee message from 'remote_bed_bedroom', type 'raw', cluster 'genOnOff', data '{"data":[1,126,253,0],"type":"Buffer"}' from endpoint 3 with groupID 0
2021-08-22T16:10:55.738Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-22T16:10:55.739Z zigbee-herdsman:controller:endpoint DefaultResponse 0x60a423fffe708538/3 6(253, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-08-22T16:10:55.739Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x60a423fffe708538:14857/3 (0,0,1)
2021-08-22T16:10:55.739Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":14857,"destendpoint":3,"srcendpoint":1,"clusterid":6,"transid":160,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,126,11,253,0]}}
2021-08-22T16:10:55.739Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,9,58,3,1,6,0,160,0,30,5,24,126,11,253,0,54]
Zigbee2MQTT:info  2021-08-22 18:10:55: MQTT publish: topic 'zigbee2mqtt/remote_bed_bedroom', payload '{"action":"3_single","battery":null,"linkquality":39}'
Zigbee2MQTT:info  2021-08-22 18:10:55: MQTT publish: topic 'zigbee2mqtt/remote_bed_bedroom', payload '{"action":"","battery":null,"linkquality":39}'
Zigbee2MQTT:info  2021-08-22 18:10:55: MQTT publish: topic 'zigbee2mqtt/remote_bed_bedroom/action', payload '3_single'
Zigbee2MQTT:debug 2021-08-22 18:10:55: Received MQTT message on 'zigbee2mqtt/gledopto_bedroom/set' with data '{"state": "ON", "transition": 0.05, "brightness": 90, "color_temp": 375}'
Zigbee2MQTT:debug 2021-08-22 18:10:55: Publishing 'set' 'brightness' to 'gledopto_bedroom'
2021-08-22T16:10:55.759Z zigbee-herdsman:controller:endpoint Command 0x00124b00210f6f77/11 genLevelCtrl.moveToLevelWithOnOff({"level":90,"transtime":1.6500000000000001}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-08-22T16:10:55.760Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-08-22T16:10:55.760Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-08-22T16:10:55.760Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-08-22T16:10:55.760Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-08-22T16:10:55.760Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-22T16:10:55.807Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,225,1,160,135]
2021-08-22T16:10:55.807Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,225,1,160,135]
2021-08-22T16:10:55.807Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [225,1,160] - 135
2021-08-22T16:10:55.807Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":225,"endpoint":1,"transid":160}
2021-08-22T16:10:55.807Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-22T16:10:55.807Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x60a423fffe708538:14857,225,0)
[... cut ...]
2021-08-22T16:10:59.463Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,47,68,129,0,0,2,7,34,92,1,1,0,78,0,171]
2021-08-22T16:10:59.463Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [231,41,0,0,27,24,227,10]
2021-08-22T16:10:59.464Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,47,68,129,0,0,2,7,34,92,1,1,0,78,0,171,231,41,0,0,27,24,227,10]
2021-08-22T16:10:59.464Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,4,42,90,0,0,0,0]
```

I don' why, I dont know if it's revelant but it just to notice it

Regards

Koenkk, 2021-08-22:
Can you make sure you are using the latest firmware first? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

It can be that there is some interference from WiFi (233 error means no acknowledgement received which). https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html

kevincaradant, 2021-09-17:
Mhh I still having some issues but less other wihtout know why.

So for the moment I will close this issue, but if the issue happens again with a relevant use case, I will open again this issue 

Regards

closed 2021-09-17 by kevincaradant