Koenkk/zigbee2mqtt #8461: battery level at null for Moes_ZS-SR4-2169 / TS0044

kevincaradant, 2021-08-21:
Hi

I bought this remote.
https://zigbee.blakadder.com/Moes_ZS-SR4-2169.html

Everything works well but it's recognized as this one:
https://www.zigbee2mqtt.io/devices/TS0044.html

The issue is just I don't have the battery level.

I just have NULL as value

[image]

How to fix this issue ? What can I do ? 

Thank you 

Regards

snp88, 2021-08-24:
look similar to my problem 
https://github.com/Koenkk/zigbee2mqtt/issues/8500

kevincaradant, 2021-08-24:
Hi

Using the question of the #8500 issue. 

this is my database entry:

```
{"id":89,"type":"EndDevice","ieeeAddr":"0x60a423fffea63d99","nwkAddr":59157,"manufId":4098,"manufName":"_TZ3000_abci1hiu","powerSource":"Battery","modelId":"TS0044","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0044","manufacturerName":"_TZ3000_abci1hiu","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1629819374595}
```

manufacturer: TS3000_abci1hi

kevincaradant, 2021-09-17:
Hi

I'm still blocked with this issue and I don't know how to help, thank you

Regards

kevincaradant, 2021-11-03:
Don't close the issue

Thank you

MJeka, 2021-11-03:
Hi.
I have the same problem. Battery state is null

_TZ3000_ufhtxr59

{"id":56,"type":"EndDevice","ieeeAddr":"0x60a423fffe8affcd","nwkAddr":47829,"manufId":4098,"manufName":"_TZ3000_ufhtxr59","powerSource":"Battery","modelId":"TS0044","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0044","manufacturerName":"_TZ3000_ufhtxr59","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1635887453343}

kobuki, 2021-11-12:
Same device, same problem here, too.

I tried reconfiguring it, but I only get an error when trying to do so:

```
MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x1234123412341234"},"error":"Device '0x1234123412341234' cannot be configured","status":"error","transaction":"x2oyk-14"}'
```

kevincaradant, 2021-12-22:
@Koenkk, could you please reopen it?

Do your hunk you can fix it?

Thank you

Koenkk, 2021-12-23:
@kevincaradant did an attempt to fix it:
- update to the latest dev branch in a few hours from now (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)
- Configure device via frontend (press yellow refresh button), right before doing that make sure to wakeup the device by pressing a button on it.

kevincaradant, 2021-12-29:
Perfect thank you it's fixed

Regards

closed 2021-12-29 by kevincaradant