Koenkk/zigbee2mqtt #8467: CC1352-P2 / USB port get stuck after HA soft reboot, only unplug/replug resolves the issue

TheStigh, 2021-08-21:
Hi,

Intel NUC's, Home Assistant OS, latest version HA 2021.8.8, latest OS 6.2, latest Z2M 1.21.0, latest fw for coordinator.

On every 2, 3, 4 or 5th soft restart of Home Assistant (running Z2M addon) the USB port seem to get stuck. I suspect it has to do with re-initialization of the USB port. I can un- and re-plug the adapter, even while Z2M are running, and it starts up without any more issues. It happen on all my installs, both release and dev, **over several versions so I assume it is not related to Z2M itself,** more how the hardware works.

Is it possible to add a hard reset function of the USB port as part of Z2M? Even a button to do this in the UI?

Have a look at this: https://github.com/jkulesza/usbreset

hughb8on, 2021-08-29:
Does the CC1352-P2  show up as a /dev/ttyACMn device, probably ttyACM0?, and if so, have you seen that the modemmanager service, installed by default in Ubuntu, can cause problems with such devices? 

ModemManager incorrectly assumes anything with a ttyACM name must be a modem, to it tries to grab it, preventing any other process accessing it.

It's unlikely you actually have modem, so you don't need that service and can disable/uninstall it.  It's what I did to cure the problem you describe affecting my CC26x2R1.

It's intermittent because it depends on timing, and it doesn't affect the majority of users because they tend to use Raspberry Pis and they don't have modemmanager installed by default.

TheStigh, 2021-08-29:
> Does the CC1352-P2 show up as a /dev/ttyACMn device, probably ttyACM0?, and if so, have you seen that the modemmanager service, installed by default in Ubuntu, can cause problems with such devices?

Hi @hughb8on  
I do run it as ACM0, but I also run Home Assistant OS, so there are no ModemManager installed.
But I'll definitely change to use device by-id as you suggest to try.

Anyway - getting a feature to reset any USB port would still be a nice to have feature. The code exist but currently we cannot run it within Home Assistant OS as we don't have sufficient access to the file system from terminal.

closed 2021-10-07 by github-actions[bot]