Koenkk/zigbee2mqtt #8468: Adding a new device Aqara QBKG38LM 1 key without zero line.

dumbodumbos, 2021-08-21:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/8463

<div type='discussions-op-text'>

<sup>Originally posted by **dumbodumbos** August 21, 2021</sup>
Hi. Aqara QBKG39LM, QBKG40LM have been added to the stable version for a long time. And they forgot about QBKG38LM. Please add it to the next release of the stable version.</div>

danieledwardgeorgehitchcock, 2021-08-24:
It looks like this has already been added to the codebase and should be available (if not already) in the next stable release. See Koenkk/zigbee2mqtt#7625

dumbodumbos, 2021-08-24:
OK thanks.  made a convector, everything worked.  but still please remember to add in the next issue.

Koenkk, 2021-08-24:
Already added indeed, available in the dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-24 by Koenkk