Koenkk/zigbee2mqtt #8469: Failed to start zigbee

AlexUaDuo, 2021-08-21:
#Bug Report

What happened
Good afternoon.
After launching the Zigbee2mqtt add-on in the web interface, the inscription (502: Bad Gateway)
Here are the logs: https://pastebin.com/m1PgvUy0
Help me please.

What did you expect to happen
Correct launch zigbee2mqtt

How to reproduce it (minimal and precise)
add-on config: https://pastebin.com/T9VnFpAW

Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2531
Adapter firmware version: I took already stitched by the Chinese

lux73, 2021-08-22:
Zigbee2MQTT:error 2021-08-21 17:15:40: Error: Error while opening serialport **'Error: Error: No such file or directory, cannot open /dev/ttyACM0'**

your Stick cannot be found by the System...

AlexUaDuo, 2021-08-22:
> 
> 
> Zigbee2MQTT:error 2021-08-21 17:15:40: Error: Error while opening serialport **'Error: Error: No such file or directory, cannot open /dev/ttyACM0'**
> 
> your Stick cannot be found by the System...

I made some progress, now I know the right port.
Only a new problem has appeared, something has no permissions there. 

Zigbee2MQTT:error 2021-08-22 19:44:55: Error: Error while opening serialport 'Error: Error: Operation not permitted, cannot open /dev/bus/usb/001/003'

lux73, 2021-08-22:
you must add your User which is running zigbee2mqtt to the "dialout" Group

sudo adduser YOUR_USER dialout

AlexUaDuo, 2021-08-22:
> 
> 
> you must add your User which is running zigbee2mqtt to the "dialout" Group
> 
> sudo adduser YOUR_USER dialout
> 
> /edit
> 
> btw your USB Path seems weird - please check: https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

[image]
[image]
[image]
Doesn't work anyway

lux73, 2021-08-22:
Docker...  havent used Docker before

could imagine that you Pass-Through your USB Device to your Docker Container: https://forums.unraid.net/topic/47595-how-do-i-pass-through-a-usb-device-to-a-docker-container/

if that not Help i dont know, sorry...

AlexUaDuo, 2021-08-22:
Unfortunately, it didn’t help.
But thanks anyway for the help.
Maybe someone else can help me.

closed 2021-10-08 by github-actions[bot]