Koenkk/zigbee2mqtt #8481: Iris contact and temperature sensor Model # IL06_1

D4M4EVER, 2021-08-22:
## What happened

Added Iris Contact and temperature sensor, model # IL06_1 power source icon is showing ? question mark.
Ref supported device page https://www.zigbee2mqtt.io/devices/IL06_1.html#iris-il06_1

## What did you expect to happen

Power Icon to show a battery icon with current power level 

## How to reproduce it (minimal and precise)

Add Iris Contact and temperature sensor.

## Debug info
Zigbee2MQTT version:    1.21.0 commit: 70891ee 
Adapter hardware:    ZZH!
Adapter firmware version:    CC2652R_coordinator_20210708

D4M4EVER, 2021-08-22:
Here is a the device info from my database.db file:

`{"id":19,"type":"EndDevice","ieeeAddr":"REDACTED","nwkAddr":29682,
"manufId":4667,"manufName":"iMagic by GreatStar",
"powerSource":"Unknown","modelId":"1116-S",
"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,
"devId":1026,"inClusterList":[0,1,3,32,1026,1280,2821,64513,64514],"outClusterList":[3,25],
"clusters":{"genBasic":{"attributes":{"modelId":"1116-S","manufacturerName":"iMagic by GreatStar",
"powerSource":0,"zclVersion":2,"appVersion":0,"stackVersion":0,"hwVersion":0}},
"ssIasZone":{"attributes":{"iasCieAddr":"REDACTED","zoneState":1}},"genPowerCfg":{"attributes":{"batteryVoltage":29}},
"msTemperatureMeasurement":{"attributes":{"measuredValue":2128}}},"binds":[{"cluster":1026,"type":"endpoint",
"deviceIeeeAddress":"REDACTED","endpointID":1},{"cluster":1,"type":"endpoint",
"deviceIeeeAddress":"REDACTED","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,
"maxRepIntval":3600,"repChange":100},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],
"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"zclVersion":2,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1629658580219}
`

nurikk, 2021-08-23:
Hi!
Can you attach complete state.json file from `Settings -> Tools -> Download state` ?

D4M4EVER, 2021-08-23:
The complete state.json file doesn't list the device, it shows all my other devices but the Iris contact sensors.

D4M4EVER, 2021-08-23:
Heres all the info for it from the state.json file, when i looked it via my cell phone the file was to big to list everything. I removed all over devices to reduce the file size.

```
`            "friendly_name": "CS-01",
            "ieee_address": "REDACTED",
            "interview_completed": true,
            "interviewing": false,
            "manufacturer": "iMagic by GreatStar",
            "model_id": "1116-S",
            "network_address": 29682,
            "power_source": "Unknown",
            "supported": true,
            "type": "EndDevice"
        },
        "0xd0cf5efffeb9fe8b": {
            "definition": {
                "description": "Contact and temperature sensor",
                "exposes": [
                    {
                        "access": 1,
                        "description": "Indicates if the contact is closed (= true) or open (= false)",
                        "name": "contact",
                        "property": "contact",
                        "type": "binary",
                        "value_off": true,
                        "value_on": false
                    },
                    {
                        "access": 1,
                        "description": "Indicates if the battery of this device is almost empty",
                        "name": "battery_low",
                        "property": "battery_low",
                        "type": "binary",
                        "value_off": false,
                        "value_on": true
                    },
                    {
                        "access": 1,
                        "description": "Indicates whether the device is tampered",
                        "name": "tamper",
                        "property": "tamper",
                        "type": "binary",
                        "value_off": false,
                        "value_on": true
                    },
                    {
                        "access": 1,
                        "description": "Measured temperature value",
                        "name": "temperature",
                        "property": "temperature",
                        "type": "numeric",
                        "unit": "°C"
                    },
                    {
                        "access": 1,
                        "description": "Remaining battery in %",
                        "name": "battery",
                        "property": "battery",
                        "type": "numeric",
                        "unit": "%",
                        "value_max": 100,
                        "value_min": 0
                    },
                    {
                        "access": 1,
                        "description": "Link quality (signal strength)",
                        "name": "linkquality",
                        "property": "linkquality",
                        "type": "numeric",
                        "unit": "lqi",
                        "value_max": 255,
                        "value_min": 0
                    }
                ],
                "model": "IL06_1",
                "supports_ota": false,
                "vendor": "Iris"
            },
            "endpoints": {
                "1": {
                    "bindings": [
                        {
                            "cluster": "msTemperatureMeasurement",
                            "target": {
                                "endpoint": 1,
                                "ieee_address": "REDACTED",
                                "type": "endpoint"
                            }
                        },
                        {
                            "cluster": "genPowerCfg",
                            "target": {
                                "endpoint": 1,
                                "ieee_address": "REDACTED",
                                "type": "endpoint"
                            }
                        }
                    ],
                    "clusters": {
                        "input": [
                            "genBasic",
                            "genPowerCfg",
                            "genIdentify",
                            "genPollCtrl",
                            "msTemperatureMeasurement",
                            "ssIasZone",
                            "haDiagnostic",
                            "manuSpecificUbisysDimmerSetup",
                            "manuSpecificSamsungAccelerometer"
                        ],
                        "output": [
                            "genIdentify",
                            "genOta"
                        ]
                    },
                    "configured_reportings": [
                        {
                            "attribute": "measuredValue",
                            "cluster": "msTemperatureMeasurement",
                            "maximum_report_interval": 3600,
                            "minimum_report_interval": 10,
                            "reportable_change": 100
                        },
                        {
                            "attribute": "batteryVoltage",
                            "cluster": "genPowerCfg",
                            "maximum_report_interval": 62000,
                            "minimum_report_interval": 3600,
                            "reportable_change": 0
                        }
                    ]
                }
            },
            
            
        "CS-01": {
            "battery": 42,
            "battery_low": false,
            "contact": true,
            "linkquality": 63,
            "tamper": false,
            "temperature": 20.1,
            "voltage": 2900
        },

````

closed 2021-09-30 by github-actions[bot]