Koenkk/zigbee2mqtt #8488: OSRAM AC01353010G SMART+ Motion Sensor: do not expose battery_low and tamper

mabi, 2021-08-23:
## Is your feature request related to a problem? Please describe
battery_low and tamper are not truly functional, as it seems that hardware support is missing, see also #507

## Describe the solution you'd like
I suggest removing them from the report.

## Describe alternatives you've considered
I tried removing them myself from the exposes section in osram.js of  zigbee-herdsman-converters, but they keep showing.

## Additional context
Please suggest how to do it correctly.
Thank you.

Koenkk, 2021-08-23:
Done, note that after updating you need to remove it from `state.json` manually.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-23 by Koenkk

mabi, 2021-08-23:
Thank you, removing from `state.json` did the trick. I will remove my local modifications and pull yours.

Koenkk, 2021-08-23:
Yes, otherwise they will be published again