Koenkk/zigbee2mqtt #8490: HA: no _linkquality entities for new devices

Masterz69, 2021-08-23:
## What happened
Just added new devices into network: MCCGQ11LM, RTCGQ11LM & WSDCGQ11LM.
Almost all looks fine. Except - no _linkquality entities created, after hours.

State for all devices contains link quality, like:
```{
    "battery": 100,
    "humidity": 61.03,
    "last_seen": "2021-08-23T13:55:41+03:00",
    "linkquality": 117,
    "pressure": 1020,
    "temperature": 17.31,
    "voltage": 3005
}```
Same info in MQTT topics as well. 
HA topics have this property:
![image](https://user-images.githubusercontent.com/15225823/130438934-808835fa-b7e6-4c25-9315-8640281cc610.png)

## What did you expect to happen
Same WSDCGQ11LM devices previously (on Z2M 1.18 version) added - all have _linkquality entities.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.21.0-4
Adapter hardware: zStack3x0 (CC2652)
Adapter firmware version: 20200921

![image](https://user-images.githubusercontent.com/15225823/130438509-ce0d901c-ffa6-4b1e-8ec5-c2b90c4ac3ea.png)

tabnul, 2021-08-23:
Same with WS-EUK03 (Aqara EU wall switch with neutral)

Koenkk, 2021-08-23:
These entities are disabled by default so that is expected, see https://github.com/Koenkk/zigbee2mqtt/pull/7684 for the rationale.

closed 2021-08-23 by Koenkk

Masterz69, 2021-08-23:
Thank You very much.