Koenkk/zigbee2mqtt #8496: Devices not responing, migrating from CC2538 to CC252RB

wizu, 2021-08-23:
I would like to migrate from CC2538 to Slae.sh CC2652RB because of some devices not responding - I have 15 units of AM43 roller blinds motor and some of them do not respond from time to time.
Beside AM43 I have 40 other devices, and most of them work correctly. I wanted to check if it's coordinator fault.

## What happened
I stopped Z2M container, changed pan_id, switched adapter to CC2652RB and started Z2M. It started, but with 0 devices available.

## What did you expect to happen
I expected all my devices to be available.

## How to reproduce it (minimal and precise)
Stop Z2M, change pan_id, swtich adapter from CC2538 to CC2652RB, start Z2M

## Debug info
Zigbee2MQTT version: 1.21.0 commit: 70891ee
Adapter hardware - from: CC2538 (https://www.tindie.com/products/GiovanniCas/cc2538-cc2592-zigbee-dongle-zb30/) to Slae.sh CC2652RB
Adapter firmware version - from : 20200211
Adapter hardware - to: Slae.sh CC2652RB
Adapter firmware version - to : 20210708

Koenkk, 2021-08-24:
Migrating from a CC2538 to CC2652R without repairing devices can be done by just plugging in the new coordinator and updating the `port` in your `configuration.yaml`.

Changing the `pan_id` requires repairing all devices, if you don't have a backup of your `data` folder I'm afraid the backup got lost and you have to repair all devices.

closed 2021-08-24 by Koenkk

wizu, 2021-08-25:
@Koenkk thanks, it almost work now ;)
But wheh i try to turn on switch i got:
`Publish 'set' 'state' to 'ikea_socket_c' failed: 'Error: Command 0x086bd7fffe5df97c/1 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'`

I've tried to place extension cable at another place, but that didn't fix it.

Koenkk, 2021-08-25:
MAC channel access failure happens when the coordinator is unable to transmit something due to interference, make sure to check your zigbee channel: https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

wizu, 2021-08-25:
@Koenkk I've set my wifi at 6 and zigbee at 25 so it shouldn't interfere. My currurent CC2538 have no problem here and i've replaced it with CC2652RB at the same place.
But looking closely at my new CC2652RB adapter from Slae.sh it looks like one leg of antenna socket isn't soldered to pcb. I will contact Slae.sh and ask about it.