Koenkk/zigbee2mqtt #8497: TS0121_plug - not polling anymore ? but not all.

Masterz69, 2021-08-23:
## What happened

At some moment had noticed : part of my [TS0121_plug ](https://www.zigbee2mqtt.io/devices/TS0121_plug.html#tuya-ts0121_plug) not polled anymore every minute.
[image]
Looks like, isn't it.

Log level is debug, but see no records about some fails or timeouts during readings from sockets.

Two of them still working - as couple of month till ~now:
```
Zigbee2MQTT:debug 2021-08-24 00:37:06: Received Zigbee message from 'plg Xbox', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":234}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-08-24 00:37:06: MQTT publish: topic 'zigbee2mqtt/plg Xbox', payload '{"current":0,"energy":131,"last_seen":"2021-08-24T00:37:06+03:00","linkquality":48,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":234}'
```
## What did you expect to happen

All sockets polled as described.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.21.0-4 
Adapter hardware: CC2652 
Adapter firmware version: zStack3x0 20200921

## P.S.
Let me know if I can provide more info about case.

mrTuomoK, 2021-08-24:
Try to set the polling interval in the config with `measurement_poll_interval ` (just an idea, might not help at all though)

Masterz69, 2021-08-24:
Re-joined 3 of them.
Now 4 of 5 looks fine - poll happens every minute.
But one (MultiV) is not - which worked yesterday. 
Mystic.

[image]

Masterz69, 2021-08-25:
So, in short.

August 22-24 in Riga district, Latvia - has multiple issues:
* polling not working partly;
* some devices not reporint for several hours;
* join fails for couple of devices.

But starting after August 24 noon - all again working fine.

Is it "solar wind" or some interference from neghbours etc. ?

Now it works on same Z2M/HA version, anything same in my household. Mystery.

closed 2021-08-25 by Masterz69