Koenkk/zigbee2mqtt #8499: Batteries values are questionable

andriej, 2021-08-24:
## What happened
My few-years-old door window's sensor from Xiaomi died. With it's battery value until the last day reported at 100%.

## What did you expect to happen
To see battery value more-real, probably like values from Xiaomi Gateway.

Was waiting with this report, but I have to admit - there's an issue with battery reporting/gathering in z2m.
Before I've moved to z2m I had most of the sensors I have now conencted to Aqara gateways - each of them was reporting their value getting lower and lower, was much easier to see which sensors had around 40% and which 60-70.
Right now it's total guess game.

What could be the reason behind it? I know the value of voltage is reported - but it doesn't change during the time pass - the sensor died had the same 'voltage' report as all other sensors have - 3035 mV. So it's purely wrong information.
What Aqara gateway does differently? May it be Xiaomi-specific handling for implementation?
I hope the discussion starts and z2m will get even better -  currently it's the last struggle I have with my installation, everything else works just fine

## How to reproduce it (minimal and precise)
Just look at list of all door-sensors I have since few years.
Can't see or even guess when they willl die, before 'availability' kicks in :/
[image]

## Debug info
Zigbee2MQTT version: 1.21.0-4
Adapter hardware: CC2652P zStack3x0
Adapter firmware version:  20210708

andriej, 2021-08-24:
And not to blame it only on door-window sensors (MCCGQ11LM), here are:

- smoke sensors (JTYJ-GD-01LM/BW aka Honeywell Xiaomi)
[image]

- xiaomi buttons ( WXKG11LM and WXKG01LM )
 [image]

- motion sensors ( RTCGQ01LM and RTCGQ11LM)
[image]

And this installation is a-lot-months old already, so it's not fault of often restarts/rebindings or not gathered enough data...

Shred99, 2021-08-25:
Are you sure that this is a problem with Zigbee2MQTT?  As far as I know, all it does is publish the values transmitted by the ZigBee sensor.

I have quite a number of Xiaomi sensors that have been in service for 2-3 years... possibly longer.  I'm currently seeing most of the door switch sensors reporting battery health in the 91% - 100% range.  I have three environment (temperature/humidity/barometric pressure) sensors that send updates quite frequently.  One of them recently needed a new battery and I was able to see the voltage and battery percentage trend down over time, before the sensor finally stopped transmitting.

andriej, 2021-08-25:
I'm almost certainly sure, as those sensors in xiaomi aqara gateways and it's app reported values that seemed reasonable. Much more reasonable. So 90% after few months is much more reasonable than 100% after 3 years.
On the same sensor.

McGiverGim, 2021-08-25:
As far as I know the sensor only sends the voltage, and z2m translates it into a percentage the best it can. 
So if you don't see a change in the voltage, you will not see any change in the percentage.
Here is where it makes the magic:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/fe7d0ec950dec5af824e153220777806f29ac59a/converters/fromZigbee.js#L4277-L4301
and here (I think it uses the 3V_2100 option):
https://github.com/Koenkk/zigbee-herdsman-converters/blob/fe7d0ec950dec5af824e153220777806f29ac59a/lib/utils.js#L132-L168
The conversion from voltage to percentage maybe can be done better/different, but first you need the voltage to be changing.
Over 3 V it will always show 100%.

andriej, 2021-08-25:
@McGiverGim I'm aware it's z2m that does the calculation. I've opened the issue because I'm somehow nervous about not having oppoturnity to i.e. exchange batteries before I go for holiday, because I can see they're low (i.e. <20%).
Main question is - what's the difference with approach of Xiaomi Aqara gateway (for specifically Xiaomi sensors), that makes their values much more reasonable?

If sensors send it's voltage, are we 101% sure it's pure voltage without any calculations done?
I guess it's not difference in protocol - older Xiaomi gateways were zigbee 2 (the ones I had), newer ones are zigbee 3 already.

McGiverGim, 2021-08-25:
It works with the data the sensors sent. Nothing more. If it is the real or some strange calculation voltage I can't say. The same does the Aqara/Xiaomi gateway as far as I know.
The method z2m uses for the percentage is the best z2m found, because z2m don't know how Aqara/Xiaomi converts the voltage to percentage. If someone knows how it does this conversion, it can be changed to be similar.

andriej, 2021-08-25:
I started to dig some once again, as maybe there's a oppoturnity for some fine tuning.
So, calculations are/were (don't know, don't use) in ZHA too: https://github.com/home-assistant/core/issues/44578
but!
I've found some nice battery-level count method, which I've checked with one of my sensors:
https://community.home-assistant.io/t/xiaomi-aqara-battery-level/36222/14

So, taking a look at:
(2.8 is 'completely dead, 2.9 should be considered to exchange):
[image]

z2m shows 100%, solid 
while:
`Math.min(Math.round((voltage - 2200) / 10), 100)` found and used in nodered is:
((3025 - 2200) / 10) -> 82,5%

and that's the value I could believe, because I remember that I was exchanging that battery some time ago already.
Let's take a look on the other one I've changed earlier:
(3085 - 2200) / 10 -> 88,5%

I'm just not sure if taking '2500mV' to account is proper way for calculation, as I don't think any xiaomi sensor will report anything with battery lower than 2.8V

It's open for discussion, I'd just like to have my fav software fine-tuned in that aspect too... and cut the myth of 100% ;-)

Koenkk, 2021-08-25:
@andriej since we focus on Xiaomi devices here, could you change the title to mention Xiaomi?

My WXKG02LM_rev1 currently reports the following so it should drop off soon, will keep an eye on the last known voltage before it stops working.

[image]

andriej, 2021-08-25:
@Koenkk this will be surely nice catch - the sensor that was the reason of starting this thread has gone out with 3035 (last value I see in HA) and I'm even more worried about it right now. ;-)

To not be blamed of being 100% pesimistic - there's something 'better' in calculating temp/hum and temp/hum/press sensor:
[image]
WSDCGQ11LM and WSDCGQ01LM

Maybe they get different counting schema and that's why their values are more to be respected? 
I don't have much more stuff from Xiaomi with batteries, but also have GZCGQ01LM outside house.
Would you believe it has 100% after winter out there? ;-) It's voltage is 3000, so below what's 100% (new battery), but z2m shows as 100%.

I hope maybe more people will join the thread and solution may come later on.

rindlerblabla, 2021-08-25:
I have some 20 of those contact sensors. Those reporting 3015-3025 mV report 100%, 2995 mV report 97%, 2985 mV report 91% and 2975 report 86%.

McGiverGim, 2021-08-25:
One of my WSDCGQ01LM is low too:

[image]

I will keep an eye on it too, but I'm pretty sure it can last some months with this battery level. And I suppose different devices will have different dead levels.

andriej, 2021-09-25:
Still valid I guess

zen2, 2021-10-16:
I'm come here following two dead batteries on xiaomi temperature/humidity sensors.
Last battery % i get from this sensors was respectively 91%(2985mv) and 100%(3005mv).

I suspect several things about % calculation:
- calculation is not linear since battery discharge is not
- calculation is depending on temperature/humidity
- different kind of sensors can have different minimum voltage to work properly
- discharge is depending on resistive load of device
- possibility that xiaomi gateway battery calculation is based with a 100% based on new battery voltage

So it's a complex topic to find the perfect solution.
I can see only two way to find an acceptable solution:

1. the best one: retro-engineering of xiaomi gateway (or may be application ?)
2. use a xiaomi gateway and try to guess a general formula depending of voltage seen, battery % from gateway/app, kind of sensors, temperature/humidity
3. create a central database of voltage for different kind of sensors from several users to determine minimum voltage for each kind of sensors

For the battery problem alert, actually it's not possible to rely on % battery for all kind of devices by experience.
I actually use last_seen report to detect died battery since sensors emit regularly and suddenly stop.
But I can't see a way to alert before battery die.

Complete PDF about CR2032 lithium battery: [High pulse drain impact on CR2032 coin cell battery capacity](https://www.dmcinfo.com/Portals/0/Blog%20Files/High%20pulse%20drain%20impact%20on%20CR2032%20coin%20cell%20battery%20capacity.pdf)

andriej, 2021-11-24:
Still valid

McGiverGim, 2021-11-24:
I'm waiting to see when my device dies. The latest report was at 40%, the 25th august. Now is at 29% (2.815V) and working...

larod241, 2021-12-01:
Hi, 

To help, my WSDCGQ11LM stop working. 
Last values before it die : 
[image]

Hope it can help.

Koenkk, 2021-12-01:
@larod241 thanks, I will keep track of these values here: https://pastebin.com/64HP9i10 (so others please also report the last voltage before it dies)

larod241, 2022-01-01:
This morning : WSDCGQ11LM stop to work. Last voltage seen : 2965 mV

McGiverGim, 2022-01-01:
> This morning : WSDCGQ11LM stop to work. Last voltage seen : 2965 mV

I've one with 2845 mV working without problem...

larod241, 2022-01-01:
> > This morning : WSDCGQ11LM stop to work. Last voltage seen : 2965 mV
> 
> I've one with 2845 mV working without problem...

I also have sensors that work under this voltage.
But, in order to capitalize and succeed in finding an average to define a real average percentage, I note, as requested, the last known voltage when a xiaomi device turns off and the battery needs to be changed.

verstrepeng, 2022-01-05:
I got 3 sensor that are working, different voltages:
3025mV reports 100%
2995mV reports 97%
3045mV reports 100%
As 100% should be 3100mV (or higher), 3025 should already be lower.
We should consider 2800 or lower as dead, new batteries can be at 3200mV, so shouldn't we just make a linear calculation between those values: 2800mv=0% and 3200mV=100% ?

How does ZHA make the calculation, because this was more correct for me.

larod241, 2022-01-05:
Hello,

It is not that simple. the actual ascent of the device must be taken into account. And it turns out that often it stops working around 2900mv ...
The important thing is to be warned before the sensor stops in order to be able to anticipate the battery change.

This is why there is a "pastebin" link in order to identify the last valuation recorded before the device where down.

verstrepeng, 2022-01-06:
Never said it was simple ;-)
I just migrates my 2 last aquara sensors from ZHA to Z2M and this time I noted the values, might be helpful...
Sensor 1 - ZHA : 66% - Z2M : 100%, 3005mV
Sensor 2 - ZHA : 59% - Z2M : 91%, 2985mV
Quite big differences, so I think it's very interesting to see how ZHA makes the calculation.

larod241, 2022-01-06:
ZHA seems to be close to reality !

zen2, 2022-01-06:
I got a Aqara Door/Window Contact sensor (MCCGQ11LM) that stop to communicate few days ago.
Last battery info was:   15% | 2655mV

If you want more data, this is the actual status of all my battery based sensors that currently report regularly:

"voltage":2855 "battery":35 Aqara WSDCGQ11LM [Temperature]
"voltage":2885 "battery":40 Aqara WSDCGQ11LM [Temperature]
"voltage":2895 "battery":41 Aqara WSDCGQ11LM [Temperature]
"voltage":2955 "battery":74 Aqara WSDCGQ11LM [Temperature]
"voltage":2955 "battery":74 Aqara WSDCGQ11LM [Temperature]
"voltage":2955 "battery":74 Aqara WSDCGQ11LM [Temperature]
"voltage":2955 "battery":74 Aqara SJCGQ11LM [waterleak]
"voltage":2965 "battery":80 Aqara WSDCGQ11LM [Temperature]
"voltage":2975 "battery":86 Aqara MCCGQ11LM [Contact]
"voltage":2975 "battery":86 Aqara MCCGQ11LM [Contact]
"voltage":2975 "battery":86 Aqara MCCGQ11LM [Contact]
"voltage":2975 "battery":86 Aqara WSDCGQ11LM [Temperature]
"voltage":2975 "battery":86 Aqara WSDCGQ11LM [Temperature]
"voltage":2985 "battery":91 Aqara MCCGQ11LM [Contact]
"voltage":2985 "battery":91 Aqara MCCGQ11LM [Contact]
"voltage":2985 "battery":91 Aqara MCCGQ11LM [Contact]
"voltage":2985 "battery":91 Aqara WSDCGQ11LM [Temperature]
"voltage":2985 "battery":91 Aqara WSDCGQ11LM [Temperature]
"voltage":2995 "battery":97 Aqara MCCGQ11LM [Contact]
"voltage":2995 "battery":97 Aqara MCCGQ11LM [Contact]
"voltage":2995 "battery":97 Aqara MCCGQ11LM [Contact]
"voltage":2995 "battery":97 Aqara MCCGQ11LM [Contact]
"voltage":2995 "battery":97 Aqara RTCGQ11LM [motion]
"voltage":2995 "battery":97 Aqara RTCGQ11LM [motion]
"voltage":2995 "battery":97 Aqara WSDCGQ11LM [Temperature]
"voltage":3002 "battery":100 Aqara WXKG01LM [switch]
"voltage":3005 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3005 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3005 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3005 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3005 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3005 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3005 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3005 "battery":100 Aqara RTCGQ11LM [motion]
"voltage":3005 "battery":100 Aqara RTCGQ11LM [motion]
"voltage":3005 "battery":100 Aqara SJCGQ11LM [waterleak]
"voltage":3005 "battery":100 Aqara SJCGQ11LM [waterleak]
"voltage":3012 "battery":100 Aqara WXKG01LM [switch]
"voltage":3015 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3015 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3015 "battery":100 Aqara RTCGQ11LM [motion]
"voltage":3015 "battery":100 Aqara WSDCGQ11LM [Temperature]
"voltage":3015 "battery":100 Aqara WSDCGQ11LM [Temperature]
"voltage":3025 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3025 "battery":100 Aqara MCCGQ11LM [Contact]
"voltage":3025 "battery":100 Aqara RTCGQ11LM [motion]
"voltage":3035 "battery":100 Aqara RTCGQ11LM [motion]
"voltage":3035 "battery":100 Aqara RTCGQ11LM [motion]
"voltage":3075 "battery":100 Aqara WSDCGQ11LM [Temperature]

I don't rely on voltage or battery % to predict batteries failure so I actually rely on last report timestamp.
Battery based device report regularly their states. Aqara ones report their states at least once a hour.
So if you got no report for 2 hours, you know the battery have died or worst you got a problem on your zigbee network.
I think actually it's the best way to detect battery failure.

It's not obvious to translate voltage to battery percent since each device doesn't need same amount of voltage to function.
in addition batteries discharge is not linear and temperature affect it too.
I got a temperature/humidity sensors in a fridge and another one in wine cave and its reports always lower voltage than similar sensors. 
So If you want to collect data to make a voltage based correlation you need at least to do it for each kind of device.
To collect temperature could be great too but difficult since you need another sensor for it.

I use this bash script to report batteries failure:

zigbee-last-report-warning.sh
```
#!/bin/bash

z2m_path="/home/zigbee2mqtt/data"
conf="$z2m_path/devices.yaml"
logs="$z2m_path/log.txt"
exclude="light\."
delta_max=7200

day="$1" ; day="${day:-today}"
[ "$day" = "today" ] && now="$(date +%s)" || now="$(date +%s --date="$day +1day 00:00:00")"

devices="$(cat "$conf" | grep friendly | cut -d' ' -f4 | grep -vE "$exclude")"
for device in $devices ; do
  list="$(cat $logs | grep "$(date +%F -d "$day").*MQTT publish: topic "\'"zigbee2mqtt/$device"\')"
  count="$(echo -e "$list" | grep -c "^.")"
  last="$(echo -e "$list" | tail -1 | cut -d' ' -f3,4 | cut -d: -f1-3)"
  delta="$(( $now - $(date +%s --date="$last") ))"
  [ $delta -gt $delta_max ] || [ $delta -lt 0 ] && notice="[WARNING]" || notice=""
  echo "$last $device [$count] $delta $notice"
done | sort -n
```
You can easily modify it to collect automatically data or make a command based sensor for home assistant.

thomashogema, 2022-01-06:
> ZHA seems to be close to reality !

When looking into this problem I found a similar issue raised at ZHA (Zigpy) which might help: https://github.com/zigpy/zha-device-handlers/issues/96 

Here they resolved the questionable battery values by changing the minimum and maximum voltage levels to `2800` and `3300` respectively in this PR  https://github.com/zigpy/zha-device-handlers/pull/174

Koenkk, 2022-01-06:
@thomashogema that calculation isn't correct either, if you look at https://pastebin.com/64HP9i10 some die at 2965

verstrepeng, 2022-01-07:
> @thomashogema that calculation isn't correct either, if you look at https://pastebin.com/64HP9i10 some die at 2965

Which is not normal...

larod241, 2022-01-12:
Hello, 

Hi,

New WSDCGQ11LM stop to work this morning. 
Last seen voltage : 2845 mV

thomashogema, 2022-01-12:
> that calculation isn't correct either, if you look at https://pastebin.com/64HP9i10 some die at 2965

@Koenkk fully agreed that it is not a perfect calculation.

However, it can provide a more accurate estimate. Given the diverse range of 'last seen voltages' for the same device, I think the best we can do is an estimation

Koenkk, 2022-01-12:
Let's gather some more voltage values when the device stops working, then we can decide what to lower bound of the calculation should be.

larod241, 2022-01-12:
for information, the full voltage after changing the battery is, in general, 3185mV

kevincaradant, 2022-01-18:
Hi
Just to give my cases

I'm running with https://www.zigbee2mqtt.io/devices/WXKG06LM.html#xiaomi-wxkg06lm

And its working but really badly, lost connection on the network every two weeks and sometimes no blue light ( or very weak) if I click too often on it in the same minute 
Battery :2980mv

The previous one stop to work at 2850 if my memory is good.

My thermometer also keep going to work at 2800mv but I got one disconnection by month. It's soon the end for it. 

Hoping that help to calculate the battery level

Cossey, 2022-02-01:
Last report from one of my `WSDCGQ11LM` before it went offline - died at 2965mV
```
Firmware build date: 20161129
Firmware version: 3000-0001
```

kevincaradant, 2022-02-02:
Hi

@Koenkk, I don't know if you saw my previous messages in the aim to add my values to your memo. 
However, I have 25 xiaomi dévices, I think it will useful to know that 75% of devices died at 2850mV and just two died at 2950mV. 

As I said I think it's most normal they died  around of 2850mv 😁.

Only one or two devices still work at 2750 / 2800mV but with some disconnections sometimes.

Nb: the xiaomi aqara devices are motion detection, open/Close door, water leak sensors, thermometers and wall switches. 

Tuya devices are remotes ( 3 devices) 

Regards

Cossey, 2022-02-02:
Does the firmware version of the device matter/affect the lowest operating mV? 
There are some discussions of the other Xiaomi sensors in this issue too but the gist is only tracking the THP sensors.

Is there a way to discharge new CR2032 batteries quickly? It would be nice to be able to connect a few of these sensors up with a gateway with developer mode on and then we could just sniff the zigbee traffic and match up the mV values with the gateway messages as they discharge.

kevincaradant, 2022-02-02:
> Is there a way to discharge new CR2032 batteries quickly? It would be nice to be able to connect a few of these sensors up with a gateway with developer mode on and then we could just sniff the zigbee traffic and match up the mV values with the gateway messages as they discharge.

In my cases, to discharge or believe it's decharged quickly I got two factors:
   - My outdoor captors ( water leak, temperature and motion detection ) have only 35% of battery when it's cold outside ( ~0 degrees ) the morning and 50% of battery the afternoon ( 8 degrees )
   - Have like me an issue of water leak which is not really waterproof now and the new CR2032 go from 100% to 40% in 5 minutes after to put it in water.

Koenkk, 2022-02-02:
@kevincaradant 

- what xiaomi devices are these?
- did the xiaomi devices also die at 2950?

kevincaradant, 2022-02-02:
~35 devices, and my xiaomi devices are:

RTCGQ11LM (x10)
SJCGQ11LM (x3)
WSDCGQ11LM (x6)
WXKG11LM (x4)
WXKG06LM (x4)
MCCGQ11LM (x6)

Around of 20 died since one year ( original battery ) . Only one or two died at 2950mV. It's uncommon. 
17 are died around of 2850mv.
And two WSDCGQ11LM still works at 2800mV but I thik it's soon the end for them. 

The average is more 2850mV for me.

Cossey, 2022-02-07:
I vote to update the values to the suggested (2850mV) in the interim - just so that I don't get any surprises with offline devices. It can be refined further later with more submissions from people (ill also submit further data from my sensors when they start to die too, which could take some time).

Koenkk, 2022-02-07:
Implemented the improved calculation:
- 2850 or lower = 0%
- 3200 or higher = 100%
- 2850-3200 = linear calculation

zen2, 2022-02-07:
I found this picture that represent relationship between voltage and temperature:
[image]

It could be nice to measure voltage at different temperature to calculate the delta that could be used on end devices that report temperature.

kevincaradant, 2022-02-07:
So scary now. Just before 90% of my devices was at 100%. Now it's different, more realistic 😁

Let's give a try to this :) 
[image]
[image]

ristomatti, 2022-02-08:
I upgraded to the latest Docker image, now running:

**Zigbee2MQTT version:** 1.23.0-dev commit: [6394d0f](https://github.com/Koenkk/zigbee2mqtt/commit/6394d0f)
**Coordinator type:** zStack3x0
**Coordinator revision:** 20210708

It seems only some of my Xiaomi sensors report based on the new calculations. Examples:

[WSDCGQ11LM](https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html#xiaomi-wsdcgq11lm) Aqara temp/humi sensor:
```json
{
    "battery": 91,
    "humidity": 32.05,
    "last_seen": 1644302968470,
    "linkquality": null,
    "pressure": 998.8,
    "temperature": 21.1,
    "voltage": 2985
}
```
[DJT11LM](https://www.zigbee2mqtt.io/devices/DJT11LM.html#xiaomi-djt11lm) Aqara vibration sensor:

```json
{
    "angle": 6,
    "angle_x": 4,
    "angle_x_absolute": 86,
    "angle_y": 10,
    "angle_y_absolute": 80,
    "angle_z": -79,
    "battery": 56,
    "last_seen": 1644304827717,
    "linkquality": 75,
    "strength": 17,
    "vibration": false,
    "voltage": 3045,
    "action": "vibration",
    "sensitivity": null
}
```

Will it just take some time for the values to update or do I need to run a newer firmware rev also?

McGiverGim, 2022-02-08:
> Implemented the improved calculation:
> 
> * 2850 or lower = 0%
> * 3200 or higher = 100%
> * 2850-3200 = linear calculation

Maybe I'm very lucky. I have for example five Xiaomi temperature devices. Only one of them is over 2850 (exactly 2875) All the others are lower. So when I update I will have a lot of devices with 0% battery, only the latest one of this list will have a positive value:
[image]

I will keep posting when they really die, to see what happens. At this moment the minimum of them have a value of 2775.

zen2, 2022-02-08:
I wrote a simple bash script to help to track voltage/battery and temperature if available:
```
#!/bin/bash
#
# Display for each device last report temperature/battery/voltage
# sorted by voltage by parsing zigbee2mqtt log file
#

z2m_path="/home/zigbee2mqtt/data"
conf="$z2m_path/devices.yaml"
logs="$z2m_path/log.txt"

devices="$(cat "$conf" | grep friendly | cut -d' ' -f4)"
for device in $devices ; do
  last="$(cat "$logs" | tac | grep -m1 " MQTT publish: topic "\'"zigbee2mqtt/$device"\' | cut -d\' -f4)"
  if grep -q '"battery":.*"voltage":' <<< "$last"; then
    battery=$( sed -E 's|.*"battery":([0-9]*)[,}].*|\1|' <<< "$last" )
    voltage=$( sed -E 's|.*"voltage":([0-9]*)[,}].*|\1|' <<< "$last" )
    if grep -q '"temperature":' <<< "$last"; then
      temperature=$( sed -E 's|.*"temperature":([0-9.]*)[,}].*|\1|' <<< "$last" )
    else
      temperature='N/A'
    fi
    echo "$device $temperature $battery $voltage"
  fi
done | sort -rnk4 | column -t -N "Device,Battery,Voltage,Temperature"
```
With current stable version 1.23.0 of zigbee2mqtt so without last modification of battery calculation I got this:
```
Device                                 Battery  Voltage  Temperature
alarm                                  N/A      100      4100
contact.congelateur                    25       100      3105
presence.escalier                      20       100      3035
presence.wc_etage                      18       100      3025
presence.bal                           18       100      3025
contact.fenetre_sdb                    19       100      3025
contact.fenetre_ch_amis                18       100      3025
thp.salon                              20.22    100      3015
thp.cuisine                            19.3     100      3015
presence.couloir_etage                 23       100      3015
contact.porte_wc_rdc                   20       100      3015
contact.fenetre_wc_rdc                 16       100      3015
switch.2                               N/A      100      3012
presence.wc_rdc                        18       100      3005
contact.porte_fenetre_salon_jardin     23       100      3005
contact.porte_fenetre_bureau_terrasse  22       100      3005
contact.porte_ch_jeux                  25       100      3005
contact.porte_ch_jeff                  23       100      3005
contact.porte_ch_charlie               21       100      3005
contact.fenetre_bureau                 23       100      3005
switch.ch_jeff                         N/A      100      3002
waterleak.exterieur                    15       97       2995
thp.wc_rdc                             15       97       2995
thp.terrasse                           14.88    97       2995
thp.bureau                             19.12    97       2995
presence.evier                         N/A      97       2995
presence.cuisine                       26       97       2995
presence.buanderie                     21       97       2995
contact.porte_wc_etage                 23       97       2995
contact.porte_entree                   25       97       2995
contact.portail                        21       97       2995
contact.fenetre_ch_charlie             23       97       2995
thp.ch_jeff                            17.86    91       2985
thp.ch_charlie                         17.83    91       2985
contact_fenetre_wc_etage               21       91       2985
contact.porte_ch_amis                  23       91       2985
contact.fenetre_cuisine                25       91       2985
contact.porte_sdb                      25       86       2975
contact.porte_fenetre_salon_terrasse   27       86       2975
contact.porte_entree_buanderie         21       86       2975
contact.fenetre_ch_jeux                28       86       2975
contact.fenetre_ch_jeff                28       86       2975
thp.ch_amis                            15.66    80       2965
thp.buanderie                          16.34    80       2965
waterleak.vide_sanitaire               23       74       2955
thp.sdb                                16.74    74       2955
thp.ch_jeux                            17.94    74       2955
thp.refrigerateur                      8.14     62       2935
waterleak.studio                       16       51       2915
thp.studio                             16.43    37       2865
thp.cave_vin                           9.8      35       2855
thp.combles                            15.09    31       2825
```
I hope this script can help this thread to make correlation.

kevincaradant, 2022-02-08:
> Maybe I'm very lucky. I have for example five Xiaomi temperature devices. Only one of them is over 2850 (exactly 2875) All the others are lower. So when I update I will have a lot of devices with 0% [...]

Hi

What's are your temperatures ? I think you are lucky because I don't remember the last time where I got less than 2800. Not sure to got it already once. 

2775 seems to me really low. Too much in the most of cases but I'm interesting to know from other people. 

BTW : @McGiverGim, I'm really interested to know if you unpaired it and try to pair yours really low devices, is it working? 
Why I ask that? Because I already got devices with around 2850 for exemple which worked just because they send a mqtt message to Z2M but when I needed to pair it again ( new channel in my case) , it never works correctly because the device turn off during the pairing ( issue about low voltage or amperage of the battery)

jthomaschewski, 2022-02-08:
> What's are your temperatures ? I think you are lucky because I don't remember the last time where I got less than 2800. Not sure to got it already once.
> 
> 2775 seems to me really low. Too much in the most of cases but I'm interesting to know from other people.

@kevincaradant Recently one of my Xiaomi temp sensor (WSDCGQ11LM) died at 2705. It was working weeks at below 2800 before turning off.

Currently I have a different sensor (same model) at 2805 and its still working without issues.
I just unpaired it and tried to repair = did not work. The LED brightness is very low but it still acknowledges pairing mode - but nothing in the log of z2m. After replacing the battery it joined instantly.
Measured voltage with multimeter now: 2815 (without load/resistor)

I remember that once an occupancy sensor (RTCGQ11LM) stopped working at higher voltage, but I'm not 100% (been a while, didn't noted the voltage).

kevincaradant, 2022-02-08:
> @kevincaradant Recently one of my Xiaomi temp sensor (WSDCGQ11LM) died at 2705. It was working weeks at below 2800 before turning off.

Ok thanks for your feedback, which temperature have the device around of it by curiosity? More like 0c or 20c ? 

> Currently I have a different sensor (same model) at 2805 and its still working without issues.
I just unpaired it and tried to repair = did not work. The LED brightness is very low but it still acknowledges pairing mode - but nothing in the log of z2m. After replacing the battery it joined instantly.
Measured voltage with multimeter now: 2815 (without load/resistor)

You exactly talk about my currently question for @McGiverGim . 
Even if the battery is low ( can send MQTT message but can't do a pairing ) , I think we can consider that the battery is close to 0% .
 It still remains my point of view indeed.

I had many times this issue. Works well at 2800 / 2850 until the next pairing with blue light which blink (low powerful) or just turn off and the device died. No pairing until I change the battery.

jthomaschewski, 2022-02-09:
> Ok thanks for your feedback, which temperature have the device (environment) by curiosity? More like 0c or 20c ?

Around 20c

> I still think 2800 or 2850 is a good average to be compatible with the most devices.
I agree. 
Didn't want to make a point against the suggested limit. Just share my experiences/numbers.

I'm happy if I get notified when battery is low - doesnt matter if its days, weeks, 1 or 2 month until they die. 
Personally I would prefer an too early warning over too late.

Too late = broken automation, no time to buy battery
Too early = possibly too early buy & replacement, but depending on the importance of the device, one can decide to just wait until it dies.

ristomatti, 2022-02-09:
> I wrote a simple bash script to help to track voltage/battery and temperature if available:
>
> ```
> #!/bin/bash
> #
> # Display for each device last report temperature/battery/voltage
> # sorted by voltage [...]

The script seems platform specific. My `data` dir does not have `devices.yaml`, maybe the equivalent is `configuration.yaml` in my case(?)-. The script also gives error as the `sort` command does not recognize parameter `-N` (gives `column: invalid option -- 'N'`). I tried changing `-N` to `-n` (for numeric) and `devices.yaml` to `configuration.yaml` -> `column: Device,Battery,Voltage,Temperature: No such file or directory`....

ristomatti, 2022-02-09:
I agree that early warning is better than late one. As @jthomaschewski said, if it reports 0% but still works, you can decide how to react. It could be perhaps be improved with a fancier (but more assuming) logic which would report a low percentage (1-3%) a the voltage some devices stop sending data but 0% only when a given last seen time has past. That would be clearly a feature request territory though. Although it'd be nice, it definitely would not go past supporting the current feature set for new devices if I were asked. :)

tjxism, 2022-02-09:
Hi all, I am new here and this post is very informative. 

I currently only have 2 Aqara sensors connected through z2m:

- Aqara TVOC air quality monitor
- Aqara P1 door & window contact sensor
Both don't expose voltage to z2m as I see in the exposed tab. 
The air monitor exposes a percentage. When I first connect it shows 27% (after a tiring pairing) and now it is back to 100%. 
The P1 door sensor simply exposes a green battery sign. 

Is this normal or it is because I have disabled some settings in z2m? I'd love to see more accurate battery status but if it's a decision on the hardware side then I'll try to trust it...

McGiverGim, 2022-02-09:
> You exactly talk about my currently question for @McGiverGim .
Even if the battery is low ( can send MQTT message but can't do a pairing ) , we can consider that the battery is close to 0% even if it "survive" .
It still remains my point of view indeed.

Yes, I think the value chosen must be the one that gives full functionality to the device. I only showed my devices, to show that we will have a lot of devices with 0% battery (and maybe some users questioning the value of it).

I usually have them between 19-21 C in winter, a little more in summer (they are inside home). One of the devices does not seem to respond to the button press, so I doubt it will pair again if needed. I will let them for curiosity to see where they really die, but seeing that others (@jthomaschewski ) have confirmed the problem to pair with very low battery, seems acceptable the current battery change done.

I have two xiaomi vibration sensors, that had problems to pair before with low battery, but I'm not too sure at which level. This vibration sensors are more power hungry, the battery last less than one year for sure. I will keep an eye on them to confirm the values are valid for this kind of sensor too.

I have too door sensors, water leak, smoke sensor, presence sensors... but this batteries last a lot more so I don't know when they will be low enough to give some information.

McGiverGim, 2022-02-09:
More info:
- One of the vibration devices has just died [DJT11LM](https://www.zigbee2mqtt.io/devices/DJT11LM.html#xiaomi-djt11lm), the latest reported battery was as high as 2.965v. This is not the first time that happens to me, I have two devices like this and usually have problems with them that die with high battery reports. Someone more has this kind of devices to see if it's only me or not? Maybe some defective batteries? They are Sony, but maybe they are a bad clone and I think the original batteries died soon too.
- One of the temperature devices (from Aqara) has just died too: [WSDCGQ11LM](https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html#xiaomi-wsdcgq11lm), the latest reading was 2.815 V.

zen2, 2022-02-09:
> The script seems platform specific. My `data` dir does not have `devices.yaml`, maybe the equivalent is `configuration.yaml` in my case? The script also gives error as the `sort` command does not [...]

Yes I use the "device:" option in my configuration to have a separate file for devices description.
So I update my script (in original comment) to work for both case: devices description in main configuration file or a specific configuration file.
I don't write platform specific bash script. It's pure bash and column is a 70's command so the error about "-N" option surprise me. This option is used to set columns headers:
` -N, --table-columns <names>      comma separated columns names`
This option have been added in linux-utils 2.30 released in 2018.
Can you give me the version you got using column --version please ?

ristomatti, 2022-02-09:
> I don't write platform specific bash script. It's pure bash and column is a 70's command so the error about "-N" option surprise me. This option is used to set columns headers:
> ` -N, --table-columns [...]

There are a surprising amount of little differences depending on the background of the tools (GNU/BSD etc.). I've struggled writing utility scripts on OS X as so many commands work differently than on Linux.

> Can you give me the version you got using column --version please ?

<details>
  <summary>Click to expand off topic content</summary>

```bash
$ column --version
column: invalid option -- '-'
usage: column [-txne] [-c columns] [-s sep] [file ...]

$ dpkg -S /usr/bin/column 
bsdmainutils: /usr/bin/column

$ dpkg -l | grep bsdmainutils
ii  bsdmainutils                         11.1.2                              armhf        collection of more utilities from FreeBSD
```

This is on Raspbian Buster but apparently my Linux Mint (~Ubuntu) laptop has the same version. Both are Debian based.

From "man column" (this BSD variant doesn't seem to support --help either)
```
 The options are as follows:

 -c      Output is formatted for a display columns wide.
 -s      Specify a set of characters to be used to delimit columns for the -t option.
 -t      Determine the number of columns the input contains and create a table.  Columns are delimited with
         whitespace, by default, or with the characters supplied using the -s option.  Useful for
         pretty-printing displays.
 -x      Fill columns before filling rows.
 -n      By default, the column command will merge multiple adjacent delimiters into a single delimiter 
         when using the -t option; this option disables that behavior. This option is a Debian GNU/Linux
         extension.
 -e      Do not ignore empty lines.
 ```
</details>

zen2, 2022-02-09:
@ristomatti : thanks to report.
<details>
  <summary>Off Topic - click to expand</summary>
 OS X is a proprietary system BSD based so need pure posix shell.
 I think the "tac command don't exist on OS X anyway.
 It seems you use debian and I'm surprised that debian uses bsd utils and not linux utils.
 Good to know in order to share script.
 BTW I have learned how to hide an off topic comment too ^^
</details>
So this is the new script that should work on linux platform:

<details>
  <summary>Script to print battery/voltage/temperature of devices [click to expand]</summary>

```
#!/bin/bash
#
# Display for each device last report temperature/battery/voltage
# sorted by voltage by parsing zigbee2mqtt log file
#
#

z2m_path="/home/zigbee2mqtt/data"
conf="$z2m_path/configuration.yaml"
logs="$z2m_path/log.txt"

if grep -q '^devices:' "$conf"; then
  device_file="$(grep '^devices:' "$conf" | cut -d: -f2 | tr -d ' ')"
  [[ $d =~ ^/ ]] && conf="$device_file" || conf="$z2m_path/$device_file"
fi

devices="$(cat "$conf" | grep friendly | cut -d' ' -f4)"
( echo "Device Battery Voltage Temperature"
  for device in $devices ; do
    last="$(cat "$logs" | tac | grep -m1 " MQTT publish: topic "\'"zigbee2mqtt/$device"\' | cut -d\' -f4)"
    if grep -q '"battery":.*"voltage":' <<< "$last"; then
      battery=$( sed -E 's|.*"battery":([0-9]*)[,}].*|\1|' <<< "$last" )
      voltage=$( sed -E 's|.*"voltage":([0-9]*)[,}].*|\1|' <<< "$last" )
      if grep -q '"temperature":' <<< "$last"; then
        temperature=$( sed -E 's|.*"temperature":([0-9.]*)[,}].*|\1|' <<< "$last" )
      else
        temperature='N/A'
      fi
      echo "$device $temperature $battery $voltage"
    fi
  done | sort -rnk4
) | column -t
```

</details>

ristomatti, 2022-02-10:
@zen2 Still fails. This reflects pretty much my experience with the cross-platformity of bash scripts.

<details>
  <summary>Click to expand off topic content</summary>

I added two lines at the start to fail fast and help with debugging:

```bash
set -euo pipefail
set -x
```

I'm getting:

```
$ ./battery-levels.sh 
+ z2m_path=/home/pi/IOTstack/volumes/zigbee2mqtt/data
+ conf=/home/pi/IOTstack/volumes/zigbee2mqtt/data/configuration.yaml
+ logs=/home/pi/IOTstack/volumes/zigbee2mqtt/data/log.txt
+ grep -q '^devices:' /home/pi/IOTstack/volumes/zigbee2mqtt/data/configuration.yaml
++ grep '^devices:' /home/pi/IOTstack/volumes/zigbee2mqtt/data/configuration.yaml
++ tr -d ' '
++ cut -d: -f2
+ device_file=
./battery-levels.sh: line 16: d: unbound variable

# With the added commands, line 16 is:
# [[ $d =~ ^/ ]] && conf="$device_file" || conf="$z2m_path/$device_file"
```
</details>

zen2, 2022-02-10:
> @zen2 Still fails. This reflects pretty much my experience with the cross-platformity of bash scripts.
My mistake: it should be '$device_file' and not '$d' for this test. I've fixed it on last script post.

[218 more replies]