Koenkk/zigbee2mqtt #8500: Missing Battery information for TS0041

snp88, 2021-08-24:
## What happened
Hi All
I have two single button scene switch both are paired and introduce in Zigbe2mqtt as Zigbee Model
TS0041

Triggering action by them working so no issue here 
Unfortunately they are from different manufacture 

No issue at all Zigbee Manufacturer _TZ3000_pzui3skt
Button send battery every time i press it 
Log:
 MQTT publish: topic 'zigbee2mqtt/ScaneSwitch', payload '{"action":"single","battery":100,"linkquality":255,"voltage":3000}'

Second button that i don't see battery status was manufacutred by Zigbee Manufacturer  _TZ3000_fkp5zyho
And when i press it  battery is null in the log also missing voltage

MQTT publish: topic 'zigbee2mqtt/FrontDoorBell', payload '{"action":"single","battery":null,"linkquality":255}'

I try to paring again and no luck issue still persist

## What did you expect to happen
Button will report the battery status
## How to reproduce it (minimal and precise)
Pair button TS0041 manufactured by  _TZ3000_fkp5zyho
Push button few times

## Debug info
Zigbee2MQTT version: 1.21.0-4
Adapter hardware: Conbee II
Adapter firmware version:

danieledwardgeorgehitchcock, 2021-08-24:
Can you paste your database.db entry for the device that isn't reporting the battery?

snp88, 2021-08-24:
HI 

DB entry 

This is entry for working device 

`{
    "id": 26,
    "type": "EndDevice",
    "ieeeAddr": "0x5c0272fffec73923",
    "nwkAddr": 54718,
    "manufId": 4098,
    "manufName": "_TZ3000_pzui3skt",
    "powerSource": "Battery",
    "modelId": "TS0041",
    "epList": [
        1
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 0,
            "inClusterList": [
                0,
                10,
                1,
                6
            ],
            "outClusterList": [
                25
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "65503": "\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\f\u0000\u0000\u0000\u0000\u0011",
                        "65506": 28,
                        "modelId": "TS0041",
                        "manufacturerName": "_TZ3000_pzui3skt",
                        "powerSource": 3,
                        "zclVersion": 3,
                        "stackVersion": 0,
                        "hwVersion": 1,
                        "dateCode": "",
                        "appVersion": 66
                    }
                },
                "genPowerCfg": {
                    "attributes": {
                        "batteryPercentageRemaining": 200,
                        "batteryVoltage": 30
                    }
                }
            },
            "binds": [
                {
                    "cluster": 1,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00212effff0658d3",
                    "endpointID": 1
                }
            ],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 255,
    "stackVersion": 0,
    "hwVersion": 1,
    "dateCode": "",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {
        "configured": 1
    },
    "lastSeen": 1629793128366
}`

This is entry for device with issue 

`{
    "id": 24,
    "type": "EndDevice",
    "ieeeAddr": "0x60a423fffe45ff3f",
    "nwkAddr": 43680,
    "manufId": 4098,
    "manufName": "_TZ3000_fkp5zyho",
    "powerSource": "Battery",
    "modelId": "TS0041",
    "epList": [
        1
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 0,
            "inClusterList": [
                0,
                1,
                6
            ],
            "outClusterList": [
                25,
                10
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "TS0041",
                        "manufacturerName": "_TZ3000_fkp5zyho",
                        "powerSource": 3,
                        "zclVersion": 3,
                        "appVersion": 65,
                        "stackVersion": 0,
                        "hwVersion": 1,
                        "dateCode": ""
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 65,
    "stackVersion": 0,
    "hwVersion": 1,
    "dateCode": "",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {},
    "lastSeen": 1629799764692
}`

As we can see they are quite diffrent ;)

snp88, 2021-08-24:
Quick diff ;)
on left device with the problem
[image]

danieledwardgeorgehitchcock, 2021-08-24:
It looks as though the device doesn't have a genPowerCfg cluster. Do you know if the battery percentage is reported when it is joined to its native hub / app?

snp88, 2021-08-24:
Unfortunately i have only conbee and HA.
I don't have native hub :(

danieledwardgeorgehitchcock, 2021-08-24:
What seems to be strange is that I have looked at the Zigbee Herdsman Converter's device entry for this device, and it references another issue Koenkk/zigbee2mqtt#6313

The interesting thing here is that the database.db entry for the _TZ3000_pzui3skt device you attached above, and the one in the post appears to have a genPowerCfg cluster so, your _TZ3000_fkp5zyho seems to be a different device that doesn't report battery, and should probably have its own device handler.

Out of curiosity, did you get any errors when pairing the device?

snp88, 2021-08-24:
No issue during the paring but method  of the paring is different

_TZ3000_pzui3skt (good devices) required to hold button  

_TZ3000_fkp5zyho (bad devices) have separate parring buton on the board 

So the board inside for sure are different ;)

danieledwardgeorgehitchcock, 2021-08-25:
While this doesn't help your battery situation, do you use the Zigbee2MQTT GUI?

I was wondering if you could paste screenshots of each of the devices about screens.

Reason for requesting this is that I cannot see anywhere in the Zigbee2MQTT codebase where the above manufacturer ID's are specified so, I am wondering how they have been picked up by Zigbee2MQTT.

As mentioned above, there probably needs to be some additions made to the supported devices to make sure these are properly handled.

If you could also post up manufacturers / models / links / documentation for the devices, I can make sure that this is all added correctly..

snp88, 2021-08-25:
Wrong one
Link to aliexpress where i bought the devices 
[https://pl.aliexpress.com/item/4001095272467.html?spm=a2g0s.9042311.0.0.16745c0f8omyCp](url) 
[image]
[image]
[image]

Good one
Link to aliexpress where i bought the devices 
[https://pl.aliexpress.com/item/1005001490468150.html?spm=a2g0o.productlist.0.0.26dd4164pchoRt&algo_pvid=21e74314-e42b-449a-8beb-c8990844a528&algo_exp_id=21e74314-e42b-449a-8beb-c8990844a528-1](url)
[image]
[image]
[image]

DavyRoswinkel, 2021-09-17:
I got the same issue with multiple TS0043 models and only one TS0041 model.

When a button is pressed on the affected switch the battery percentage returns: null%
Weird issue is it also reports:

action
Triggered action (e.g. a button click)    Empty string("") 

While the button does fire the action ( in this case turning on or off a light)

In the device list ( z2m ui) the battery % ( is blinking red)
Zigbee Model: TS0041
Zigbee Manufacturer: _TYZB02_keyjqthh
( one button switch)

Zigbee Model: TS0043
Zigbee Manufacturer:  _TYZB02_key8kk7r
( 3 button switch)
Zigbee Model: TS0043
Zigbee Manufacturer: _TZ3000_qzjcsmar
( 3 button switch)

The model that works: 
Zigbee Model: TS0041
Zigbee Manufacturer:  _TZ3400_keyjqthh
( 1 button switch)

kevincaradant, 2021-10-02:
Any update on this issue?
I think a lot of devices under tuya have this issue about battery level 

Thank you

closed 2021-11-09 by github-actions[bot]