Koenkk/zigbee2mqtt #8501: seMetering 'currentSummDelivered' not assigned to endpoint

Ungederon, 2021-08-24:
Hi,
I have made a custom device for metering energy from 4 devices and cant get it working because "currentSummDelivered" value is  not assigned to correct endpoint. "activePower" from 'haElectricalMeasurement' cluster works fine.
```
Starting Zigbee2MQTT version 1.21.0 (commit #70891eec)
Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"}'
```
Code fragment in custom_devices.js:
```
          fromZigbee: [fz.electrical_measurement, fz.metering],
	  toZigbee: [],
	  exposes: [
			e.energy().withEndpoint(1),
			e.energy().withEndpoint(2),
			e.energy().withEndpoint(3),
			e.energy().withEndpoint(4),
                        e.power().withEndpoint(1),
			e.power().withEndpoint(2),
			e.power().withEndpoint(3),
			e.power().withEndpoint(4),
                   ],
	  meta: {multiEndpoint: true},
	  configure: async (device, coordinatorEndpoint, logger) =>
	  {
            const ep1 = device.getEndpoint(1);
            await reporting.bind(ep1, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
            await readPowerMultDiv(ep1, logger);
	    await reporting.readMeteringMultiplierDivisor(ep1);

            const ep2 = device.getEndpoint(2);
            await reporting.bind(ep2, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
            await readPowerMultDiv(ep2, logger);
	    await reporting.readMeteringMultiplierDivisor(ep2);

            const ep3 = device.getEndpoint(3);
            await reporting.bind(ep3, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
            await readPowerMultDiv(ep3, logger);
	    await reporting.readMeteringMultiplierDivisor(ep3);

            const ep4 = device.getEndpoint(4);
            await reporting.bind(ep4, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
            await readPowerMultDiv(ep4, logger);
	    await reporting.readMeteringMultiplierDivisor(ep4);
	  }
```
And here is fragment from log:
```
debug 2021-08-24 11:15:08: Received Zigbee message from '0x00124b001f853aba', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1]}' from endpoint 1 with groupID 0
info  2021-08-24 11:15:08: MQTT publish: topic 'zigbee2mqtt/0x00124b001f853aba', payload '{"energy":1,"energy_1":null,"energy_2":null,"energy_3":null,"energy_4":null,"linkquality":47,"power_1":0,"power_2":0,"power_3":37,"power_4":38,"power_l1":60,"power_l2":0,"power_l3":86,"power_l4":238}'
debug 2021-08-24 11:15:08: Received Zigbee message from '0x00124b001f853aba', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 2 with groupID 0
info  2021-08-24 11:15:08: MQTT publish: topic 'zigbee2mqtt/0x00124b001f853aba', payload '{"energy":0,"energy_1":null,"energy_2":null,"energy_3":null,"energy_4":null,"linkquality":47,"power_1":0,"power_2":0,"power_3":37,"power_4":38,"power_l1":60,"power_l2":0,"power_l3":86,"power_l4":238}'
debug 2021-08-24 11:15:08: Received Zigbee message from '0x00124b001f853aba', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 2 with groupID 0
debug 2021-08-24 11:15:08: Received Zigbee message from '0x00124b001f853aba', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,7]}' from endpoint 3 with groupID 0
info  2021-08-24 11:15:08: MQTT publish: topic 'zigbee2mqtt/0x00124b001f853aba', payload '{"energy":0,"energy_1":null,"energy_2":null,"energy_3":null,"energy_4":null,"linkquality":47,"power_1":0,"power_2":0,"power_3":37,"power_4":38,"power_l1":60,"power_l2":0,"power_l3":86,"power_l4":238}'
info  2021-08-24 11:15:08: MQTT publish: topic 'zigbee2mqtt/0x00124b001f853aba', payload '{"energy":7,"energy_1":null,"energy_2":null,"energy_3":null,"energy_4":null,"linkquality":47,"power_1":0,"power_2":0,"power_3":37,"power_4":38,"power_l1":60,"power_l2":0,"power_l3":86,"power_l4":238}'
debug 2021-08-24 11:15:08: Received Zigbee message from '0x00124b001f853aba', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10]}' from endpoint 4 with groupID 0
info  2021-08-24 11:15:08: MQTT publish: topic 'zigbee2mqtt/0x00124b001f853aba', payload '{"energy":10,"energy_1":null,"energy_2":null,"energy_3":null,"energy_4":null,"linkquality":47,"power_1":0,"power_2":0,"power_3":37,"power_4":38,"power_l1":60,"power_l2":0,"power_l3":86,"power_l4":238}'
```
As You can see from the log, value of "currentSummDelivered" is assigned to "energy", not to "energy_1", "energy_2", "energy_3" or "energy_4".

danieledwardgeorgehitchcock, 2021-08-24:
Can you paste your database.db entry for this device?

Ungederon, 2021-08-24:
```
{
	"id": 2,
	"type": "Router",
	"ieeeAddr": "0x00124b001f853aba",
	"nwkAddr": 25405,
	"manufId": 0,
	"manufName": "DGS",
	"powerSource": "Mains (single phase)",
	"modelId": "Power Meter",
	"epList": [
		1,
		2,
		242,
		3,
		4
	],
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 1281,
			"inClusterList": [
				0,
				3
			],
			"outClusterList": [
				0
			],
			"clusters": {
				"genBasic": {
					"attributes": {
						"modelId": "Power Meter",
						"manufacturerName": "DGS",
						"powerSource": 1,
						"zclVersion": 1,
						"hwVersion": 1,
						"dateCode": "20210707"
					}
				},
				"haElectricalMeasurement": {
					"attributes": {
						"acPowerMultiplier": 1,
						"acPowerDivisor": 1,
						"activePower": 12
					}
				},
				"seMetering": {
					"attributes": {
						"multiplier": 1,
						"divisor": 1,
						"currentSummDelivered": [
							0,
							1
						]
					}
				}
			},
			"binds": [
				{
					"cluster": 2820,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0014b88763",
					"endpointID": 1
				},
				{
					"cluster": 1794,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0014b88763",
					"endpointID": 1
				}
			],
			"configuredReportings": [],
			"meta": {}
		},
		"2": {
			"profId": 260,
			"epId": 2,
			"devId": 1281,
			"inClusterList": [
				0,
				3
			],
			"outClusterList": [
				0
			],
			"clusters": {
				"genBasic": {
					"attributes": {}
				},
				"haElectricalMeasurement": {
					"attributes": {
						"acPowerMultiplier": 1,
						"acPowerDivisor": 1,
						"activePower": 0
					}
				},
				"seMetering": {
					"attributes": {
						"multiplier": 1,
						"divisor": 1,
						"currentSummDelivered": [
							0,
							0
						]
					}
				}
			},
			"binds": [
				{
					"cluster": 2820,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0014b88763",
					"endpointID": 1
				},
				{
					"cluster": 1794,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0014b88763",
					"endpointID": 1
				}
			],
			"configuredReportings": [],
			"meta": {}
		},
		"3": {
			"profId": 260,
			"epId": 3,
			"devId": 1281,
			"inClusterList": [
				0,
				3
			],
			"outClusterList": [
				0
			],
			"clusters": {
				"genBasic": {
					"attributes": {}
				},
				"haElectricalMeasurement": {
					"attributes": {
						"acPowerMultiplier": 1,
						"acPowerDivisor": 1,
						"activePower": 85
					}
				},
				"seMetering": {
					"attributes": {
						"multiplier": 1,
						"divisor": 1,
						"currentSummDelivered": [
							0,
							7
						]
					}
				}
			},
			"binds": [
				{
					"cluster": 2820,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0014b88763",
					"endpointID": 1
				},
				{
					"cluster": 1794,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0014b88763",
					"endpointID": 1
				}
			],
			"configuredReportings": [],
			"meta": {}
		},
		"4": {
			"profId": 260,
			"epId": 4,
			"devId": 1281,
			"inClusterList": [
				0,
				3
			],
			"outClusterList": [
				0
			],
			"clusters": {
				"genBasic": {
					"attributes": {}
				},
				"haElectricalMeasurement": {
					"attributes": {
						"acPowerMultiplier": 1,
						"acPowerDivisor": 1,
						"activePower": 100
					}
				},
				"seMetering": {
					"attributes": {
						"multiplier": 1,
						"divisor": 1,
						"currentSummDelivered": [
							0,
							10
						]
					}
				}
			},
			"binds": [
				{
					"cluster": 2820,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0014b88763",
					"endpointID": 1
				},
				{
					"cluster": 1794,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0014b88763",
					"endpointID": 1
				}
			],
			"configuredReportings": [],
			"meta": {}
		},
		"242": {
			"profId": 41440,
			"epId": 242,
			"devId": 97,
			"inClusterList": [],
			"outClusterList": [
				33
			],
			"clusters": {},
			"binds": [],
			"configuredReportings": [],
			"meta": {}
		}
	},
	"hwVersion": 1,
	"dateCode": "20210707",
	"zclVersion": 1,
	"interviewCompleted": true,
	"meta": {
		"configured": -1818186631
	},
	"lastSeen": 1629805290337
}
```

danieledwardgeorgehitchcock, 2021-08-24:
There's a couple of things that may be causing this.. 1st one, I can see in the comments in fromZigbee that When using this converter also add the following to the configure method of the device: `await readEletricalMeasurementConverterAttributes(endpoint);`

Secondly, I don't think the converter for metering is designed for multiple endpoints so, will just report back the one attibute. It may be worth creating a new fromZigbee converter with the electrical_measurement converter as a base and a postfixWithEndpointName to see if you can yeild any further results that way?

danieledwardgeorgehitchcock, 2021-08-24:
Ignore the second comment - I can see that the electrical_measurement converter does indeed use the `postfixWithEndpointName`

Ungederon, 2021-08-24:
reporting.js does not contain `await readMeteringPowerConverterAttributes(endpoint);` function but contains `async function readMeteringMultiplierDivisor(endpoint);` - this function is used.
And will try to create local converter later today...

danieledwardgeorgehitchcock, 2021-08-24:
You are correct - Reporting doesn't however, fromZigbee does so, if you just paste the two entries as they are above in your config entry (taking into account your multiple endpoints), you may see some more information come through :-)

Ungederon, 2021-08-24:
Thanks for help, now it works :)
Did following:
```
const fzLocal = {
    dgsSeMetering: {
        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            payload = fz.metering.convert(model, msg, publish, options, meta);
            let energy_ep = utils.postfixWithEndpointName('energy', msg, model);
            payload[energy_ep] = payload.energy;
            delete payload.energy;
            return payload;
        },
    },
};
```
and 
`fromZigbee: [fz.electrical_measurement, fzLocal.dgsSeMetering],`

closed 2021-08-25 by Ungederon