Koenkk/zigbee2mqtt #8502: Philips Hue white ambiance GU10 350 Lumen 9290019533 5309030P9 not supported

nexro567, 2021-08-24:
Added the new bulb with hand.
Please make it soon supported because they sell them in the new Philips Hue Spot light Runner Extension EAN 8718696175392.

## Information about the device + link

`device_type
Router
zigbee_model
5309030P9
zigbee_manufacturer
Philips
support_status
not_supported

ieee_address
0x0017880109d76a85
network_address
0x2960
firmware_build_date
20210422
firmware_version
1.88.2`

## data/database.db entry of the device

`{"id":2,"type":"Router","ieeeAddr":"0x0017880109d76a85","nwkAddr":10592,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"5309030P9","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,64515,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"5309030P9","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":3,"dateCode":"20210422","swBuildId":"1.88.2"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":3,"dateCode":"20210422","swBuildId":"1.88.2","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1629807098255}`

Configuration.yaml:

`external_converters:
  - 5309030P9.js`

5309030P9.js:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['5309030P9'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '5309030P9', // Vendor model number, look on the device for a model number
    vendor: 'Philips', // Vendor of the device (only used for documentation and startup logging)
    description: 'Philips Hue White Ambiance GU10 light', // Description of the device, copy from vendor site. (only used for
    extend: extend.light_onoff_brightness_colortemp_color(),

};

module.exports = definition;`

danieledwardgeorgehitchcock, 2021-08-24:
You should be able to extend the existing device handler in Zigbee Herdsman Converters for the white GU10.

Do you want me to submit a Pull Request with the changes on your behalf?

nexro567, 2021-08-24:
> You should be able to extend the existing device handler in Zigbee Herdsman Converters for the white GU10.
> 
> Do you want me to submit a Pull Request with the changes on your behalf?

Hi @danieledwardgeorgehitchcock,

Thanks for the reply. The whtite GU10 works fine now. If you can make a pull request, yes please :).

Stephan.

danieledwardgeorgehitchcock, 2021-08-24:
Done!

Based this on information provided above, and [here](https://www.philips-hue.com/en-gb/p/hue-white-ambiance-runner-single-spotlight-ext/5309030P9)

closed 2021-08-24 by Koenkk