Koenkk/zigbee2mqtt #8503: Initial build fails on Widows

cinadr, 2021-08-24:
## What happened

After checking out 1.21.0and running npm ci (which finished ok) I got error while running npm start:

```
PS C:\zigbee2mqtt> npm start

> zigbee2mqtt@1.21.0 start
> node index.js

Building Zigbee2MQTT... (initial build), failed
node:child_process:333
      ex = new Error('Command failed: ' + cmd + '\n' + stderr);
           ^

Error: Command failed: npm run build
'node_modules' is not recognized as an internal or external command,
operable program or batch file.

    at ChildProcess.exithandler (node:child_process:333:12)
    at ChildProcess.emit (node:events:379:20)
    at maybeClose (node:internal/child_process:1065:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:296:5) {
  killed: false,
  code: 1,
  signal: null,
  cmd: 'npm run build'
}
```

## What did you expect to happen

Expect to finish initial build.

## How to reproduce it (minimal and precise)

1. Clean install on Windows
2. npm ci
3. npm start

## Debug info
Windows server 2019 x64
Node version: 15.6.0
NPM version: 7.21.0
Zigbee2MQTT version: 1.21.0
Adapter hardware:  CC2656RB (Slaesh)
Adapter firmware version: 

## Workaround:
changed package.json slashes to backslashes and it started up:
```
  "scripts": {
    "build": "node_modules\\.bin\\tsc && node index.js writehash",
    "build-watch": "node_modules\\.bin\\tsc --watch",
```

Koenkk, 2021-08-24:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/8410, should already be fixed in the dev branch.

closed 2021-08-24 by Koenkk

cinadr, 2021-08-24:
Thanks, sorry for duplicate...