Koenkk/zigbee2mqtt #8504: Z2M Not Started. TypeError: Cannot read property 'length' of null

idMasjid, 2021-08-24:
## What happened
I have tried different solutions but I can't start the Z2M service.
It shows me a type error: TypeError: Cannot read property 'length' of null
I did indicate the logical link in the configuration file
log file here:
https://pastebin.com/JCnARzhX
## What did you expect to happen
Zigbee2Mqtt Started
## How to reproduce it (minimal and precise)
see above
## Debug info
Zigbee2MQTT version: 1.21.0-4 in homeassistant
Adapter hardware: usb-Silicon_Labs_slae.sh_cc2652rb_stick
Adapter firmware version:?? I don't know

Koenkk, 2021-08-24:
Can you try reflashing your coordinator with the latest firmware?

avdgaag1984, 2021-10-07:
how can i see the last firmware on the zigbee stick?

diazdeespada, 2021-12-03:
I have the same problem. how can install the last firmware? thanks

avdgaag1984, 2021-12-04:
I have don it with this way: 
https://electrolama.com/radio-docs/flash-ti-flash-prog/

rhpijnacker, 2021-12-12:
I'm running in to this problem too.
Is it possible to fix this without flashing firmware?
This used to work until yesterday (upgrade home automation OS and other components)

avdgaag1984, 2021-12-14:
> I'm running in to this problem too. Is it possible to fix this without flashing firmware? This used to work until yesterday (upgrade home automation OS and other components)

what is the type number of your zigbee stick?

rhpijnacker, 2021-12-14:
> what is the type number of your zigbee stick?

Never mind, I got it working in the end. Turns out I had another component open the device at the same time which was causing issues.