Koenkk/zigbee2mqtt #8505: Hue wall switch module - No converter available

bonobo86, 2021-08-24:
## What happened
Pressing button on philips hue wall switch leads to "No converter available"
## What did you expect to happen
Proper reaction regarding pressing the button.
## How to reproduce it (minimal and precise)
Press button after device joins the network.
## Debug info
Zigbee2MQTT version:1.21.0
Adapter hardware:CC26X2R1
Adapter firmware version:1.2.7 - zStack3x0

Logs:https://pastebin.com/HqE2tma1

Koenkk, 2021-08-24:
Can you try to reconfigure the device via the frontend? (press the yellow button after  clicking on the device on the frontend)

bonobo86, 2021-08-24:
Hi,
still the same behavoir. 
First time I tried your advise, I got the following error:
https://pastebin.com/YZR0nByS
I think it is because the device was asleep. I tried your advice right after pressing the button once, to get the device awake, but I got no feedback. And the next time I tried to press the button, the same "no converter" message was in the logs.

Koenkk, 2021-08-25:
>  but I got no feedback. 

You should at least see a `Successfully configured` in the log, can you try again?

bonobo86, 2021-09-03:
Hi,
I found time to repeat the steps with a restart of zigbee2mqtt. Now it was successful, but I receive an error that the new device ID is unknown.

**info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x7cb03eaa00a986ee"},"type":"entity_not_found"}'
error Entity '0x7cb03eaa00a986ee' is unknown**

Here the logs(copied form the browser, sorry for the mess):https://pastebin.com/Jc4rAnWL