Koenkk/zigbee2mqtt #8510: Aqara Smart Lightstrip Driver / ZNDDMK11LM

keniji, 2021-08-25:
## Information about the device + link
There is no link currently, but Aqara has a new product "Aqara Smart Lightstrip Driver" (model: ZNDDMK11LM / ZNDD). As I checked with some retailer, this product could be an upgrade from "Aqara Smart Dimmer Controller T2" (model: [ZNTGMK11LM](https://www.zigbee2mqtt.io/devices/ZNTGMK11LM.html / ZNTG), which Z2M already supported. And I found I can hardly find ZNTG on the China market, most of the retailers only sell ZNDD now.
Altough I don't have ZNTG on my hand, I think ZNDD has the same design as ZNTG, so I use the same code to add ZNDD, but, I found that I cannot change ZNDD to "color temperature" mode, which ZNTG does not supported, that is, there is no such option for me to do that. So I went to buy an Aqara Gateway and add ZNDD with Aqara's official APP, and get the below screenshots:
[image]
_Lightings to be supported_

[image]
_When we switch ZNDD from "RGBW mode" to "Dual color temperature mode", it will prompt 2 lights to be chosen_

[image]
_When ZNDD is under "Dual color temperature mode", 2 lightstrips can be controlled individually_

## data/database.db entry of the device
`{"id":11,"type":"Router","ieeeAddr":"0x04cf8cdf3c8a6dc7","nwkAddr":32014,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.dimmer.rcbac1","epList":[1,2,21,242,3,31,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,4,3,8,5,768,6,2820,1794,2],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.dimmer.rcbac1","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":21,"stackVersion":2,"hwVersion":2,"dateCode":"08-13-2020"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":24,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":269,"inClusterList":[3,4,5,6,8,768,64704],"outClusterList":[],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":24,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"epId":4,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":269,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"31":{"epId":31,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":2,"hwVersion":2,"dateCode":"08-13-2020","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1629880154722}`

Please feel free to let me know if any further log is needed, thanks!

keniji, 2021-08-28:
> [How To Sniff Zigbee Traffic](https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

Have some tries but still fail to decrypt the message -- I'm stuck in getting the Transport Key (network key).

I am able to find the right channel for my Aqara E1 hub and sinff it's traffics, then I tried to pair the new device (ZNDDMK11LM) to it, but around the "Association Response" packet, which shows "Association Successfully":
[image]

I can only found this message seems have the Transport Key -- but it's undecoded:
[image]

I've congfigured the Trust Center link key before doing the sniffing, so am I still missing anything?
[image]

keniji, 2021-09-14:
Hi @Koenkk , so sorry to involve you here, but I seem to have stuck in getting the Transport Key to decode the payloads... Do you have any idea about how I can do that? Is it because the added Trust Center link key is wrong?

Koenkk, 2021-09-14:
@keniji the transport key should indeed be encrypted with the trust link center key: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html#3-sniffing-traffic (note that some bridges like hue use a different key)

keniji, 2021-09-15:
> @keniji the transport key should indeed be encrypted with the trust link center key: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html#3-sniffing-traffic (note that some bridges like hue use a different key)

Thanks for the reply. I understand that the transport key is encrypted with the TLC key, and I've already added it, so I guess the TLC key used by my hub ([Aqara Hub E1](https://homekitnews.com/2021/08/21/aqara-hub-e1-review/)) could be different, do you have any idea about how I can get the correct one? I found in [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/6524#issuecomment-854991491) @tavalin is also using the same hub, but not sure which TLC key he was using...

tavalin, 2021-09-15:
> > @keniji the transport key should indeed be encrypted with the trust link center key: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html#3-sniffing-traffic (note that some bridges like hue use a different key)
> 
> Thanks for the reply. I understand that the transport key is encrypted with the TLC key, and I've already added it (5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39), so I guess the TLC key used by my hub ([Aqara Hub E1](https://homekitnews.com/2021/08/21/aqara-hub-e1-review/)) could be different?? do you have any idea about how I can get the correct one? I found in [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/6524#issuecomment-854991491) @tavalin is also using the same hub, but not sure which TLC key he was using...

The TLC is the same, but every time you reset the hub to factory, it will generate a new transport key.  I think I had to pair an old ZigBee device in order to extract the transport key, rather than pair a ZigBee 3.0 device.

keniji, 2021-09-15:
> The TLC is the same, but every time you reset the hub to factory, it will generate a new transport key. I think I had to pair an old ZigBee device in order to extract the transport key, rather than [...]

You right @tavalin ! I paired a SJCGQ11LM and the transport key is out now... Many thanks!!

Hi @Koenkk , so I got the log here:
[switch_mode_for_ZNDDMK11LM.zip](https://github.com/Koenkk/zigbee2mqtt/files/7169475/switch_mode_for_ZNDDMK11LM.zip)
The traffic should be clean since there is no other device on the same channel, and the network key should be: `90:fa:71:43:bd:eb:11:ab:f3:45:9b:c4:23:fb:50:05`
Would you please take a look and see how we can switch the device's mode with Z2M? Thanks in advance!

Koenkk, 2021-09-15:
@keniji this converter should do it:

```js
ZNDDMK11LM_mode: {
    key: ['device_mode'],
    convertSet: async (entity, key, value, meta) => {
        const lookup = {'rgbw': 1, 'dual_ct': 3};
        value = value.toLowerCase();
        utils.validateValue(value, Object.keys(lookup));
        await entity.write('aqaraOpple', {0x050f: {value: lookup[value], type: 0x23}}, manufacturerOptions.xiaomi);
        return {state: {device_mode: value}};
    },
},
```

Add it to the `toZigbee` list and send e.g. `zigbee2mqtt/my_device/set` with payload `{"device_mode": "rgbw"}`

keniji, 2021-09-15:
> @keniji this converter should do it:
>
> ```js
> ZNDDMK11LM_mode: {
>     key: ['device_mode'],
>     convertSet: async (entity, key, value, meta) => {
>         const lookup = {'rgbw': 1, 'dual_ct': [...]

Thanks again @Koenkk ! The lightstrips are currently not beside so I will test it tomorrow. And BTW how do I control the lightstrips separately when the device is under the "dual_ct" mode? Will there be any topic for me to do that then?

Koenkk, 2021-09-15:
> Will there be any topic for me to do that then?

Can you provide a sniff of how this works with the Xiaomi gateway?

keniji, 2021-09-16:
> > Will there be any topic for me to do that then?
> 
> Can you provide a sniff of how this works with the Xiaomi gateway?

Sure.

As I mentioned before, when I switch the device from RGBW to dual CT, there are 2 lights can be selected:
[image]
Since I'm not sure if only select one of them will make any difference when I turn off/on the light, I captured 3 logs for different scenarios:
[control_state_for_ZNDDMK11LM_01.zip](https://github.com/Koenkk/zigbee2mqtt/files/7174707/control_state_for_ZNDDMK11LM_01.zip)
`turn_off_light_1_then_light_2.pcapng` : Both lights are enabled, and I turn off light 1, then turn off light 2.
`turn_off_only_exist_light_1.pcapng` : Only `Light 1` is enabled, and I turn off the only light 1.
`turn_off_only_exist_light_2.pcapng` : Only `Light 2` is enabled, and I turn off the only light 2.

And I captured 2 more logs for switching the device to dual CT mode:
[switch_mode_for_ZNDDMK11LM_02.zip](https://github.com/Koenkk/zigbee2mqtt/files/7174714/switch_mode_for_ZNDDMK11LM_02.zip)
`enable_dualCT_only_light_1.pcapng` : Only `Light 1` is selected to be enabled.
`enable_dualCT_only_light_2.pcapng` : Only `Light 2` is selected to be enabled.

BTW this device also supports the "Power-off memory" feature:
[image]
So can I use the same command to enable/disable it? Just like [SSM-U01](https://www.zigbee2mqtt.io/devices/SSM-U01.html)?

Koenkk, 2021-09-18:
Try with adding the following to your definition:

```js
        exposes: [e.light_brightness_colortemp().withEndpoint('l1'), e.light_brightness_colortemp().withEndpoint('l2')],
        endpoint: (device) => {
            return {l1: 1, l2: 2};
        },
```

keniji, 2021-09-19:
> Try with adding the following to your definition:
> 
> ```js
>         exposes: [e.light_brightness_colortemp().withEndpoint('l1'), e.light_brightness_colortemp().withEndpoint('l2')],
>         endpoint: (device) => {
>             return {l1: 1, l2: 2};
>         },
> ```

Had a test just now...

Should I add the "ZNDDMK11LM_mode" converter to toZigbee.js directly? (If yes, where should I map the file as I am using docker)
Since when I add the converted to the external converter file, I get an error:
```
Using '/app/data' as data directory
(node:7) UnhandledPromiseRejectionWarning: SyntaxError: Unexpected token ':'
(Use `node --trace-warnings ...` to show where the warning was created)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:7) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```
And btw this is the external converter file now:
[ZNDDMK11LM.zip](https://github.com/Koenkk/zigbee2mqtt/files/7192334/ZNDDMK11LM.zip)

Koenkk, 2021-09-20:
@keniji you can define the converter locally, example: https://github.com/Koenkk/zigbee2mqtt.io/blob/28e2cb326db99c8e17b9d9477a90d597c1b6f77f/docs/externalConvertersExample/freepad_ext.js#L43

keniji, 2021-09-20:
> @keniji you can define the converter locally, example: https://github.com/Koenkk/zigbee2mqtt.io/blob/28e2cb326db99c8e17b9d9477a90d597c1b6f77f/docs/externalConvertersExample/freepad_ext.js#L43

Thanks, so my files is now:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    ZNDDMK11LM_mode: {
        key: ['device_mode'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = {'rgbw': 1, 'dual_ct': 3};
            value = value.toLowerCase();
            utils.validateValue(value, Object.keys(lookup));
            await entity.write('aqaraOpple', {0x050f: {value: lookup[value], type: 0x23}}, manufacturerOptions.xiaomi);
            return {state: {device_mode: value}};
        },
    },
};

const definition = {
    zigbeeModel: ['lumi.dimmer.rcbac1'],
    model: 'ZNDDMK11LM',
    vendor: 'Xiaomi',
    description: 'Aqara Smart Lightstrip Driver',
    fromZigbee: [],
    toZigbee: [tzLocal.ZNDDMK11LM_mode],
    exposes: [e.light_brightness_colortemp().withEndpoint('l1'), e.light_brightness_colortemp().withEndpoint('l2')],
    endpoint: (device) => {
        return {l1: 1, l2: 2};
    },
};

module.exports = definition;
```
But when I send `{"device_mode": "dual_ct"}` to `zigbee2mqtt/light_test/set` (the device is now named `light_test`) I get below erros in log:
```
Zigbee2MQTT:debug 2021-09-20 23:56:06: Received MQTT message on 'zigbee2mqtt/light_test/set' with data '{"device_mode": "dual_ct"}'
Zigbee2MQTT:debug 2021-09-20 23:56:06: Publishing 'set' 'device_mode' to 'light_test'
Zigbee2MQTT:error 2021-09-20 23:56:06: Publish 'set' 'device_mode' to 'light_test' failed: 'ReferenceError: utils is not defined'
Zigbee2MQTT:debug 2021-09-20 23:56:06: ReferenceError: utils is not defined
    at Object.convertSet (/app/dist/util/externally-loaded.js:16:13)
    at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:230:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:412:81)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
Zigbee2MQTT:info  2021-09-20 23:56:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'device_mode' to 'light_test' failed: 'ReferenceError: utils is not defined'","meta":{"friendly_name":"light_test"},"type":"zigbee_publish_error"}'
```
(Sending `{"device_mode": "rgbw"}` will get similar errors)
And the UI is:
[image]

Koenkk, 2021-09-20:
Below `const reporting = require('zigbee-herdsman-converters/lib/reporting');` try adding `const utils = require('zigbee-herdsman-converters/lib/utils');`

keniji, 2021-09-21:
> Below `const reporting = require('zigbee-herdsman-converters/lib/reporting');` try adding `const utils = require('zigbee-herdsman-converters/lib/utils');`

Thanks!

With you suggestion and according to the new errors, I added below lines to my file:
```
const utils = require('zigbee-herdsman-converters/lib/utils');
const herdsman = require('zigbee-herdsman');
...
const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};
```
And now when I send `{"device_mode": "dual_ct"}` to `zigbee2mqtt/light_test/set`, now I get errors in log:
```
Zigbee2MQTT:debug 2021-09-21 13:08:23: Received MQTT message on 'zigbee2mqtt/light_test/set' with data '{"device_mode": "dual_ct"}'
Zigbee2MQTT:debug 2021-09-21 13:08:23: Publishing 'set' 'device_mode' to 'light_test'
Zigbee2MQTT:debug 2021-09-21 13:08:23: Received Zigbee message from 'light_test', type 'attributeReport', cluster 'aqaraOpple', data '{"1295":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-09-21 13:08:23: No converter available for 'ZNDDMK11LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"1295":3}'
Zigbee2MQTT:info  2021-09-21 13:08:23: MQTT publish: topic 'zigbee2mqtt/light_test', payload '{"device_mode":"dual_ct","last_seen":1632200903378,"linkquality":99}'
Zigbee2MQTT:info  2021-09-21 13:08:23: MQTT publish: topic 'zigbee2mqtt/light_test', payload '{"device_mode":"dual_ct","last_seen":1632200903383,"linkquality":96}'
```

And the device is still uncontrolable from the frontend, when I try to set/refresh the state, I get errors in log:
```
Zigbee2MQTT:debug 2021-09-21 13:15:29: Received MQTT message on 'zigbee2mqtt/light_test/l1/get' with data '{"state_l1":""}'
Zigbee2MQTT:error 2021-09-21 13:15:29: No converter available for 'state' (undefined)
```
Any suggestion? Thanks again!

Koenkk, 2021-09-21:
Setting of device_mode seems to work correctly (`No converter available for 'ZNDDMK11LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"1295":3}'` can be ignored for now).

To control change `fromZigbee`/`toZigbee` to:

```js
    fromZigbee: [fz.color_colortemp, fz.on_off, fz.brightness],
    toZigbee: [tzLocal.ZNDDMK11LM_mode, tz.light_onoff_brightness, tz.light_colortemp],
```

keniji, 2021-09-22:
> Setting of device_mode seems to work correctly (`No converter available for 'ZNDDMK11LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"1295":3}'` can be ignored for now).
>
> To [...]

Thanks Koenkk!

I think the device works fine now under the dual CT mode, thats great because I will use this mode mainly... But I continue to send `{"device_mode": "rgbw"}` to `zigbee2mqtt/light_test/set` this time, to see if I can switch the device to RGBW mode (although actually I don't have a RGBW lightstrip beside me now), but at least from the frontend the device is still under the dual CT mode:
[image]
And see this in logs:
```
Zigbee2MQTT:debug 2021-09-22 21:38:34: Received MQTT message on 'zigbee2mqtt/light_test/set' with data '{"device_mode": "rgbw"}'
Zigbee2MQTT:debug 2021-09-22 21:38:34: Publishing 'set' 'device_mode' to 'light_test'
Zigbee2MQTT:debug 2021-09-22 21:38:34: Received Zigbee message from 'light_test', type 'attributeReport', cluster 'aqaraOpple', data '{"1295":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-09-22 21:38:34: No converter available for 'ZNDDMK11LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"1295":1}'
Zigbee2MQTT:info  2021-09-22 21:38:34: MQTT publish: topic 'zigbee2mqtt/light_test', payload '{"brightness":254,"brightness_l1":12,"brightness_l2":32,"color_mode":"color_temp","color_mode_l1":"color_temp","color_mode_l2":"color_temp","color_temp":153,"color_temp_l1":153,"color_temp_l2":291,"device_mode":"rgbw","last_seen":1632317914920,"linkquality":96,"state":"ON","state_l1":"OFF","state_l2":"OFF"}'
Zigbee2MQTT:info  2021-09-22 21:38:34: MQTT publish: topic 'zigbee2mqtt/light_test', payload '{"brightness":254,"brightness_l1":12,"brightness_l2":32,"color_mode":"color_temp","color_mode_l1":"color_temp","color_mode_l2":"color_temp","color_temp":153,"color_temp_l1":153,"color_temp_l2":291,"device_mode":"rgbw","last_seen":1632317914925,"linkquality":96,"state":"ON","state_l1":"OFF","state_l2":"OFF"}'
```
I also tried to re-pair the device after sending the payload, then send the payload again, nothing seems changed... So is it suppose to change it's mode automatically (with a corresponding frontend) after receiving the payload? Or should I do something to make it changed?

Koenkk, 2021-09-22:
@keniji try repairing it close to the coordinator. 

Note that the exposes cannot be dynamically changed, so it will always be displayed as 2 lights in the frontend. I also suggest to change `exposes: [e.light_brightness_colortemp().withEndpoint('l1'), e.light_brightness_colortemp().withEndpoint('l2')],` to `exposes: [e.light_onoff_brightness_colortemp_color().withEndpoint('l1'), e.light_onoff_brightness_colortemp_color().withEndpoint('l2')],`

keniji, 2021-09-23:
> @keniji try repairing it close to the coordinator.
>
> Note that the exposes cannot be dynamically changed, so it will always be displayed as 2 lights in the frontend. I also suggest to change [...]

Thanks again Koenkk!

First of all my file now is:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const herdsman = require('zigbee-herdsman');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const tzLocal = {
    ZNDDMK11LM_mode: {
        key: ['device_mode'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = {'rgbw': 1, 'dual_ct': 3};
            value = value.toLowerCase();
            utils.validateValue(value, Object.keys(lookup));
            await entity.write('aqaraOpple', {0x050f: {value: lookup[value], type: 0x23}}, manufacturerOptions.xiaomi);
            return {state: {device_mode: value}};
        },
    },
};

const definition = {
    zigbeeModel: ['lumi.dimmer.rcbac1'],
    model: 'ZNDDMK11LM',
    vendor: 'Xiaomi',
    description: 'Aqara Smart Lightstrip Driver',
    fromZigbee: [fz.color_colortemp, fz.on_off, fz.brightness],
    toZigbee: [tzLocal.ZNDDMK11LM_mode, tz.light_onoff_brightness, tz.light_colortemp, tz.light_color],
    exposes: [e.light_brightness_colortemp_color().withEndpoint('l1'), e.light_brightness_colortemp_color().withEndpoint('l2')],
    endpoint: (device) => {
        return {l1: 1, l2: 2};
    },
};

module.exports = definition;
```
The frontend looks and acts normal but the RGBW just not work, no matter which mode I switch the device to... As I neither actually have a RGBW lightstrip nor have any plan to buy one recently, and this model seems not very popular, I am thinking to just leave the job to someone who really need the RGBW feature in future... Or maybe I can pretend there is a RGBW light and we can further look into how to make both modes work? Since I only need to see if both connected CT lightstrips will change when I set the pretended-RGBW lightstrip to a color.

And BTW the frontend is now:
[image]

Koenkk, 2021-09-23:
Lets leave out the RGBW indeed for now, could you make a PR?

keniji, 2021-09-24:
> Lets leave out the RGBW indeed for now, could you make a PR?

Sure, let me test the `xiaomi_light_power_outage_memory` feature tonight and then I will try a PR, thanks!

keniji, 2021-09-24:
Hi @Koenkk , when I'm preparing the PR, I found the work to support this device on the dual CT mode is not finish yet...

When this model `ZNDDMK11LM` is first unboxing, it's under the RGBW mode, since not everyone has Aqara gateway, I think people need to use Z2M to initial it and switch it to the dual CT mode. However, as I mentioned yesterday, when I use `{"device_mode": "dual_CT"}`, I can only enable/control the first lightstripe/channel, and the second one is uncontrolable, the log is here:
```
debug 2021-09-24 22:47:32: Received MQTT message on 'zigbee2mqtt/light_test/l2/set' with data '{"state_l2":"OFF"}'
debug 2021-09-24 22:47:32: Publishing 'set' 'state' to 'light_test'
error 2021-09-24 22:47:55: Publish 'set' 'state' to 'light_test' failed: 'Error: Command 0x04cf8cdf3c8a6dc7/2 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 6503 - 2 - 4 - 6 - 11 after 10000ms)'
debug 2021-09-24 22:47:55: Error: Command 0x04cf8cdf3c8a6dc7/2 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 6503 - 2 - 4 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
info  2021-09-24 22:47:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'light_test' failed: 'Error: Command 0x04cf8cdf3c8a6dc7/2 genOnOff.off({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 6503 - 2 - 4 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"light_test"},"type":"zigbee_publish_error"}'
```
I'm thinking maybe the second one is not enabled by the command?

Please see [this reply](https://github.com/Koenkk/zigbee2mqtt/issues/8510#issuecomment-920554532), as when I use `Aqara Home` APP to switch to the dual CT mode, there are 2 lights can be enabled, so I sniff the traffics again with all 3 scenarios:
[issue_8510_20210924_logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/7226772/issue_8510_20210924_logs.zip)
`enable_dualCT_both_lights.pcapng`: Switch to dual CT with both `light 1` and `light 2` enabled.
`enable_dualCT_only_light1.pcapng`: Switch to dual CT with only `light 1` enabled.
`enable_dualCT_only_light2.pcapng`: Switch to dual CT with only `light 2` enabled.

Would you please help to check this again? Or if the command is all right, then do you have any idea about what will cause only `light 1` can be controlled, after the device is switch to dual CT from RGBW? Many thanks again!

Koenkk, 2021-09-24:
Can you try with this external converter? I've removed the option to set the device_mode, it should be in dual ct both by default:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const herdsman = require('zigbee-herdsman');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const definition = {
    zigbeeModel: ['lumi.dimmer.rcbac1'],
    model: 'ZNDDMK11LM',
    vendor: 'Xiaomi',
    description: 'Aqara smart lightstrip driver',
    fromZigbee: [fz.color_colortemp, fz.on_off, fz.brightness],
    toZigbee: [tz.light_onoff_brightness, tz.light_colortemp, tz.light_color],
    exposes: [e.light_brightness_colortemp().withEndpoint('l1'), e.light_brightness_colortemp().withEndpoint('l2')],
    endpoint: (device) => {
        return {l1: 1, l2: 2};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        await entity.write('aqaraOpple', {0x0509: {value: 1, type: 0x23}}, manufacturerOptions.xiaomi);
        await entity.write('aqaraOpple', {0x050f: {value: 3, type: 0x23}}, manufacturerOptions.xiaomi);
    },
};

module.exports = definition;

```

keniji, 2021-09-25:
> Can you try with this external converter? I've removed the option to set the device_mode, it should be in dual ct both by default:
>
> ```js
> const fz = [...]

Thanks @Koenkk !

It seems not work, I switch the device to RGBW mode with `Aqara Home` APP, then pair it to Z2M, then I got the same behavior as before, that is, I can only control the first lightstripe/channel, please see logs:
```
info  2021-09-25 15:48:32: Successfully interviewed 'light_test', device has successfully been paired
info  2021-09-25 15:48:32: Device 'light_test' is supported, identified as: Xiaomi Aqara smart lightstrip driver (ZNDDMK11LM)
info  2021-09-25 15:48:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara smart lightstrip driver","exposes":[{"endpoint":"l1","features":[{"access":7,"description":"On/off state of this light","endpoint":"l1","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","endpoint":"l1","name":"brightness","property":"brightness_l1","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","endpoint":"l1","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp_l1","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"access":7,"description":"Color temperature after cold power on of this light","endpoint":"l1","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup_l1","type":"numeric","unit":"mired","value_max":500,"value_min":150}],"type":"light"},{"endpoint":"l2","features":[{"access":7,"description":"On/off state of this light","endpoint":"l2","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","endpoint":"l2","name":"brightness","property":"brightness_l2","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","endpoint":"l2","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp_l2","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"access":7,"description":"Color temperature after cold power on of this light","endpoint":"l2","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup_l2","type":"numeric","unit":"mired","value_max":500,"value_min":150}],"type":"light"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZNDDMK11LM","supports_ota":false,"vendor":"Xiaomi"},"friendly_name":"light_test","ieee_address":"0x04cf8cdf3c8a6dc7","status":"successful","supported":true},"type":"device_interview"}'
info  2021-09-25 15:48:32: Configuring 'light_test'
error 2021-09-25 15:48:33: Failed to configure 'light_test', attempt 1 (ReferenceError: entity is not defined
    at Object.configure (/app/dist/util/externally-loaded.js:28:9)
    at Configure.configure (/app/lib/extension/configure.js:137:45)
    at Configure.onZigbeeEvent (/app/lib/extension/configure.js:119:18)
    at Controller.callExtensionMethod (/app/lib/controller.js:412:81)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at Controller.onZigbeeEvent (/app/lib/controller.js:284:24))
[... cut ...]
info  2021-09-25 15:49:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'light_test' failed: 'Error: Command 0x04cf8cdf3c8a6dc7/2 genOnOff.off({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 51447 - 2 - 16 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"light_test"},"type":"zigbee_publish_error"}'
```
Any idea?

Koenkk, 2021-09-25:
What if you set it to dual ct from the aqara bridge and then pair it?

keniji, 2021-09-26:
> What if you set it to dual ct from the aqara bridge and then pair it?

Tests of setting the device to dual CT from the `Aqara Home` APP then pair it to Z2M:
Enable both lights in the APP -- Both lights are working fine in Z2M.
Enable only light 1 in the APP -- It's strange that no light is controlable, I get "timeout" errors in the log.
Enable only light 2 in the APP -- Only light 2 works in Z2M.

I am surprised that both lights are uncontrolable if I only enable light 1 in the APP, I will further test this and update the log here later.

keniji, 2021-09-27:
> Tests of setting the device to dual CT from the `Aqara Home` APP then pair it to Z2M:
> Enable both lights in the APP -- Both lights are working fine in Z2M.
> Enable only light 1 in the APP -- It's [...]

Hi @Koenkk , so I confimred that the reason why `light 1` is uncontrolable yesterday is because I have click `light 2` carelessly, when `light 2` is not enabled, so the device is waiting the response from `light 2` and was not able to echo for `light 1`, so the confimred test result is below:

Enable both lights in the APP -- Both lights are working fine in Z2M.
Enable only light 1 in the APP -- Only light 1 works in Z2M.
Enable only light 2 in the APP -- Only light 2 works in Z2M.

I am not familiar with the ZigBee protocol, but I always see the logs tell that the device "requested OTA", would it be the reason?

Koenkk, 2021-09-27:
I found a bug in the code, with below it should set it in dual_ct mode by default:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const herdsman = require('zigbee-herdsman');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const definition = {
    zigbeeModel: ['lumi.dimmer.rcbac1'],
    model: 'ZNDDMK11LM',
    vendor: 'Xiaomi',
    description: 'Aqara smart lightstrip driver',
    fromZigbee: [fz.color_colortemp, fz.on_off, fz.brightness],
    toZigbee: [tz.light_onoff_brightness, tz.light_colortemp, tz.light_color],
    exposes: [e.light_brightness_colortemp().withEndpoint('l1'), e.light_brightness_colortemp().withEndpoint('l2')],
    endpoint: (device) => {
        return {l1: 1, l2: 2};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.write('aqaraOpple', {0x0509: {value: 1, type: 0x23}}, manufacturerOptions.xiaomi);
        await endpoint.write('aqaraOpple', {0x050f: {value: 3, type: 0x23}}, manufacturerOptions.xiaomi);
    },
};

module.exports = definition;

```

keniji, 2021-09-28:
> I found a bug in the code, with below it should set it in dual_ct mode by default:
>
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

Finally!! Many thanks again @Koenkk !!

Below is the external converter has been confirmed worked, I've formated the code by comparing `ZNTGMK11LM`'s so I think the module can be added right after `ZNTGMK11LM`:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const herdsman = require('zigbee-herdsman');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

module.exports = [
    {
        zigbeeModel: ['lumi.dimmer.rcbac1'],
        model: 'ZNDDMK11LM',
        vendor: 'Xiaomi',
        description: 'Aqara Smart Lightstrip Driver',
        exposes: [e.light_brightness_colortemp().withEndpoint('l1'), e.light_brightness_colortemp().withEndpoint('l2')],
        endpoint: (device) => {
            return {l1: 1, l2: 2};
        },
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.write('aqaraOpple', {0x0509: {value: 1, type: 0x23}}, manufacturerOptions.xiaomi);
            await endpoint.write('aqaraOpple', {0x050f: {value: 3, type: 0x23}}, manufacturerOptions.xiaomi);
        },
        extend: extend.light_onoff_brightness_colortemp_color({supportsHS: true}),
    },
];
```

Koenkk, 2021-09-28:
Thanks, as a last thing, can you provide the color temp range? How-to: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature

keniji, 2021-09-28:
> Thanks, as a last thing, can you provide the color temp range? How-to: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature

My pleasure :)
```
Zigbee2MQTT:info  2021-09-29 07:40:35: Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}
```

Koenkk, 2021-09-29:
Added, assuming this can be closed now.

closed 2021-09-29 by Koenkk

keniji, 2021-10-09:
> but how to switch to RGBW mode?

Sorry that I don't have RBGW lightstrip so I can't test to add this feature, so currently this is not yet supported by Z2M, I think you can open an issue and see what data is needed for adding it, and if so you're welcome to involve me there, in case there is anything I can do.

BaRaD5, 2025-06-06:
Not sure if the issue is still active, but i cannot seem to exit RGBW mode under Z2M, and switcing to dual ct mode will not make the L2 function so im stuck using RGBW (which is OK in my use)

Is there any way to progress this issue?