Koenkk/zigbee2mqtt #8511: Tuya Smart DIN  relay  Life ZigBee, 16 А

meandes, 2021-08-25:
## Information about the device + link
. Tuya Smart Life ZigBee, 16 А

https://aliexpress.ru/item/1005002983983361.html?spm=a2g39.orderlist.0.0.56894aa6pW9WGt&_ga=2.66528622.1702638691.1629749563-797364589.1629749563

Managed to add with adding this device to tuya.js (line 710 ) 

   { 
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_cphmq0q7'},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_ew3ldmgx'},           
            {modelID: 'TS011F', manufacturerName: '_TZ3000_ps3dmato'},
            **{modelID: 'TS011F', manufacturerName: '_TZ3000_8bxrzyxz'},**
            {modelID: 'TS011F', manufacturerName: '_TZ3000_jvzvulen'}],
        model: 'TS011F_plug',                             
        description: 'Smart plug (with power monitoring)',
        vendor: 'TuYa',                                       
        whiteLabel: [{vendor: 'LELLKI', model: 'TS011F_plug'}],
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],              
        configure: async (device, coordinatorEndpoint, logger) => {    
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);                         
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();                                                       
        },                         

## data/database.db entry of the device
..
{"id":25,"type":"Router","ieeeAddr":"0x0c4314fffe98d6ef","nwkAddr":31456,"manufId":4098,"manufName":"_TZ3000_8bxrzyxz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_8bxrzyxz","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":224,"rmsCurrent":0,"activePower":0,"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,0]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001940bc0c","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b001940bc0c","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001940bc0c","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1629809935867}root@smarthouse:/opt/zigbee2mqtt/data# 

And pairing log

info  2021-08-23 23:43:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0c4314fffe98d6ef"},"type":"pairing"}'
debug 2021-08-23 23:43:32: Device '0x0c4314fffe98d6ef' announced itself
info  2021-08-23 23:43:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe98d6ef","ieee_address":"0x0c4314fffe98d6ef"},"type":"device_announce"}'
info  2021-08-23 23:43:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0c4314fffe98d6ef"},"type":"device_announced"}'
debug 2021-08-23 23:43:33: Received Zigbee message from '0x0c4314fffe98d6ef', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS011F"}' from endpoint 1 with groupID 0
debug 2021-08-23 23:43:33: Skipping message, definition is undefined and still interviewing
debug 2021-08-23 23:43:33: Received Zigbee message from '0x0c4314fffe98d6ef', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_8bxrzyxz"}' from endpoint 1 with groupID 0
debug 2021-08-23 23:43:33: Skipping message, definition is undefined and still interviewing
debug 2021-08-23 23:43:35: Received Zigbee message from '0x0c4314fffe98d6ef', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
debug 2021-08-23 23:43:35: Skipping message, definition is undefined and still interviewing
debug 2021-08-23 23:43:37: Received Zigbee message from '0x0c4314fffe98d6ef', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2021-08-23 23:43:37: Skipping message, definition is undefined and still interviewing
debug 2021-08-23 23:43:37: Received Zigbee message from '0x0c4314fffe98d6ef', type 'readResponse', cluster 'genBasic', data '{"appVersion":67}' from endpoint 1 with groupID 0
debug 2021-08-23 23:43:37: Skipping message, definition is undefined and still interviewing
debug 2021-08-23 23:43:37: Received Zigbee message from '0x0c4314fffe98d6ef', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
debug 2021-08-23 23:43:37: Skipping message, definition is undefined and still interviewing
debug 2021-08-23 23:43:39: Received Zigbee message from '0x0c4314fffe98d6ef', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2021-08-23 23:43:39: Skipping message, definition is undefined and still interviewing
debug 2021-08-23 23:43:39: Received Zigbee message from '0x0c4314fffe98d6ef', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
debug 2021-08-23 23:43:39: Skipping message, definition is undefined and still interviewing
debug 2021-08-23 23:43:40: Received Zigbee message from '0x0c4314fffe98d6ef', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2021-08-23 23:43:40: Skipping message, definition is undefined and still interviewing
info  2021-08-23 23:43:40: Successfully interviewed '0x0c4314fffe98d6ef', device has successfully been paired
warn  2021-08-23 23:43:40: Device '0x0c4314fffe98d6ef' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_8bxrzyxz' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
info  2021-08-23 23:43:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x0c4314fffe98d6ef","ieee_address":"0x0c4314fffe98d6ef","status":"successful","supported":false},"type":"device_interview"}'
info  2021-08-23 23:43:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x0c4314fffe98d6ef","supported":false},"type":"pairing"}'

Koenkk, 2021-08-25:
Already supported in the dev branch: https://github.com/Koenkk/zigbee-herdsman-converters/blob/bd281afe13a9da2fcdbbdb92cc5af36c3f5f5768/devices/tuya.js#L1126 (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-25 by Koenkk

Z18Aplha, 2021-09-24:
I still have the same issue. My device appears as unknown with 

```
Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_dpo1ysak'
```

The manufacturer is different to the one in the mentioned fingerprint in [ https://github.com/Koenkk/zigbee-herdsman-converters/blob/bd281afe13a9da2fcdbbdb92cc5af36c3f5f5768/devices/tuya.js#L1126]( https://github.com/Koenkk/zigbee-herdsman-converters/blob/bd281afe13a9da2fcdbbdb92cc5af36c3f5f5768/devices/tuya.js#L1126)