Koenkk/zigbee2mqtt #8512: TuYa Smart water/gas valve modelID: 'TS0601', manufacturerName: '_TZE200_vrjkcam9'

im0t0, 2021-08-25:
## Information about the device + link
TuYa Smart water/gas valve
[https://aliexpress.ru/item/1005002832975190.html](https://aliexpress.ru/item/1005002832975190.html)

I tried to create an external converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_vrjkcam9'}],
        model: 'PM02D-TYZ',
        vendor: 'TuYa',
        description: 'Smart water/gas valve',
        extend: extend.switch(),
    };

module.exports = definition;
```

But the switch didn't react. In the log: 
Valve closed:

> Received Zigbee message from '0x847127fffe748211', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID 0
> No converter available for 'PM02D-TYZ' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":0}'

Valve open:

> Received Zigbee message from '0x847127fffe748211', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID 0
> No converter available for 'PM02D-TYZ' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":0}'

## data/database.db entry of the device
..

danieledwardgeorgehitchcock, 2021-08-25:
Given that this is a valve, I think it would be more appropriate to base this on a cover type device rather than a switch

im0t0, 2021-08-25:
Did the same as for similar Smart water/gas valve
[model: 'SM-AZ713'](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/useelink.js)
[model: 'ZN231392'](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js)

danieledwardgeorgehitchcock, 2021-08-25:
Could you try this - it may not work / may need tweaking as I am coding blind here:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    tuya_valve_test: {
        cluster: 'manuSpecificTuya',
        type: ['commandGetData', 'commandSetDataResponse'],
        convert: (model, msg, publish, options, meta) => {
            return {'state': msg.data.data[0] === 1 ? 'ON' : 'OFF'};            
        },
    },
};

const device = {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_vrjkcam9'}],
        model: 'PM02D-TYZ',
        vendor: 'TuYa',
        description: 'Smart water/gas valve',
        fromZigbee: [fzLocal.tuya_valve_test],
        toZigbee: [tz.on_off],
        meta: {multiEndpoint: true},
        exposes: [
            e.switch(),
        ],
    };

module.exports = device;
```

im0t0, 2021-08-25:
Now correctly displays the state of the switch when controlled on the device. But in Z2M it is not possible to control the switch.

danieledwardgeorgehitchcock, 2021-08-25:
Anything in the logs?

Can you paste your data/database.db entry for the device?

I have got a feeling that it doesn't use the genOnOff cluster at all for control...

Koenkk, 2021-08-25:
This one has already been added (https://github.com/Koenkk/zigbee-herdsman-converters/pull/2917/files) and is supported in the dev branch.  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-25 by Koenkk

spfuture, 2022-01-05:
I have the same PF-PM02D-TYZ device, but.

Debug Received Zigbee message from '0x5c0272fffe0a5688', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_yxcgyjf1","modelId":"TS0601"}' from endpoint 1 with groupID 0

Received Zigbee message from '0x5c0272fffe0a5688', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":0}' from endpoint 1 with groupID 0

Please add this device?