Koenkk/zigbee2mqtt #8514: Philips Hue 1100 Lumen Support

narc792, 2021-08-25:
Tried adding a the new device into zigbee2mqtt but adding the external converter through Home Assistant's file editor (both the file editor add on and the configuration on the supervisor page for zigbee2mqtt) resulted in not being able to start zigbee2mqtt properly. Happy to try adding it again myself if anyone has advice. 

## Information about the device + link
This is the new E26 Philips Hue White and Color Ambience Bulb  1100 lumen LCA007

https://www.philips-hue.com/en-us/p/hue-white-and-color-ambiance-1-pack-e26/046677563257

The code below should contain all the necessary information to add it.

const definition = {
        zigbeeModel: ['LCA007'],
        model: '9290024687',
        vendor: 'Philips',
        description: 'Hue White and Color Ambiance E26 1100 lumen',
        meta: {turnsOffAtBrightness1: true},
        extend: hueExtend.light_onoff_brightness_colortemp_color(),
        ota: ota.zigbeeOTA,
};
..

## data/database.db entry of the device
..

danieledwardgeorgehitchcock, 2021-08-25:
Is that your full external converter file above or just a snippet? If it is the full version, you are missing your includes at the beginning and the exports at the end which is why Zigbee2MQTT is probably not starting..

it should look something like this:

```javascript
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');

const device = {
    zigbeeModel: ['LCA007'],
    model: '9290024687',
    vendor: 'Philips',
    description: 'Hue White and Color Ambiance E26 1100 lumen',
    meta: {turnsOffAtBrightness1: true},
    extend: hueExtend.light_onoff_brightness_colortemp_color(),
    ota: ota.zigbeeOTA,
};

module.exports = device;
```

Do you know what the colorTempRange values are?

narc792, 2021-08-25:
> Is that your full external converter file above or just a snippet? If it is the full version, you are missing your includes at the beginning and the exports at the end which is why Zigbee2MQTT is probably not starting..
> 
> it should look something like this:
> 
> ```js
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> 
> const device = {
>     zigbeeModel: ['LCA007'],
>     model: '9290024687',
>     vendor: 'Philips',
>     description: 'Hue White and Color Ambiance E26 1100 lumen',
>     meta: {turnsOffAtBrightness1: true},
>     extend: extend.light_onoff_brightness_colortemp_color(),
>     ota: ota.zigbeeOTA,
> };
> 
> module.exports = device;
> ```
> 
> Do you know what the colorTempRange values are? Information on how to do that can be found [here](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature)

I'm pretty new at this but so I'm not super sure on how to properly do that but it should be (153, 500).

narc792, 2021-08-25:
> Is that your full external converter file above or just a snippet? If it is the full version, you are missing your includes at the beginning and the exports at the end which is why Zigbee2MQTT is [...]

Yes I have the additional things in the converter file. This is the complete one. 

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['LCA007'],
        model: '9290024687',
        vendor: 'Philips',
        description: 'Hue White and Color Ambiance A19 1100 lumen',
        meta: {turnsOffAtBrightness1: true},
        extend: Extend.light_onoff_brightness_colortemp_color(),
        ota: ota.zigbeeOTA,
};

module.exports = definition;`

danieledwardgeorgehitchcock, 2021-08-25:
You’ve got a capital E on extend

narc792, 2021-08-25:
> You’ve got a capital E on extend

Ahh. I think its capital because I tired hueExtend at one point. Not sure if I need to include all of that from the philips.js database or not, but it was definitely a roadblock.

danieledwardgeorgehitchcock, 2021-08-25:
No, you should just need the standard extend that you have referenced in your above file if you are using light_onoff_brightness_colortemp_color

narc792, 2021-08-25:
> No, you should just need the standard extend that you have referenced in your above file if you are using light_onoff_brightness_colortemp_color

So I fixed the capital E but zigbee2mqtt still will not start up. Logs report this.

```js
> node index.js
(node:330) UnhandledPromiseRejectionWarning: ReferenceError: ota is not defined
(Use `node --trace-warnings ...` to show where the warning was created)
(node:330) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:330) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

```

Is the way I'm referencing the converter incorrect? 

[image]