Koenkk/zigbee2mqtt #8552: Data from Xiaomi WXKG01LM devices not published to queue immediately

plaes, 2021-08-28:
## What happened
After upgrade to Zigbee2MQTT version 1.21.0 (from 1.17), I noticed that data based on Xiaomi WXKG01LM `action` is not sent immediately, but instead seems to sit in some kind of queue (until any other non-WXKG01LM device publishes data).

## How to reproduce it (minimal and precise)

1. Start following the logfile..
2. Press the Xiaomi WXKG01LM button
3. Observe for ~30 seconds that nothing is happening in the zigbee topics
4. Publish something else on the `zigbee2mqtt/xxx` topics
5. Data from Xiaomi WXKG01LM event is published

For example (first column is seconds):
```
T=00.0 - Push button on WSDCGQ01LM device
T+0.5s - Push Xiaomi WXKG01LM immediately after that
T+1.0s (notice that its data gets published)
... wait...
T+15.0s - Push button on WSDCGQ01LM
... data arrives from both WSDCGQ01LM and WXKG01LM
```

## Debug info
Zigbee2MQTT version:1.21.0 / 
Adapter hardware: CC2652P (dongle from https://smartlight.me/smart-home-devices/zigbee-devices/zigbee-coordinator-v4-cc2652p )
Adapter firmware version: 20210708 (and 20210319)

IctBtx, 2021-09-02:
Can confirm I'm also seeing the same behaviour after upgrading Zigbee2MQTT from 1.20.0 to 1.21.1 and HA from 2021.8.8 to 2021.9.0 on RPI4.

Zigbee2MQTT version: 1.21.1
Coordinator firmware: 20200805
Home Assistant: 2021.9.0

Koenkk, 2021-09-02:
Could you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Koenkk, 2021-09-02:
@pedrolamas sounds the same as your problem (https://github.com/Koenkk/zigbee2mqtt/issues/8271)

pedrolamas, 2021-09-02:
Agreed, it does sound similar!

Koenkk, 2021-09-02:
@IctBtx does downgrading z2m solve the problem?

IctBtx, 2021-09-02:
@Koenkk I did try reverting to my 1.20.0 backup earlier but still got the same behaviour.

I've also since upgraded the coordinator firmware to 20210708 but no change.

I'll get the herdsman debug log to you when I'm back at home but there might be a small delay before that happens.

(BTW Thanks for z2m - it's very rare I've ever had a problem with it!)

IctBtx, 2021-09-03:
@Koenkk Herdsman debug log attached. I single clicked "xiaomi_switch_4" roughly once every second, and you can see all the events grouped together at the end of the file. Hope this helps. 

[herdsman-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/7105829/herdsman-debug.log)

Koenkk, 2021-09-03:
Can you check if updating to the latest-dev solves the issue? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

IctBtx, 2021-09-03:
Just updated to dev and no change I'm afraid.

bossgos, 2021-09-05:
Same here :( 1.21.1 and HA core 2021.9.3. Response takes minutes from Xiaomi button and motion sensor..

Koenkk, 2021-09-05:
Seems the same as https://github.com/Koenkk/zigbee2mqtt/issues/8623 , looks to be related to the kernel

closed 2021-09-05 by Koenkk

plaes, 2021-09-07:
Yup, confirmed it being a kernel issue.